betaflight/betaflight #5314: Frsky R-XSR f.port default output values

Gozzz, 2018-02-25:
[image]

I'm using the R-XSR with frsky f.port firmware on a CLR4 (bardwell F4), and a BFF3. Both are showing the same default output when the RX is powered and the TX is not. See the screenshot attached. Once TX is powered the output values are normal. They fly fine and lua/telemetry seems to be working properly, failsafe also seems to work as when the TX is powered off again it drops throttle value below 900us. 

I've emailed Frsky but the BF f.port Wiki suggested providing feedback here.

Diff from the CLR4-
```# version
# Betaflight / CLRACINGF4 (CLR4) 3.3.0 Feb 18 2018 / 13:34:25 (f07bf40d1) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE

# beeper

# map

# serial
serial 2 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_hardware = NONE
set min_check = 1005
set rssi_channel = 8
set serialrx_provider = FPORT
set blackbox_device = NONE
set dshot_idle_value = 229
set motor_pwm_protocol = DSHOT1200
set pid_process_denom = 1
set osd_warnings = 63
set osd_rssi_alarm = 40
set osd_cap_alarm = 1000
set osd_tim2 = 769
set osd_vbat_pos = 2465
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2486
set osd_throttle_pos = 2454
set osd_current_pos = 2478
set osd_mah_drawn_pos = 2471
set osd_warnings_pos = 2441
set osd_disarmed_pos = 2410
set osd_stat_max_spd = OFF
set osd_stat_bb_no = OFF
set vtx_band = 2
set vtx_channel = 4
set vtx_freq = 5790

# profile
profile 0

set dterm_lowpass_type = PT1
set p_pitch = 47
set i_pitch = 46
set p_roll = 47
set i_roll = 46
set d_roll = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150
set roll_srate = 79
set pitch_srate = 79
set yaw_srate = 79

# ```

mikeller, 2018-02-25:
@gozz: The values sent by your RX when not connected to the TX are the failsafe values, and have to be set on the RX. When using Betaflight, these values should **always be set to 'no signal'**, otherwise Betaflight's failsafe mechanism will not be triggered when the connection to the TX is lost, and you'll be watching your craft fly away over the horizon (if you're lucky). In your case, it looks like you set your RX to send fixed values on failsafe. Please consult your RX's manual to find out how to change the failsafe setting to 'no signal'.

Gozzz, 2018-02-25:
Thanks mikeller. The R-XSR is bound and failsafe is set to no pulses. No pulses only seems to work if the RX and TX are powered and connected then RX loses signal/TX powered off. Otherwise If the RX is powered before TX the behavior is as shown above.

mikeller, 2018-02-25:
@Gozzz: Ok, that sounds like a problem with the r-XSR FrSky firmware then. I'll raise an issue with FrSky.

Gozzz, 2018-02-25:
Thanks @mikeller. I know it isn't a huge issue as BF won't arm with throttle high but it is still disconcerting to see that as default along with aux 1 being high.

ttjs82, 2018-02-25:
[image]

I'm seeing exactly same channel values on KakuteF4V2 AIO using the R-XSR FPort protocol and failsafe set to no pulses. Luckily the channel values and safety features on Betaflight still mean that the quad did not arm even with these channel values coming from the receiver if I would by accident plug in the battery before turning on the Tx. However, perhaps with some other channel order and arming switch channel configuration there is the potential of the props spinning when plugging in the battery. So a fix would be nice.

Gozzz, 2018-02-25:
@ttjs82 This was also my concern.

Gozzz, 2018-02-25:
I'm running the FCC 180111 firmware on mine.

ttjs82, 2018-02-26:
Another thing @Gozzz. I seem to be getting the correct telemetry reading while using F.Port even without enabling the Telemetry setting in betaflight. I guess the telemetry data transmission is always on with the F.Port. In addition also the RSSI information seems to be working correctly without using a separate channel to send it from the TX back to the receiver. This seems to be the case at least on the KakuteF4V2 AIO. So this frees up one channel especially if trying to stay with just max 8 channels.

Arakon, 2018-02-26:
I just checked with a normal XSR and get the "normal" centered values with the transmitter off, so it seems to be R-XSR specific.

ttjs82, 2018-02-26:
Thanks for testing out @Arakon. I suppose you were also using the F.Port firmware on the regular XSR? - I also did additional test with R-XSR using the regular firmware (not the F.Port beta). With the regular firmware the throttle channel shows value 885 (set in the failsafe valid pulse range) when powering up the quad with Tx turned off and no pulses failsafe setting enabled during binding. With the F.Port firmware the channel values are in the same situation as shown in the previous screenshots. So the F.Port firmware failsafe settings definitely seem to have something wrong

Arakon, 2018-02-26:
Yep, I used the f.port firmware for the XSR, "No pulse" as failsafe setting.

robertlacroix, 2018-02-27:
Not sure if you guys need another data point, but I can confirm I have the same problem on R-XSR:

[image]

XSR looks better (not sure if AUX4 is intentionally at 1675):

[image]

bluehallu, 2018-02-28:
I have a different but related problem. I'm using international firmware and R-XSR. Default values are all fine, but turning off the Taranis doesn't trigger failsafe in the FC regardless of Failsafe being set to No Pulses or Receiver even though the receiver has the red blinking light. Instead, I just see the correct default values with throttle at 885 and the rest at 1500 but no failsafe.

mikeller, 2018-03-01:
@bluehallu: This seems to be a different problem, occurring with a different version of the firmware. Can you please open a new issue, so we can keep track of the two independently?

Gozzz, 2018-03-01:
@mikeller @bluehallu , FWIW this is the same behavior I get on the FCC 180111 f.port R-XSR firmware if I power down the TX _after_ the RX and TX have been connected to simulate signal loss. Throttle at 885, aer at 1500. I assumed the values would trigger FC failsafe behavior but I'm not certain as I didn't get an Rxlost warning when I tested it, only a drop to the ground. 

I have received a reply from from my initial inquiry from frsky asking for FC firmware version and target. I gave them the info and linked them to this thread. No word back as of yet.

mikeller, 2018-03-01:
@Gozzz: This sounds pretty much like the failsafe gets activated, but not displayed properly. Can you simulate a failsafe, and then go into CLI and do a `status`? Would be interesting to see if `FAILSAFE` appears as an arming disabled reason.

Gozzz, 2018-03-01:
@mikeller status after signal loss/shutting down TX is as follows. 

# status
System Uptime: 36 seconds
Current Time: 2000-12-15T22:55:51.227+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, Vref=3.28V, Core temp=30degC, GYRO=MPU6000
SD card: Startup failed
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 0, config size: 2242, max available config: 16384
CPU:5%, cycle time: 125, GYRO rate: 8000, RX rate: 659, System rate: 10
Arming disable flags: RXLOSS CLI MSP

#

ttjs82, 2018-03-01:
@Gozzz @mikeller I do get the normal failsafe notification on betaflight with my F.port R-XSR & Kakute F4 V2 AIO if I turn off the TX while the quad was in armed state motors running.

# status
System Uptime: 143 seconds
Current Time: 2018-03-01T22:01:05.115+00:00
Voltage: 143 * 0.1V (4S battery - WARNING)
CPU Clock=168MHz, Vref=3.29V, Core temp=40degC, GYRO=ICM20689, ACC=ICM20689, BARO=BMP280
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 1, config size: 2251, max available config: 16384
CPU:9%, cycle time: 127, GYRO rate: 7874, RX rate: 653, System rate: 9
Arming disable flags: FAILSAFE RXLOSS NOPREARM CLI MSP ARMSWITCH

However, the channels still show the strange 1733 & 1306 values if I power up the quad with lipo (to power the R-XSR) without Tx and check the channels in Betaflight. If I connect the quad with just the usb and no lipo (R-XSR doesn't get powered) then the channel values have 885 for throttle and 1500 for others. So the R-XSR injects the strange channel values if it's powered on without the radiolink while using F.Port protocol.

bluehallu, 2018-03-01:
Wait, is motor arming required for failsafe to trigger? My bad, I was trying while disarmed 😅

mikeller, 2018-03-02:
@bluehallu: Yes, failsafe is only triggered when armed - or else you'd get a lot of angry beeping when bench testing.

Gozzz, 2018-03-02:
I just received a beta firmware from frsky overnight (180228_beta). Looks like the issue is fixed, default power on, no signal behavior is the same as @robertlacroix XSR screenshot above. 

Signal loss behavior is the same, throttle drops to 885, aer stick to 1500, and it signals rxloss in cli as disarming flag.

I replied to them with the results. Thanks for all the information and help!

ttjs82, 2018-03-02:
That's great news @Gozzz . Hopefully it's up for download soon as both FCC and EU-LBT versions. I suppose you received the FCC version of the beta firmware?

robertlacroix, 2018-03-03:
@Gozzz great news! Could you upload the beta somewhere?

mikeller, 2018-03-03:
Thanks for testing, @Gozzz.

@robertlacroix: Here you go. Please report back on how the testing went.

[R-XSR_FPORT_LBT180228_Beta.zip](https://github.com/betaflight/betaflight/files/1776920/R-XSR_FPORT_LBT180228_Beta.zip)
[R-XSR_FPORT_FCC180228_Beta.zip](https://github.com/betaflight/betaflight/files/1776921/R-XSR_FPORT_FCC180228_Beta.zip)

ttjs82, 2018-03-03:
I tested out the LBT version and everything seems to be fine with it also.

mikeller, 2018-03-03:
Thanks for testing, @ttjs82. I'll let FrSky know, so they can release the new firmware on their website.

mikeller, 2018-03-03:
@Gozzz: Can this be closed?

ttjs82, 2018-03-03:
@mikeller I used the serialpassthrough to update the receiver as changing the connector to update directly with QX7 was too difficult. The serialpassthrough worked well.

parnz, 2018-03-03:
@ttjs82 Any direction on this method?

mikeller, 2018-03-03:
@parnz: See https://github.com/betaflight/betaflight/issues/3364#issuecomment-312121695.

ttjs82, 2018-03-03:
Certainly @parnz 

To start of devices I used was Kakute F4 V2 AIO and the R-XSR receiver. The flight controller doesn't supply power to the receiver while the controller is powered through usb which helps a lot in the update process. Download the FrSky update tool from their website https://www.frsky-rc.com/usb-to-s-port/ 

I had the Rx connected to Uart1 Tx pad initially and the port setting tab on betaflight having the serial rx enabled for the F.Port protocol. What I had to do was disabling the serial rx option and choose smartport from the telemetry output option. The quad was powered through lipo in this phase.

[image]

After saving this option I reconnected to Betaflight and then went to the CLI tab. There I gave the command serialpassthrough 0 57600. Where is the number you get from the UART number you have the receiver connected minus 1. So in my case Uart 1 becomes 0. After this command I unplugged the lipo to power down the receiver (but leave the usb connected). Next I opened the update tool. Make sure the com port setting shows the same com port you had when connecting betaflight.  In the update utility choose the update file you want to update to the receiver. After choosing the file the tool starts to search for the receiver. Now I powered the quad with lipo to turn on the receiver. Then pressing the download button starts the process of updating. If everything goes fine it shows a success message. Then I closed the tool and powered down the quad. Finally go back to betaflight and change the serial rx back on and disable the smartport telemetry option. Also make sure the configuration tabs also has the F.Port chosen. Take notice that disabling the serialrx option also empties all the modes tab options so you have to put back the arm switch settings and such.

This process worked for me but the problem can be power cycling the receiver if the flight controller feeds power to it when using usb connection and removing the connection cable to receiver is difficult. The instructions on this page https://github.com/betaflight/betaflight/wiki/Serial-Pass-Through was missing the thing about putting the uart port number to the seriallpassthrough command. Also enabling the smartport setting is something I did differently but I thing the github instruction page doesn't mention anything about that because I think it's written from the point of view where there is SBUS and Smartport connections in use initially and we start from the situation of F.Port.

Good luck with the update.

Gozzz, 2018-03-03:
@mikeller yeah looks good to me. Thanks again.

closed 2018-03-03 by Gozzz

parnz, 2018-03-04:
@ttjs82 Thanks for a very details instruction 👍 
But my fc is Matek F405 mini and it power up the receiver when connected to PC.
Look like it more difficult than update with my X7S :/

brucesdad13, 2018-03-10:
@mikeller the beta firmware fixes the default output values for me as well. Tested the FCC version plugging into configurator before tx has been turned on all values look normal now. 👍  Any idea when FrSky will update the FPort firmware on their downloads page?

PsyDafke, 2018-03-11:
@brucesdad13, seems FrSky have put the latest F.Port FW in the regular section, instead of the separate F.Port beta section of the downloads page.

Gozzz, 2018-03-12:
@PsyDafke You need to hook fport from receiver to uart 3 or 1 TX pad on BFF3. I'm using uart 1 without the inversion hack on the RX with the following CLI on my BFF3-

serial 0 64 115200 57600 0 115200

set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON

set tlm_halfduplex = OFF

PsyDafke, 2018-03-13:
Thanks @Gozzz,
I'm aware that the receiver needs to be connected on the TX (this is not my 1st build with F.PORT).
The weird thing is, that I get Telemetry data to my Taranis.
I checked my settings again (currently the Receiver is on UART3 TX, but also tried UART1 & 2 previously).
UART2 is also available on the board, btw (there is some confusion/error with the silkscreen print though).
Only difference I noticed was the tlm_halfduplex was ON in my config, so I changed that parameter hoping it would be the solution... sadly it wasn't.

PsyDafke, 2018-03-17:
UPDATE!!!
Apparently, it was a user error on my end. Seemed that I accidentally held the bind button on the R-XSR too long (same function to change between SBUS and CPPM output)! This still seems to 'work' with the FPORT firmware!
Now have both control and telemetry working just fine.
Hope this helps out others hunting down F.Port issues.
Now, let's go out and maiden this baby! 👍

odixxx, 2018-03-28:
after updating from 180111 to 180228
problems with telemetry lua
sensors all work
video problems https://youtu.be/4Al4QsMfxQQ
ps. at times it normally works

Gozzz, 2018-03-29:
@odixxx I get a delay sometimes but it usually shows up. How long do have you waited?

odixxx, 2018-03-29:
long tested - it sometimes appears
as it is incomprehensible happens randomly
today it's back 180111

CharlosFPV, 2018-07-15:
I just updated the firmware of the r xsr. I keep loosing telemetry and my rssi drops like crazy at a close distance. I’m using an lumenier f4 aio. F port with the x-lite. Any suggestions?