betaflight/betaflight #5316: BF3.3 NAZE target gyro initialization failure

etracer65, 2018-02-25:
Recording this while recognizing the F1 targets are no longer officially supported.

From at least BF3.3RC1 gyro initialization fails for the NAZE target.  Not sure if this is a generic F1 problem or specific to target.  I was testing with a Naze32 rev 5 with a MPU6050 (I2C) gyro.

BF3.2.5 works without problems.

I'll try to figure out when the problem started.

etracer65, 2018-02-25:
Dug out an old Naze32 rev 6 with a MPU6500 (I2C) and it also fails to initialize the gyro.  So at least eliminates a problem with a specific gyro.  Still could be only related to the NAZE target or affecting all F1's.  I don't have any other F1's to test.

The accelerometer works properly but the gyro stays in calibration phase indefinitely and the sensors tab show no output for the gyro.

mikeller, 2018-02-25:
Just making a guess here, but could this be related to the fact that the NAZE's gyro is I2C? I think the original SPRACINGF3 was I2C as well, so one of these could be used to verify.

etracer65, 2018-02-25:
Tested a SPRACINGF3 target with I2C MPU6050 and it worked fine.  So it's still either a generic F1 issue or specific to the NAZE target.

etracer65, 2018-02-26:
I determined that it's https://github.com/betaflight/betaflight/pull/4986 that caused the problem - not sure why yet.

mikeller, 2018-02-26:
Could the switch to `float` be pushing the F1 over the edge, or causing optimisation problems? Have you tried with slower loop times?

etracer65, 2018-02-26:
Found it.  ` src/main/sensors/sensors.h`

```
@@ -40,7 +40,7 @@ typedef union flightDynamicsTrims_u {
     flightDynamicsTrims_def_t values;
 } flightDynamicsTrims_t;
 
-#define CALIBRATING_GYRO_CYCLES             1000
+#define CALIBRATING_GYRO_CYCLES             10000
 #define CALIBRATING_ACC_CYCLES              400
 #define CALIBRATING_BARO_CYCLES             200 // 10 seconds init_delay + 200 * 25 ms = 15 seconds before ground pressure settles
 ```

Changing the calibration cycles back to 1000 fixes the problem.  I'm guessing that F1 is taking too long to do 10000 cycles.

Also explains why I noticed gyro calibration taking a little longer on F4's as well.

@adrianmiriuta Was the change to 10000 gyro calibration cycles intentional or maybe leftover from debugging?

mikeller, 2018-02-26:
@etracer65:

> Was the change to 10000 gyro calibration cycles intentional or maybe leftover from debugging?

I think that was the outcome of a discussion with @ledvinap that brought the insight that 1000 cycles was not long enough.

mikeller, 2018-02-26:
Also, for a fix, add a conditional define for F1, with an comment as to why?

etracer65, 2018-02-26:
@mikeller The extra floating point is definitely hurting the poor little F1.  It's really just a combination of the extra floating point time the extra calibration cycles that's just too much.  The F1 seems to be limited to about 1K/1K now to have stable loop times.

Experimentally it looks like 1500 cycles is the upper limit for calibration completing.

> Also, for a fix, add a conditional define for F1, with an comment as to why?

I was going to suggest exactly this.  Just go back to 1000 for the F1 and not waste any more time working on it.  I'll put together a pull request.

mikeller, 2018-02-26:
@adrianmiriuta: Good point. Do you want to look into this for 3.4? I think in the meantime we should still push the change proposed above, to make the F1 targets useable again.

adrianmiriuta, 2018-02-26:
@mikeller 
Ok

atomiclama, 2018-03-06:
Built from master. 

Tested and now working on a CC3D

@adrianmiriuta Thanks

etracer65, 2018-03-06:
Fixed by https://github.com/betaflight/betaflight/pull/5320

closed 2018-03-06 by etracer65