betaflight/betaflight #5317: Proposal: Support for blind / A/B testing with profiles

mikeller, 2018-02-25:
Considering the recent discussions (see #5231) about user demand for a way to do unbiased testing of different (filtering) options, I propose the following:

Add a boolean parameter `blind_test_profiles`. If this parameter is set to `true`, the flight controller will use a random value (LSB of gyro noise?) to randomly select between profile 1 and 2, and load the selected profile. (To make this work for the filtering case at hand, filter settings would have to be moved into profiles as well.)

This will enable the user to set up profiles 1 and 2 to be two different but similar setups they want to compare the flight performance for in an unbiased way. They can then perform a number of test flights with black box enabled, and note down their rating / observations for every flight. After the flights, they can download the logs, and determine from the header of every log which profile has been used for each flight. They can then group their findings by setup, allowing them to determine which one they like better (without perceptual bias).

ctzsnooze, 2018-02-25:
awesome idea

etracer65, 2018-02-25:
I like the idea, assuming we won't be adding too much stuff to the profiles that we run into space issues again.  Probably a very limited user group though...  @joshuabardwell would probably make good use of it.

I could see myself using this for PID tuning as finding the final 10% is very subjective.

etracer65, 2018-02-25:
What would be even better for filter testing would being able to switch "filter profiles" while in flight.  But I realize that's probably not practical.

mikeller, 2018-02-26:
Moving settings into profiles should not significantly increase the firmware size, and only slightly increase the configuration size.

mikeller, 2018-02-27:
@DanNixon: Certainly worth consideration. But on the other hand, if users use this, and then check the selected profile before flying, they are just cheating themselves out of good data. They can just disable the OSD element (most won't be using it anyway), and not look into the CMS before flying. At least having the data retrievable in CMS would be helpful, in cases where users want to test with setups that cannot use logging, or want to do quick field tests.

hydra, 2018-02-28:
you could add the profile to the post-flight screen.

ProfessorStoke, 2018-06-20:
I like this idea. This would help not only individual pilots in determining whether a particular setting is working for them, but would also give developers another tool to determine if a particular feature is worth the effort to include or maintain.

I'm picturing a procedure activated through the OSD where the first arm is profile A, the second arm is profile B, and the third arm is profile X. After the last disarm the OSD asks for your guess, and the process repeats until you have enough data or you're bored. You then exit the procedure to see your results and whether or not the difference between A and B was statistically significant. If you have a three way switch you could also switch between A, B, and X in flight and use the OSD to guess what X is on disarm.

limonspb, 2021-03-09:
> you could add the profile to the post-flight screen.

about to comment that. That would be nice.
In this case, the feature will be useful even on FCs without a blackbox.

closed 2026-09-26 by blckmn