betaflight/betaflight #5319: Bardwell CLR4 RX Arm Issues

brybell, 2018-02-26:
1. OSD doesn't work, it only shows the battery average voltage in the middle of my screen. I have everything setup correctly and it should be working. Using JBardwell AIO FC.

When plugged into BF, it is saying an RXLOSS for arming disable, which is not possible since it receives a signal from my TX on the Receiver tab, and I am able to arm and fly when not plugged into the BF. 

OSD is not working correctly. Using Spedix 30am 32bit ESC, Bardwell FC, EMAX motors. Flew fine yesterday on 3.2.

`# version
# Betaflight / CLRACINGF4 (CLR4) 3.3.0 Feb 18 2018 / 13:34:25 (f07bf40d1) MSP API: 1.37

# reset configuration to default settings
defaults nosave

# name
name Frog

# resources

# mixer

# servo

# servo mix

# feature
feature -OSD
feature MOTOR_STOP
feature TELEMETRY
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 13 1 1700 2100 0
aux 2 28 3 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set min_check = 1015
set rssi_channel = 9
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 2
set pid_process_denom = 1
set osd_vbat_pos = 2282
set osd_rssi_pos = 2282
set osd_tim_1_pos = 2282
set osd_flymode_pos = 2282
set osd_throttle_pos = 2282
set osd_craft_name_pos = 2282
set osd_avg_cell_voltage_pos = 2282

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0`

brybell, 2018-02-26:
Only average cell voltage displays, and it is in the middle left of the screen, oddly placed. The only other thing that shows up are warnings. No RSSI, Craft Name etc. Tried enabling from BF as well as the OSD menu, which works.

jflyper, 2018-02-26:
```
set osd_vbat_pos = 2282
set osd_rssi_pos = 2282
set osd_tim_1_pos = 2282
set osd_flymode_pos = 2282
set osd_throttle_pos = 2282
set osd_craft_name_pos = 2282
set osd_avg_cell_voltage_pos = 2282
```

They are set to display on the exact same position on the screen; no wonder you can't see them separately.

Please issue CLI `defaults` and start configuring from scratch.

closed 2018-02-26 by brybell