betaflight/betaflight #5321: BF 3.3.0 RC2 Omnibus F7 v2 does not initialize ICM gyro when 32K mode is switched on

dovcharov, 2018-02-26:
1. I'm running Betaflight 3.3.0 RC2 on Omnibus F7 V2 FC (original, from Airbot). I started to get following issues:
a) I am running ICM gyroscope (set gyro-to-use = 0)
b) I turn on 32K mode
c) I set 16K/16K gyro-PID loops
d) Other settings are: Multishot as ESC protocol (32K or 16K PWM speed), acc disabled, baro disabled, FPort receiver protocol, Airmode on, Dynamic filter on, Telemetry on, OSD on, gyro notch 1 and 2 filters disabled, Gyro LPF filter - PT1.

Once i save config, disconnect FC from PC and plug in the battery, the quad does not arm displaying CALIB error message on OSD. If I reconnect quad to PC and go to Sensors tab in betaflight configurator, I see that gyro doesn't give out anything. 

2. If I connect FC first via USB, and then plug in the battery, the gyro works fine.
3. If I disable 32K mode and switch back to 8K/8K, the ICM gyro works fine - I plug in the battery and quad arms and flies well.

DieHertz, 2018-02-26:
Try increasing `moron_threshold` in CLI.
Does it happen with RC1?

mikeller, 2018-02-26:
@dovcharov: I think you are falling victim to too much gyro noise:
- gyros show a lot more noise when run at 32 kHz;
- for some boards, insufficiently buffered voltage converters seem to exacerbate gyro noise when the board is powered from battery.

Possible solutions:
- increase `moron_threshold`;
- if that does not help, switch back to 8 kHz gyro.

dovcharov, 2018-02-26:
Thank you for your comments. I will try in the evening and update you. On RC1 I had moron_threshold increased, and was able to fly with FKF.

DieHertz, 2018-02-26:
Have you crashed or dropped your quad? These gyros deteriorate after crashes and not even moron threshold helps in this case

dovcharov, 2018-02-26:
Definitely no. I didn't have any major crashes lately.

jflyper, 2018-02-26:
Can you paste output from CLI `diff` for the configuration that doesn't initialize the gyro?

dovcharov, 2018-02-26:
Here's my diff. With increased moron_threshold, gyro may occasionally calibrate, but this happens rarely and I get crasy motor jerks. 
`
# name
name AbooQuad

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# map

# serial
serial 1 0 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 27 1 1700 2100 0
aux 2 13 1 1300 2100 0
aux 3 26 5 1700 2100 0
aux 4 35 3 1700 2100 0
aux 5 36 4 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set moron_threshold = 150
set gyro_use_32khz = ON
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 128
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 32000
set ibata_scale = 179
set yaw_motors_reversed = ON
set small_angle = 90
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_warnings = 63
set osd_vbat_pos = 2498
set osd_rssi_pos = 2553
set osd_tim_1_pos = 2466
set osd_tim_2_pos = 33
set osd_flymode_pos = 2531
set osd_throttle_pos = 2509
set osd_vtx_channel_pos = 377
set osd_crosshairs = 2248
set osd_ah_sbar = 200
set osd_ah_pos = 2248
set osd_current_pos = 2486
set osd_mah_drawn_pos = 2520
set osd_craft_name_pos = 2539
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2091
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2082
set osd_esc_rpm_pos = 2102
set osd_stat_max_spd = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set debug_mode = NOTCH
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100

# rateprofile
rateprofile 0

set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 105
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75
`

DieHertz, 2018-02-26:
I think it may be related to the recent bug with gyro calibration `uint16_t` overflow that we've found in #5320.
The number of cycles expected for gyro calibration in current code is  `(10000 / 15.625 * 10000) % 65536 = 43008`. That's a lot and may actually never be achieved.
@etracer65 @mikeller your opinion? Does it sound reasonable?

jflyper, 2018-02-26:
@dovcharov My F7V2 loaded with your diff does finish calibration immediately, so the problem is hardware oriented.

And like @adrianmiriuta say, additional bypass caps on acc/gyro fix it in some case. You may want to contact Airbot and ask if you are eligible for a free gen2 ICM gyro box. I'm not sure if it will be offered for free, because
- Your symptom developed overtime
- F7V2 has a MPU6000 in the gyro box, and ICM onboard.
Worth asking anyways, though.

Since this is not a firmware problem, I will be closing this.

jflyper, 2018-02-26:
What? We have a problem on calibration?

adrianmiriuta, 2018-02-26:
@jflyper 
@DieHertz 

>2. If I connect FC first via USB, and then plug in the battery, the gyro works fine.

It is power supply induced EMI gyro noise for sure !

dovcharov, 2018-02-26:
So, is my solution - capacitor on 5v rail, or is it a software-related issue?

adrianmiriuta, 2018-02-26:
@dovcharov 
It is a hardware related issue !

jflyper, 2018-02-26:
> So, is my solution - capacitor on 5v rail, or is it a software-related issue?

Cap on 3V3 rail, as close as possible to the onboard gyro chip may help.

jflyper, 2018-02-26:
I was wrong about the gyro placement. The ICM is gyrobox-based. Ask Airbot for the gen2 (capped) ICM gyrobox.

DieHertz, 2018-02-26:
I think it's a combination of two issues.
One of them - asking for too many samples due to bug discovered in #5320, another - the amount of electrical noise, which makes collecting so many samples with acceptable variance nearly impossible.

adrianmiriuta, 2018-02-26:
@DieHertz 
I had three defective omnibus F4 corner and one omnibus F7  with the ICM20608 gyro !  
The problem is due to capacitor tolerance , EMI interference, bad board design  !
read the post !
[https://www.rcgroups.com/forums/showpost.php?p=38947338&postcount=511](https://www.rcgroups.com/forums/showpost.php?p=38947338&postcount=511)

They do not work with bf versions before #5320 either !

DieHertz, 2018-02-26:
#5320 fixes the `uint16_t` overflow, not introduces the bug. I'm not disputing that ICM-20608-G has noise issues unless properly filtered, read carefully, please.

adrianmiriuta, 2018-02-26:
@DieHertz 
sorry I expressed myself wrong !

the overflow bug was introduced with #4986 
and  these boards do not work with bf versions prior  to this  also.

dovcharov, 2018-02-26:
With 3S battery, my ICM gyro does not initialize at 16K/16K as well.

adrianmiriuta, 2018-02-26:
@dovcharov 
thx.
can you boot up from USB power
plug in lipo 4s
make a screenshot from the gyro trace in the motors tab
like this one 
[http://www.silverbug.org/26.11.2017_corner_issue/Screenshot%20from%202017-11-09%2012-52-55.png](http://www.silverbug.org/26.11.2017_corner_issue/Screenshot%20from%202017-11-09%2012-52-55.png)

and post it somewhere 
pls.

dovcharov, 2018-02-27:
Done. 
https://yadi.sk/i/K_GDpDBB3SoiJo - this is when my drone lies still on the bench
https://yadi.sk/i/5tnptdIM3SoiJh - this is when I take a drone with a hand and swing it slightly

adrianmiriuta, 2018-02-27:
@dovcharov 
you have some static gyro noise,
but its's far away from what I experienced with the defective gyros.
I think @DieHertz is right and it is some combination of hardware software issue.
Try a combination cap on 5V rail and increased moron_threshold.
This is a new Issue , I think it will take some time to figure it out .

dovcharov, 2018-02-27:
Btw, i checked the static noise on MPU6000 gyro on the same flight controller. It's relatively the same as static noise on ICM gyro in 16/16 mode

adrianmiriuta, 2018-02-27:
@dovcharov 
you have to plug in the Lipo ! 
else you do not have EMI noise from the 5V stepdown regulator.
your last gyro traces are flawless.

dovcharov, 2018-02-27:
You are correct. I will re-make the measurements in the evening.

dovcharov, 2018-03-02:
Done. Updated static noise graphs with 4S battery plugged in.
MPU6000, 8K/8K - https://yadi.sk/i/n_D9PjkZ3Sornc
ICM20608, 8K/8K - https://yadi.sk/i/FiMNOXCQ3Sw3YX
ICM20608, 16K/16K - https://yadi.sk/i/nGofqSOv3Sw3Yn

And Airbot sent me a replacement gyro box for free.

DieHertz, 2018-03-06:
@dovcharov if you could record a log with `debug_mode = gyro` with default `moron_threshold` (the one which doesn't allow calibration to pass) - I'd really appreciate it. I have some ideas how to get rid of this behaviour but need data.

nitro20, 2018-03-19:
dont laugh but i run 3s..and i have this problem on three out four flipf4 mpu icm20608 gyro FCs...heres what my BBL looked like....my gyro looks like that all the time..FKF dynamic, with notches off and D notch on..i have emax30 amp bullet esc. on the one that worked the escs are XS20A DYS bhheliS with Dshot 600 and 16k/16k, but it flew GREAT, no problems...

this is 1/8 throttle, idles ok...but i dont trust it....
[image]

nitro20, 2018-03-19:
i have a post in beginers questions bout this at intofpv with pics of the of the quads fc and esc..and i tryed a flipf4 FC just plain jane on the bench..No go..

andrejpodzimek, 2018-04-06:
I have the same (pseudo-)problem. With `gyro_to_use = 1`, only up to 8k/8k modes are possible, which is (sort of) expected. With `gyro_to_use = 0` (which indeed causes `ICM20608G` to be shown in `status`) activating the 32k mode yields an arming failure. I find this surprising, because

1. it doesn't matter whether I actually set the frequency above 8 kHz; just selecting the 32 kHz support triggers the failure, even with 8k/8k settings.
2. the gyro appears to work fine, as far as the Betaflight configurator is concerned. The drone 3D model moves as it should etc. It only fails to calibrate (no 3 beeps) and then fails to arm (with `CALIB` among the reasons).

This is quite a minor issue, as there's no strong technical reason for 32 kHz and no evidence that a human pilot could ever tell the difference between 8 kHz and 32 kHz in a rigorous blinded experiment. However, I'll be more than happy to test any updates on this on my `OMNIBUSF7V2` build.

etracer65, 2018-04-06:
@andrejpodzimek You likely just need to increase `moron_threshold` when using the ICM gyro at 32KHz.  In that setting there is a lot more noise and it can prevent gyro calibration (calibration thinks you're moving the quad).

andrejpodzimek, 2018-04-06:
Yes, that's it! Thanks for the hint. I've just tested a few `moron_threshold` values with 32 kHz support enabled:

- 48: never arms
- 150: never arms
- 175: almost always arms
- 200: always arms

This^^^ was at 8k/8k frequencies.

Sadly, switching from 8k/8k to 16k/8k increases the CPU load from 22% to 60%, which looks somewhat unsafe to fly. So it looks like I can't get above 8k with this configuration anyway. (It has a barometer, a compass, a GPS and 30 programmable LEDs.)

thiagohmoreira, 2018-04-08:
I have a "similar" issue, with loads of noise on the gyro when VBAT is connected. On my case it don't seems to matter if 32k is enabled, or what is the sampling rate. If I don't connect the VBAT on the board, use only the 5V from ESCs rail, it looks perfect.

The moment I connect anything to VBAT, this crazy noise comes into play. The higher the voltage, the bigger the noise. I thing this might be a problematic board or board layout.. the way I see, it doesn't seems to have anything to do with software.

And as a separate comment, this issue seems to be related to issue #4772 also.

flash-v, 2018-04-18:
Have a same issue. 
I used a 220 uF 6.3 V electrolytic capacitor on a 5 V rail and left the default moron_threshold=48.

DieHertz, 2018-04-18:
The 3V rail should be filtered as close to gyro as possible, 5V likely won't help

flash-v, 2018-04-18:
@DieHertz   Hmm.  In my case, the capacitor on the 5V rail helped

DieHertz, 2018-04-18:
I'm happy it helped you, it won't work for everyone though. The ideal solution would be for manufacturers to filter gyro power properly, as Airbot does now

flash-v, 2018-04-20:
@DieHertz Yes, you were right, I had to add a capacitor to the gyro's power circuit. 
Thanks for optimization F7, it works just fine in 32/32/dshot1200 mode.

DieHertz, 2018-04-20:
@flash-v Glad to hear the success stories :+1:
There are some things to iron out, but pretty much usable as is :-)

thiagohmoreira, 2018-04-20:
@flash-v Would you mind to post a picture of what/how you did it? I'm not sure I have the needed skills to do the same, but I really interested in get this board working at this point :)

flash-v, 2018-04-21:
**@thiagohmoreira**  I'm add a tantalum cup of 220 uF (№ 2,3). 

[image]

Perhaps this is not the best place for soldering capacitors, but it works. (№4 - U for 3.3 V for gyro). 

Video of flight https://youtu.be/hd37aOqAMHw  (there was a wind of about 9-10 m/s and copter not fine tune) 

and log of this flight. [LOG00010.BFL.zip](https://github.com/betaflight/betaflight/files/1934098/LOG00010.BFL.zip)

thiagohmoreira, 2018-04-21:
@flash-v thank you very much for the picture and the diagram!

Unfortunately to me it didn't solved the issue. I didn't had any tantalum cap at had, so I decided to give it a shot with the ceramic 100nF I had and it didn't changed the gyro noise (at least not that I can tell by looking on the sensors tab). Should it also work? Or I must use a tantalum one?

DieHertz, 2018-04-21:
According to Airbot it must be tantalum

closed 2018-05-28 by stale[bot]