betaflight/betaflight #5324: [Feature request] - Auto-update fonts when flashing

MR444, 2018-02-26:
Will it be possible to auto-update the font when flashing? I had two boards where the Main Batt Usage looked very weired and it took some time to find out why and how to fix that. I found a solution here:
https://github.com/betaflight/betaflight/issues/3753

mikeller, 2018-02-27:
@MR444: This is currently not possible, since there is no simple way to detect what font is currently installed. This means we'd have to assume it needs updating for every firmware update, potentially overwriting a custom font that the user has installed. There is some discussion in #3461 about adding metadata to the font that is installed, making it possible to detect the name / type, but it seems to be pretty much stalled. This would be a prerequisite for what you are proposing.

MR444, 2018-03-01:
Thanks, metadata would be great! I take a look at discussion #3461!

closed 2018-03-01 by MR444