betaflight/betaflight #5329: SmartAudio Lite support (Unify Pro Nano)

robertlacroix, 2018-02-27:
I can't get SmartAudio to work on a TBS Unify Pro Nano. According to the [manual](https://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf) it's using `SmartAudio Lite`:

> (requires 3.3V level and soft- or hardware pull low)

DieHertz, 2018-02-27:
Will have a look at the manual tonight unless someone else does. I'm also waiting for my parcel with one of them soon.

DieHertz, 2018-02-27:
`io/vtx_smartaudio.c`:
```
        portOptions_e portOptions = SERIAL_STOPBITS_2 | SERIAL_BIDIR_NOPULL;
#if defined(USE_VTX_COMMON)
        portOptions = portOptions | (vtxConfig()->halfDuplex ? SERIAL_BIDIR | SERIAL_BIDIR_PP : SERIAL_UNIDIR);
#else
        portOptions = SERIAL_BIDIR;
#endif
```
There's indeed no pulldown defined, will have to make that configurable I assume.

DieHertz, 2018-02-27:
Tell me your target and I'll give you a hex off the latest RC to check.

robertlacroix, 2018-02-27:
OBSD, but you can also open a PR. I can compile ;-)

DieHertz, 2018-02-27:
I have to decide on how to solve it the best way, thus opening a PR is a bit preliminary.
I'll give you a commit to cherry-pick onto latest RC if you like.

DieHertz, 2018-02-27:
https://github.com/diehertz/betaflight/tree/unify-nano-smartaudio-lite-support

RipperDrone, 2018-02-27:
got the TBS Nano as well, will try to compile from this branch and test - need to find time to do some soldering, though... target = OBF4

RipperDrone, 2018-02-27:
[image]

robertlacroix, 2018-02-27:
@DieHertz boom, works. Thanks a lot for the quick turnaround! I gotta catch some sleep now and didn't check if the regular SmartAudio still works btw. Can do in the morning.

DieHertz, 2018-02-27:
Thanks for your feedback. I need to muse about it for a while to resolve it in a generic manner

jflyper, 2018-02-27:
>SmartAudio LITE
>SmartAudio LITE is used on the tiny Unify PRO nano. The software protocol is the exact same as on Smartaudio V2.
 > However there are two limitations:
> - It requires 3.3V level (Standard Smartaudio accept 0.9V audio level).
> - It requires software pull low

It's a fresh news that the Standard SmartAudio accepts 0.9V audio level, whatever that means, and the phrase "software pull low" is equally cryptic. Do LITE retains audio compatibility with the line pulled low? Pull low always??? How hard can the pull low would be?

ledvinap, 2018-02-27:
@DieHertz : 
It may be best to use 2 bits to represent 4 pull states:
0 - default (pullup nor normal, pulldown for inverted)
1 - nopull
2 - pullup
3 - pulldown

On init, 0-default may be mapped to 2/3 according to port inversion (and OD)

DieHertz, 2018-02-27:
Thank you Petr, I did the same when I added unusual parity settings for SBUS which also requires more than 2 values, but it got dropped in favor of a fully-configurable serial line.
I'll probably still use this enum and 2 bits or a real bitfield to represent everything necessary for serial line parameters in BF 3.4, suggestions are welcome

ledvinap, 2018-02-27:
@DieHertz : `enum` is fine, just don't stick to one-bit-per-function paradigm ..

DieHertz, 2018-02-27:
I'll do something similar to the trick done in this PR:
https://github.com/betaflight/betaflight/pull/4371

codecae, 2018-02-27:
A hardware pull down might be a viable alternative

Gozzz, 2018-02-27:
I'm running a TBS Unify Pro Nano on a CLR4 target (Bardwell F4) 3.3 RC3 build IIRC, SA works but has a delay of around 3 seconds when setting bands/channels through Taranis x9d+ lua script. I'm just mentioning this for additional information.

dienstagfpv, 2018-02-27:
...so any chance to tweak it somehow to get this running via softserial on 3.2.5 or 3.3 rc2?

mikeller, 2018-02-28:
@dienstagfpv: I think adding an external hardware pull down resistor to the VTX cable will be the best short term solution. Unfortunately TBS did not bother to make Betaflight aware of this change in due time for us to add support for this to the upcoming 3.3. release. :-(

dienstagfpv, 2018-02-28:
@mikeller adding an external hardware pull down wont quite work with softserial i guess ;) what a pity!

DieHertz, 2018-02-28:
Why? It should work okay.

codecae, 2018-02-28:
Softserial mimics a hardware uart.  Should work the same.

jdd1777, 2018-03-01:
Just wanted to mention it works fine for me on 3.3 on lumenier micro lux v2 softserial with nano unify.

RipperDrone, 2018-03-01:
tried the Nano SmartAudio Light on UART3, works like a charm. However, power settings offered over Lua are 25/200/500/800mW whereas hardware only supports 25+50mW. Do I need a special .lua version?

RipperDrone, 2018-03-01:
has been answered by @DieHertz on #slack: It's an enum array, so setting it to nominal 200 will be 50mW, effectively.

GhostMaster9, 2018-03-03:
Does not work with dys f4 pro mini and tbs unify nano and final betaflight release. :(

tbs-trappy, 2018-03-03:
ok, first of all my apologies for not communicating this properly. I have just had some serious discussions internally why I have not been made aware or informed of all of this stuff, and I apologize for any confusion that has caused.

the standard in itself didn't break, it seems the nano is just more susceptible for not receiving a logic zero at the start. the manual entry "requires software pull low" was an extremely shitty explanation. We're updating the manual as I'm writing this. 

The pull low has always been part of the SA specification (each command starts with zero). The nano is just more stringent for this to be present as it appears at this point in time. any fix for this issue should not break compatibility with any existing transmitters and the SA standard, and it will not require a special parameter as far as we know at this point in time. The Unify Pro Nano adheres to the same SA standard as all other Unify Pro, but may just be more picky with non-compliance. As we have seen from the AKK-bug previously, the Unify Pro is very tolerant towards SA-incompatibility.

I'm not an engineer and this may sound stupid to smart people. I'm here to help anyone that needs hardware to test this and TBS internally will also be purchasing the hardware that is causing issues and doing our own testing and will report back early next week. There is no excuse for this shitty communication from us and I have put measures in place that this will not happen again. If you are a betaflight developer and you don't already have a Unify Pro Nano, please let me know. Our office is still open for another hour so I can ship out free Unify Nano's so they arrive to your doorstep monday morning. trappy - at - gmail - dot - com

DieHertz, 2018-03-03:
@tbs-trappy Thank you Raphael, I have ordered one right at the day of Crossfire Nano release, hope it'll arrive soon. Configurable serial line should help with support for Unify Nano.

tbs-trappy, 2018-03-03:
OK, keep us posted DieHertz, thanks for the update. I packed a couple of Nano's and had them expressed out, hoping for monday arrival. I just chased down UPS and hoping they will pick up the packages.

Jugsy, 2018-03-03:
Put together a new build with Jbf4 and unify nano, smartaudio worked on initial testing over a 24h period including hover test. For some reason has stopped working since.

sirPerna, 2018-03-03:
@DieHertz when we added smartaudio to the nano vtx we tested it with betaflight on colibri race target and directly off crossfire receivers. As we didn't change the protocol and it was working flawless on our testing we didn't see a reason to inform betaflight. The weird part is it work with some targets but doesn't with others. 
It's fixable for all targets by using a pull down which makes us believe the audio wire is not properly idle when the command is sent. 
Before we add a fix we should try to figure out what exactly happens on the not working targets. Adding a proper preamble may fix the issue too. 
If someone can hook up a scope and paste the wave here we probably see it. In the meantime we order some boards so we can test it ourself as well.

jflyper, 2018-03-03:
@sirPerna 
> Adding a proper preamble may fix the issue too.

For this method to be deployed, some one has to carry out regression test against all existing working cases of SmartAudio VTXs and hard/soft serial ports of flight controllers.

This is how standard works.

sirPerna, 2018-03-03:
@jflyper yeah sure. I think we should try to find the root source of this issue as a first step. 
It works directly off crossfire receivers without firmware modification or pull down. So does it with some betaflight targets. 

It would be awesome to get some help on testing from betaflight side. Otherwise just give us some more time.

DieHertz, 2018-03-03:
To do so we need to have hardware samples. Mine is currently "being imported" somewhere in Moscow, after 2 weeks of being in transit. It'll probably arrive in a week or so.

spuder, 2018-03-04:
Unify Pro Nano works with OMNIBUSF4SD
I find it often takes a few seconds for the settings to change, and smart audio doesn't always register on the first attempt.

jflyper, 2018-03-04:
Reminder to my self: One thing that comes to my mind is the handling of `SERIAL_BIDIR_NOPULL` option.

jflyper, 2018-03-04:
@sirPerna
One big problem we have is that devs have not been given an physical and datalink layer interface specifications. All we knew when we did 2.0 was

[image]

We need to know more. How vtx end is terminated at each phase of the half-duplex protocol, how a line should be kept during idle period, etc.

brucesdad13, 2018-03-04:
For whatever it's worth, SmartAudio is working well on my AlienWhoop V2 F4 with TBS Unify Pro Nano VTX connected to the TBS Tiny Camera. The TBS Tiny Camera has SmartAudio Input from the fc and then an Output to the Nano vtx and I am using it in this configuration. I'm changing VTX settings via Taranis X7 Lua scripts and FrSky R-XSR uninverted.

robertlacroix, 2018-03-05:
@spuder interesting! My problems were on a Hobbywing xrotor F4 (`OMNIBUSF4SD`). Betaflight 3.2.5 or 3.3.0?

spuder, 2018-03-05:
I've only tested 3.3.0 on the xrotor

joshmlewis, 2018-03-06:
I have two confirmed cases of smart audio not working with the TBS Nano on Hobbywing Xrotor F4 with 3.3. It uses the OMNIBUSF4SD target. Happy to test or provide feedback.

robertlacroix, 2018-03-06:
@joshewis yep same here. fwiw I’m using UART6 for SmartAudio

spuder, 2018-03-06:
I may have been wrong. Today I find I can't get smart audio working on 3.3.0 with the OMNIBUSF4SD at all.

I tried compiling this commit that @DieHertz authored, and smart audio is now working perfectly. 

    https://github.com/DieHertz/betaflight/commit/0f68790506ef8f29c25802c76f793e9a39793e1e

bkeinert, 2018-03-07:
Can confirm the issue on an OMNIBUSF4SD based FC (some 16x16 F4), soft-serial TX via LED_STRIP. Worked sporadically until I shortened the cables for final assembly. I can try both - a hardware pull down (10k?) and the suggested software pull down.

sirPerna, 2018-03-11:
We're also getting a lot of issue reports about the VTX resetting(channel, band, power) every time you power cycle your quad using verison 3.3.0. When they go back to an older version it works fine. Was there anything changed related to this? Should I make a separated issue for this or is there a relation to this issue here?

wind0r, 2018-03-11:
@sirPerna update your lua scripts and the issue is gone

Bobbaone, 2018-03-15:
is it possible to flash a little "fix" for the smartaudio problem, or is there another version coming with a fix in it?

davedmr, 2018-03-30:
Any update of this. Running hobbywing omnibus f4sd and want to use Unify pro nano. Happy to go with hardware pulldown if someone can let me know what what value resistor and where to connect it. Many thanks

DieHertz, 2018-03-30:
For now use 10kOhm to ground on SmartAudio pin

davedmr, 2018-03-30:
Thanks. Sorry to be slow. I am very much a newbie to all this. So do I connect the GND/resistor to tx pad I will using for the smart audio. Or do I connect it some where else. Many thanks.

DieHertz, 2018-03-30:
You've got it right, a 10K resistor between `GND` and SmartAudio `Tx` pad.

davedmr, 2018-03-30:
Many thanks Diehertz. Appreciate your help.

GhostMaster9, 2018-03-30:
WTF, having to add one resistor to use the smartaudio is just too painful for someone who payed for an original tbs product... maybe it's better to just buy akk and be happy, they do work on betaflight 3.3 now. 
It's a shame what tbs is making, this should be already fixed.

jflyper, 2018-03-30:
@brucesdad13
Can you identify the circuit condition on the output side of the SA? Pull-up, pull-down, such things.

robertlacroix, 2018-03-31:
@DieHertz why don't we merge in your change? It works fine on the Unify Nano and a Unify Race I tried.

codecae, 2018-03-31:
Building a pull-down into the code could affect people that use audio by sending it to ground.  This is a workaround rather than a fix.

RipperDrone, 2018-03-31:
I understand. Still I'd prefer majority of users (NOT using audio, like there is not a single one in our little group here, at least we are 25 people) getting a simple solution offered which is software based, rather than having to solder in tiny components... :-(

Maybe could there be a CLI switch to trigger the software pulldown? Can't be so code size heavy...

codecae, 2018-03-31:
There have been some discussions and possibly some things in the works that may provide extra configuration for one off scenarios like this without hard coding them into the protocol implementation.

DieHertz, 2018-03-31:
You will have a software solution in 3.4, which is still some months away. We have to do that right.

For now either cherry-pick my commit or use a hardware pulldown

RipperDrone, 2018-03-31:
fair enough :-)

[18 more replies]

closed 2019-02-21 by robertlacroix