betaflight/betaflight #5330: OSD Screen: "RTC Date Time" not included in Diff dump when set from Configurator

Arakon, 2018-02-27:
[image]

When pasting a "diff all" dump to a fresh flash/full erase, the RTC setting remains unset, so it's not included in the diff output.

mikeller, 2018-02-28:
@Arakon: Looks like the entry for RTC stats in the CLI settings was forgotten. But the RTC time during flight dumps just fine for me:

```
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.0 Mar  1 2018 / 01:24:07 (b0ff928af) MSP API: 1.37

[...]

set osd_rtc_date_time_pos = 2470
```

Arakon, 2018-02-28:
Might be that it wasn't actually set in this particular quad then, so it's likely just the RTC stats afterall.

PR #5339 (2018-02-28): Added missing `osd_stat_rtc_date_time` to CLI.
Fixes #5330.
closed 2018-02-28 by mikeller