betaflight/betaflight #5331: RSSI from xsr CH8 (AUX4) on UART1

geekness, 2018-02-27:
Hi all,

I've asked this question in the rcgroups thread, and also on discord and didn't get any help, so im coming here in the hope of getting an expert.

Im new to Betaflight and am putting together my first BF quad build.
Im not new to building quads though.

Im having an issue with setting up my RSSI in BF.
Im using an FRSKY XSRf4O, which has a built in xsr receiver on uart 1. The XSR receiver outputs RSSI onto CH8 (Betaflight AUX4) automatically.

What happens is that the value of the AUX4 on the receiver tab is stuck on 1971 (or maybe 1791?), even if I turn the transmitter off. 
I have all the endpoints set to 1000-1500-2000, +-4.

I tried to set an output in the mixer on my x9d, but then found a video saying that the xsr receivers don't need it done that way. It's just done automatically through CH8 (AUX4) . I have telemetry coming from my FC into the X9D and can see the RSSI. When I had the RSSI set as an output on CH8 in my mixer, there was no change to the CH8 value in the receiver tab in BF.

If I start with the FC connected to BF, but with the X9D turned off, then AUX4 is at 1500. When I switch the X9D on, AUX4 goes up to 1971.

If I do a "set rssi_invert = ON", there is no change.

```# version
# Betaflight / FRSKYF4 (FRF4) 3.2.4 Jan 10 2018 / 22:49:20 (90d32fd56) MSP API: 1.36

# name
name Purple

# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 A01

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 2 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1400 2100
aux 1 1 0 1800 2100
aux 2 13 1 1800 2100

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set acc_trim_roll = 10
set min_check = 1015
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_kill_switch = ON
set pid_process_denom = 1
set osd_cap_alarm = 1500
set osd_tim1 = 2561
set osd_tim_2_pos = 33
set osd_throttle_pos = 225
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set vcd_video_system = 2
set camera_control_key_delay = 125

# profile
profile 0

# rateprofile
rateprofile 0
```

https://www.frsky-rc.com/wp-content/uploads/2017/07/Manual/XSRF4O/01140067-XSRF4O.pdf
https://img.banggood.com/file/products/20160119004828XSR(2).pdf

Hoping somebody here can help. TIA.

jflyper, 2018-02-27:
Let us see a screen shot (screen copy) of the receiver tab when the embedded XSR is bound to a TX.

geekness, 2018-02-27:
Hi, I didnt notice your response on RCG, sorry about that.

Anyway....you're right. I had my failsafe setting to have AUX4 set to "hold", which kept the RSSI level constant. 
Thanks Heaps, issue can be closed now.

closed 2018-02-27 by geekness

jflyper, 2018-02-27:
Happy props :)