betaflight/betaflight #5332: omnibus f4 error

msolergelpi, 2018-02-27:
Ya instale el firmware a una controladora omnibus micro f4 nueva correctamente, el problema es que cada vez que quiero guardar una configuración "save and reboot" me redirige a la pagina principal de betaflight con el error "Failed to close serial port".

Ya actualize todos los dirvers, el zading y la placa la puse en modo "boot" y sigue dando el mismo problema.

Con otras placas no tengo ningún problema.

Gracias de ante mano.

GTorreil, 2018-02-27:
What does it look like in English ? Thanks !

mikeller, 2018-02-27:
@msolergelpi: English is the language commonly used in Betaflight development, and the language that pretty much everybody in here speaks. In this respect, if you want your question to be understood by the majority of the people that can help you, please write in English.

msolergelpi, 2018-02-28:
I already installed the firmware to a new micro f4 bus controller correctly, the problem is that every time I want to save a task I save and reboot redirects me to the betaflight main page with the error "Failed to close serial port".

I already updated all the dirvers, the zading and the board I put it in "boot" mode and it still gives the same problem.

closed 2018-02-28 by msolergelpi