betaflight/betaflight #5333: 1.5khz blackbox log: Wrong p-Interval?

Plasmatree, 2018-02-27:
Hi,
using 1.5khz logging frequency on my Matek f405 mini produces logs that freeze blackbox viewer as well as blackbox_decode. At 2khz no problem.  

The only difference inspecting the header is (line 19/20):
```
H P interval:1/4
H P denom:64
``` 
@2khz and 
```
H P interval:0/1
H P denom:48
``` 
@1.5khz

setup is running bf 3.2 with 8k/8k

Changing P interval to 1/1 solved the problem. 
Seems like I'm not the only one experiencing this issue: [blackbox-tools/issues/19](https://github.com/cleanflight/blackbox-tools/issues/19
)
Could there be a problem with 1.5khz beeing an odd denominator of 8khz? 
Not sure if this is a blackbox-tools or BF issue.

Have a great day,
Flo 

```
# version
# Betaflight / MATEKF405 (MKF4) 3.2.0 Oct 12 2017 / 15:59:13 (f2f34b4) MSP API: 1.36

# name
name guinnipig

# resources
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource ADC_CURR 1 NONE

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR
feature DYNAMIC_FILTER

# beeper
beeper -DISARM_REPEAT
beeper -ON_USB

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1150
aux 1 1 1 900 1150
aux 2 4 1 1800 2075
aux 3 13 2 1850 2100
aux 4 26 2 900 1150
aux 5 28 1 1750 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_trim_pitch = 4
set rssi_channel = 16
set blackbox_p_ratio = 48
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 5
set failsafe_kill_switch = ON
set bat_capacity = 850
set current_meter = ESC
set battery_meter = ADC
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_tim1 = 2561
set osd_rssi_pos = 2105
set osd_tim_1_pos = 2518
set osd_tim_2_pos = 33
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2082
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2497
set osd_mah_drawn_pos = 2466
set osd_craft_name_pos = 2506
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2489
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set vcd_video_system = 1

# profile
profile 0

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set p_pitch = 60
set i_pitch = 45
set p_roll = 58
set d_roll = 35

# rateprofile
rateprofile 0

set rc_rate = 140
set rc_rate_yaw = 140
set rc_expo = 5
set tpa_rate = 0
```

DieHertz, 2018-02-27:
Please be so kind as to attach a sample log :)

Plasmatree, 2018-02-27:
I just noticed that by just arming once (without starting to log) the internal 32mb flash is filled up with 16mb nothingness (added a corresponding log to the examples). 
Might this be related?

Plasmatree, 2018-02-28:
updated to 3.3 rc3:
Memory utilization is now ok (all 32mb can be used). The sample logs linked above are updated (and way smaller).

Initial problem still exists: 
Logs recorded at 1.5khz freeze logviewer/blackbox tools unless p-interval is changed from 0/1 to 1/1.

ledvinap, 2018-02-28:
IMO current calculation is wrong, it should be: `blackboxGetRateNum()` = 1 and `blackboxGetRateDenom()` = `blackboxPInterval`

(@martinbudden  ? )

DieHertz, 2018-02-28:
Will test it tonight, haven't seen Martin in a while.