betaflight/betaflight #5335: OSD display problem (with telemetry switch = ON)

tfoutfou, 2018-02-27:
Hardware :   Spracing f3 --  Frsky X4R SB --  Micro Minim OSD --  TBS VTX with smartaudio

BF : 3.2.5     OSD : 1.7.4.0

UART1 : USB  shared with smartport telemetry (on a switch with option  tlm_switch = ON )
UART2 : MSP connected to OSD
UART3 : X4RSB
softserial 1 : TBS Smartaudio

Hi , with the hardware and configuration above .
When i enable the telemetry switch on a RC dedicated channel i get an OSD problem in the MODE display.
Instead of having  : [STAB] [HOZN] [ACRO]  when i enable the telemetry on the switch the MODE change to [WPT] : 0     (which is not logic at all since it as nothing to do with any mode whatsoever )

So i think the problem is part of betaflight sending a wrong "MODE" to scarab OSD.

Could you please tell me if it come from BF and maybe fix it. Thank you.

mikeller, 2018-02-27:
@tfoutfou: Looks like scarab OSD reads modes by index. Betaflight has switched to using permanent ids for modes some time ago, which has the advantage of the ids not changing when new modes are added / old modes deleted.
scarab OSD should switch to reading permanent mode ids from Betaflight. Please open  a ticked with scarab OSD for this.

ShikOfTheRa, 2018-02-28:
I have had a look. Betaflight has assigned telemetry to the id that was previously used to GPS mission in MSP protocol.

This is easy to address now I understand the changes.
I have implemented for 1.8 

Suggest this is closed and refer to https://github.com/ShikOfTheRa/scarab-osd/issues/478

closed 2018-02-28 by tfoutfou

mikeller, 2018-03-01:
Thanks for looking into this, @ShikOfTheRa.