betaflight/betaflight #5338: Steady Buzzer Tone

mfocht001, 2018-02-28:
Is there a way to get a steady buzzer tone? I am trying to get a steady voltage from the buzzer tone in order to toggle a switch. This would allow people to hot swap quads in a two quad setup. Power two quads at once and switch camera/vtx power to the second quad after battery runs low on the first. This would speed up practice sessions. I am doing this now with seperate arm switches and different video frequencies but it could be so much easier with a basic switch feature. If there is a hacky way, i will try that also.

Uses:

1. Use as a universal switch for other functions.
2. Hot swap quads for practise session
3.2 camera setup ( rearview camera ) 

Appologies if there is a specific place for feature request. let me know if/where to post.

jflyper, 2018-02-28:
Wrote a simple guide for BF wiki.
[Pinio and PinioBox](https://github.com/betaflight/betaflight/wiki/Pinio-and-PinioBox)

mfocht001, 2018-02-28:
These are great resources. Many thanks. From a technical perspective this looks to be possible. But, I was wondering if there could be a feature added to simply turn the buzzer into a switch. Emphasis on feature request. 

At the moment, I am certainly going to use pinio to accomplish this.

From a GUI perspective, having a simplistic option to turn the buzzer into a switch would greatly simplify this. The pinio method is not at all intuitive to figure out. Even having a decent amount of experience. BUT it is awesome there is even this option.

The reason I choose the buzzer is that almost all flight controllers have that port. A simple slider in betaflight to activate the buzzer port as a basic switch. That is the feature request.

Arakon, 2018-02-28:
You'll have to keep in mind that the buzzer pin can't handle high power applications.. i.e. actually switching the power to a camera, vtx, etc will fry the pin.

mfocht001, 2018-02-28:
@jflyper  Thanks for the guide. This helps describe a few things. It does leave me with more questions though. How do we know what pins are occupied already? How do we find out what pins the flight controller is using for the LED or Buzzer pin? What is a BOX? What is a mode? To be able to look at your FC and say I want to reasign the buzzer port, how would you go about that process? 

Example:
1. Commands to find available port
2.Commands to resign buzzer port.
3.Command to check new pin assignment.

These are the questions that an average non-developer user would have. Can you add some of those commands to your guide? You have a great technical explanation of what pinio _does_  but not how to _use_ pinio from an end users perspective. The working examples would seal that guide up nicely!

mfocht001, 2018-02-28:
@Arakon Yes, that did occur. But at least having a constant voltage to drive a $1.00 transistor would open up many many options and possibilities.

jflyper, 2018-02-28:
@mfocht001 Thanks for the suggestion.

> What is a BOX? What is a mode?
Take a look at Modes tab on your configurator. Each entry is a box.
For further explanation, search the web with "betaflight setting up modes".

> 1. Commands to find available port
`resource list` prints current active assignments.

> 2. Commands to resign buzzer port.
See below.

> 3. Command to check new pin assignment.
Do `save` and when the FC comes back again, go into CLI and do `resource list` again.

You should also take a glance at what variations of `resource` command does.
https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping

As per your preference of using the buzzer output for your application, you might be able to do it by
```
resource list # Find out which pin beeper is using.
resource beeper none
resource pinio <original-beeper-pin>
```

Also note that most buzzer implementation uses MOSFET based low side switch.