betaflight/betaflight #5344: DYS F4 V2 Pro Flightcontroller wiht Xsr-m  F.Port.

JetHunter83, 2018-03-01:
I have a DYS F4 Pro V2 and would like to run it with an Xsr-m receiver. Receiver is updated with the current F.Port firmware. Empfanger runs too, I can control the copter. But telemetry does not work. I have tested all Uarts without success. Receiver runs only on Uart 1 but without telemetry. I have the command:                set serialrx_halfduplex = OFF
set serialrx_inverted = ON up and down ... 
One idea?

ttjs82, 2018-03-01:
I have the previous version DYS F4 Pro without the 9 volt BEC. I got it working with F.port using the R-XSR receiver by doing the uninversion mod. So I connected the receiver by soldering the uninverted pad to the flight controller TX1 pad. Then I set the serialrx_halfduplex = ON. Also used the Betaflight 3.3.0 RC3. If I tried to connect the receiver to the flight controller without the mod I couldn't get it working in F.port mode.

I guess that the newer DYS F4 PRO V2 doesn't have the needed bidirectional inverter on any of the UARTs either. That's why you need to do the uninversion mod also for the XSR-m receiver https://goo.gl/bC12Jk Unfortunately it isn't as easy as with the R-XSR receiver which has the solder pad ready for this. There are also some flight controllers that work without the uninversion mod like Kakute F4 V2 AIO which has the SP (smartport) pad that you can use to connect F.port receiver without a mod (have tested this with R-XSR receiver).

JetHunter83, 2018-03-01:
Stimmt der Hack-Mode ist nicht einfach.. Ich muss aber vorher den XSR-m vorher zurück flashen. Weil nachdem Mode Hack geht kein flashen mehr... Ich werde mal schauen ob ich das schaffe. Falls ja muss ich ganz normal Sbus aktivieren und den Smartport aktivieren, stimmts?

mikeller, 2018-03-01:
@JetHunter83: I never thought about the implications of the uninverted SmartPort hack and flashing, good point. You can still flash the RX by using `serialpassthrough` in Betaflight, see https://github.com/betaflight/betaflight/issues/3364#issuecomment-312121695.
It might still be worthwhile opening a feature request to support flashing of RX with uninverted hack in OpenTX / er9x - the hardware should support it.
I don't know what you mean by 'activate SBus and SmartPort'. Once everything is set up, only using FPort should work fine.

JetHunter83, 2018-03-01:
Ja man muss auf jeden Fall vorher flashen, bevor man den Hack Mode ausführt. War ein denk Fehler von mir, Sbus braucht man ja dann nicht.  Ich lass den XSR-M einfach auf F.Port geflasht. Ich werde berichten ob es klappt.

PsyDafke, 2018-03-09:
Have similar issue on Betaflight F3 board (tried both UART 1 & 3 TX's), but in my case the telemetry works, but control doesn't (R-XSR receiver with latest F.Port FW). Other setup with a SRF3 target works fine w/ R-XSR connected on it's UART2 TX.

PsyDafke, 2018-03-17:
UPDATE!!!
Apparently, it was a user error on my end. Seemed that I accidentally held the bind button on the R-XSR too long (same function to change between SBUS and CPPM output)! This still seems to 'work' with the FPORT firmware!
Now have both control and telemetry working just fine.
Hope this helps out others hunting down F.Port issues.
Now, let's go out and maiden this baby! 👍

kpmzam, 2018-03-27:
I successfully got my r-xsr working with Fport on the DYS F4 Pro V2. I did not need to do the uninversion hack. I have s.port connected to the sbus pad (uart2 rx?) and think I just lucked out that the inverted is bidirectional?
BUT there's a problem. I'm having failsafes at pretty close range and rssi seems pretty unstable. Checked everything else/compared to other r-xsr, 99% sure it's firmware related at this point. Using 2018-03-08 180228

kpmzam, 2018-05-30:
@pavel-d I would have no problem posting a diff dump if I hadn't already reverted it back to the normal 2 wire s.port and sbus configuration shortly after posting here due to the link being totally unreliable and failsafing at close range. 
With normal setup it's fully reliable and nothing changed other than wiring and firmware. So i think it's not worth pursuing until frsky releases updated firmware.
As far as i remember the cli settings were mostly stock? If not for one of the inverter options being enabled

pavel-d, 2018-05-30:
@kpmzam, I see, thanks for explaining. Did you use uninverted hack for s.port signal? on which UART?

DocentRC, 2018-07-19:
@kpmzam There is a small problem. The DYS Pro F4 VFlight Controller does not have a UART2 port. It has only UART1, UART3 and UART6 and only UART RX1 is for SBUS.