betaflight/betaflight #5345: Configuring Barometer on SPRacing F3 Evo

myuunosuke, 2018-03-01:
Hi all, 

I'm trying to configure a barometer module for SPRacing F3 Evo.
I was be able to configure UART3 to i2c, and successfully get "baro_i2c_device" to be 1.
However, when I go to the Sensors tab on BetaFlight configurator, the checkbox for barometer does not let me check and does not show the barometer info.
I would appreciate if someone could point me out what I'm missing or doing wrong...
[image]

Here is my setup...
- FC:  SPRacing F3 Evo based FC for Brushed motors (Firmware:  SPRacing F3 Evo 3.2.5)
https://www.banggood.com/Eachine-32bits-F3-Brushed-Flight-Control-Board-Based-On-SP-RACING-F3-EVO-For-Micro-FPV-Frame-p-1076530.html?rmmds=search&cur_warehouse=CN

- MS5611 Module
https://www.banggood.com/MS5611-GY-63-Atmospheric-Pressure-Sensor-Module-IICSPI-Communication-p-965980.html?rmmds=myorder&cur_warehouse=CN

Barometer has been connected to FC as following,
- SCL pin from the MS5611 module is connected to UART3 TX on FC
- SDA pin from the MS5611 module is connected to UART3 RX on FC
- VCC pin from the MS5611 module is connected to 3.3V on FC
- GND pin from the MS5611 module is connected to GND on FC

On BetaFlight Configurator, Barometer has been enabled.
[image]

Resource has been changed as following,
- Freed uart 3 tx and rx
- Assigned freed pins to i2c 2
  CLI Command:  "resource i2c_scl 2 B10", "resource i2c_sda 2 B11"

# resource
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource PPM 1 A15
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
resource I2C_SCL 1 B06
resource I2C_SCL 2 B10
resource I2C_SDA 1 B07
resource I2C_SDA 2 B11
resource LED 1 B08
resource TRANSPONDER 1 A08
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 A15
resource ADC_BATT 1 A04
resource ADC_RSSI 1 B02
resource ADC_CURR 1 A05

# get baro_i2c_device
baro_i2c_device = 1
Allowed range: 0 - 5

etracer65, 2018-03-01:
Did you try `set baro_hardware = MS5611` and did you set `baro_i2c_address` to the I2C address for your baro device?

myuunosuke, 2018-03-01:
Thank you for your quick reply!
The baro hardware was originaly set to "Auto", and baro address was set to "0".
I have changed as you suggested and still the same result.
[image]

myuunosuke, 2018-03-01:
I checked status on my FC and there were 2 i2c errors found.
[image]

jflyper, 2018-03-01:
@myuunosuke 
You have multiple problems.
- SPRACINGF3EVO does not support I2C2.
- PB10 and PB11 are not valid pins for I2C2.
- Your baro is configured to be on I2C1 (baro_i2c_device = 1).

jflyper, 2018-03-01:
One of your option would be to use UART2_TX (PA14) as I2C1_SDA and UART2_RX (PA15) as I2C1_SCL.

Other options require non-trivial mods on software and hardware.

myuunosuke, 2018-03-01:
Oh..., that's why....

> PB10 and PB11 are not valid pins for I2C2.
> Your baro is configured to be on I2C1 (baro_i2c_device = 1).

Is this because F3 only can have "i2c1" or because im using wrong pins?

If I only can use default i2c pins on this SPRacing F3 Evo, is there any way accessing it to add my baro (any soldering pads or anything)?

myuunosuke, 2018-03-01:
It seems like UART2 is configured for a receiver, is it possible to move the receiver setting to UART3?
I've looked at data sheet and it seems like UART3 is not configurable for i2c.

[image]

[image]

jflyper, 2018-03-01:
> It seems like UART2 is configured for a receiver, is it possible to move the receiver setting to UART3?

That's what I've recommended above.

> I've looked at data sheet and it seems like UART3 is not configurable for i2c.

That's what I've pointed out above.

myuunosuke, 2018-03-02:
I moved my baro module to Uart2 and still the same result... I have 2 i2c errors showing up.

Pin configuration is now...
- FC UART2 TX -> Baro i2c SDA (White wire)
- FC UART2 RX -> Baro i2c SCL (Yellow wire)
- FC 3.3V -> Baro i2c VCC
- FC GND -> Baro GND

[image]

# resource
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
resource I2C_SCL 1 A15
resource I2C_SDA 1 A14
resource LED 1 B08
resource TRANSPONDER 1 A08
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ADC_BATT 1 A04
resource ADC_RSSI 1 B02
resource ADC_CURR 1 A05

# get baro_bustype
baro_bustype = I2C
Allowed values: NONE, I2C, SPI

# get baro_i2c_device
baro_i2c_device = 1
Allowed range: 0 - 5

# get baro_hardware
baro_hardware = MS5611
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280

# get baro_i2c_address
baro_i2c_address = 119
Allowed range: 0 - 119

# status
System Uptime: 121 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500
SD card: None inserted
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 2, config size: 2019, max available config: 4096
CPU:17%, cycle time: 129, GYRO rate: 7751, RX rate: 49, System rate: 9
Arming disable flags: RXLOSS ANGLE CLI

jflyper, 2018-03-02:
Need output from `resource list`, to verify PA14 and PA15 are actually assigned to I2C1 and working.

Also need a hi-res pic of the baro module... both sides.

jflyper, 2018-03-02:
Also noted that the baro module should be powered by 5V, not 3V3, as the module has 3V3 LDO.

jflyper, 2018-03-02:
Also need you to measure voltage at PS against GND (Voltage at PS determines interface mode--  I2C or SPI --).

jflyper, 2018-03-02:
Oh wait... this GY-63 module is designed to interface with 5V systems; it has onboard level shifters...

- If we power this module with 3V3, then internal voltage would be lower then 3V3...
- Vcc (interface level) is still 3V3 in this case, and higher than internal voltage due to drop out voltage at LDO, so the level shifter works.
- MS5611 supply voltage is 1.8-3.6V...

Yeah, power the module with 3V3. Your current power wiring should be okay.

myuunosuke, 2018-03-02:
[image]

[image]

\# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT
A05: FREE
A06: FREE
A07: FREE
A08: TRANSPONDER
A09: FREE
A10: FREE
A11: FREE
A12: FREE
A13: FREE
A14: I2C_SDA 1
A15: I2C_SCL 1
B00: FREE
B01: FREE
B02: ADC_RSSI
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: FREE
B07: FREE
B08: LED 1
B09: MPU_CS
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: SDCARD_DETECT
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

jflyper, 2018-03-02:
@myuunosuke 
You seem to have everything connected and configured correctly.
If the issue persists, you'll need a logic analyzer to trouble shoot.

Only thing I can think of is the operation of the level shifter under 3V3 and 3V3 - {dropout voltage} condition.

Sorry I couldn't help you.

myuunosuke, 2018-03-03:
@jflyper 
Thank you very much for for helping me on this!
Even though we couldn't solve the problem, all the information you gave me was very useful.

I will try the sensor with my arduino to see if the sensor even works or not.
Also, I will try another barometer such as BMP180 and see if it makes any difference.

Thanks,

myuunosuke, 2018-03-04:
I've tested with BMP180.
Now, I only get one i2c error showing up.
Is there any way to see what's those i2c errors are generated from?

The end result is still the same..., the check box is not letting me check on the sensor tab. 
 
[image]

jflyper, 2018-03-04:
It's really unfortunate that SPRACINGF3EVO target does not support BMP180... 😭 It does support BMP280...

jflyper, 2018-03-04:
Custom firmware that supports BMP180 instead of BMP280 is the only chance.

myuunosuke, 2018-03-04:
Oh..., I thought BMP180 is the upgraded version of BMP085, and works the same...

The other thing I'm thinking about is soldering small wire to the SMT original i2c pins (b6, and b7) and connect them to MS5611 module.

Will it work if i directly connect those b6 and b7 to the module?
Do i need to add some pullups or any other things between?

Thanks,

jflyper, 2018-03-04:
> Will it work if i directly connect those b6 and b7 to the module?

I should have said this; the SPRACINGF3EVO _*ONLY*_ supports BMP280. No MS5611 nor BMP180.

myuunosuke, 2018-03-09:
> I should have said this; the SPRACINGF3EVO ONLY supports BMP280. No MS5611 nor BMP180.

That's the answer...
I've purchased BMP280, but I decided to use full size FC with pre-installed baro for now.
When I have time to work on this tiny boad again, I will come back and post some results.

Thank you!

closed 2018-03-09 by myuunosuke

Vp541, 2018-10-18:
@myuunosuke Did you get time to setup the brushed sp racing F3 evo with the BMP280 Barometer? Really curious on the outcome as I'm also working on something similar as I need altitude hold on the Brushed FC's for my project.

Vp541, 2018-10-18:
@myuunosuke @jflyper @etracer65 Out of your experience with Brushed FCs, Is there any with Barometer/ can hold altitude? I've built a brushed FC quad which I can control using PPM over WiFi(ESP12E module) from my smartphone. But, I want to add Altitude hold Feature to it. I use a Brushed SP racing F3 Evo FC and my wifi module takes up the UART2. I can use a ESP-01 on UART1 and make room for the barometer on UART2. Kindly let me know the best approach for a Brushed Quad to hold Altitude. Thanks in advance.

etracer65, 2018-10-18:
Betaflight no longer has any altitude hold code enabled so there is no option.

josemidevel, 2018-10-30:
I have iNav working on that fc. Maybe could be an option.
Cheers

DaniilVeselov, 2019-04-05:
I just connected MPU280 to this board. Betaflight shows BARO !

actomaz, 2019-04-12:
Hello All!! 

> I just connected BMP280 to this board. Betaflight shows BARO !

Tell me more about this, did it work?

Thank you

ARsourav, 2019-04-19:
Hii guys..I'm using EACHINE FL F3 EVO BRUSHED flight controller and i have also a BMP280 barometer.but how to connect them in uart1 port

mikeller, 2019-04-29:
@ARsourav: BMP280 is I2C or SPI, not UART, so you will have to use one of these ports on your board. Ask your manufacturer where to find them.

Vp541, 2019-06-16:
@DaniilVeselov @jflyper @myuunosuke COuld anyone of you please shed some light on this case?
I tried doing exactly as in this article but configured the I2C to UART1 an connected the BMP280. But, no luck. Any help is appreciated.

# dump
# Betaflight / SPRACINGF3EVO (SPEV) 4.0.3 Jun  1 2019 / 11:55:06 (094cfc956) MSP API: 1.41 / FEATURE CUT LEVEL 8
batch start
name -
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A15
resource LED_STRIP 1 A08
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource I2C_SCL 1 A09
resource I2C_SCL 2 NONE
resource I2C_SDA 1 A10
resource I2C_SDA 2 NONE
resource LED 1 B08
resource LED 2 NONE
resource LED 3 NONE
resource TRANSPONDER 1 A08
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource ESCSERIAL 1 A15
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 A04
resource ADC_RSSI 1 B02
resource ADC_CURR 1 A05
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
[... 501 lines cut ...]
set horizon_tilt_expert_mode = OFF
set motor_output_limit = 100
set auto_profile_cell_count = 0
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
batch end

NANYO-SHINONOME, 2020-05-26:
[image]
[image]
[image]
[image]
[image]

yurem, 2020-10-07:
The issue was closed. But there is no full instruction how to do this. Can you help me with this?

yurem, 2020-10-12:
@NANYO-SHINONOME Can you give instruction how to connect it?

NANYO-SHINONOME, 2020-10-13:
i2c port  CPU direct connect

yurem, 2020-10-13:
@NANYO-SHINONOME Do you mean these 2 pins?
[image]

Why blue wire on the board is needed?

NANYO-SHINONOME, 2020-10-15:
TM32F303CCT6

Which firmware are you using?  I'm using betaflight.

no need blue wire
maybe test

NANYO-SHINONOME, 2020-10-16:
The blue wire just connected the boot pin.  Ignore it as it doesn't matter.

2020年10月15日(木) 15:15 東雲南陽 <ajvict@gmail.com>:

yurem, 2020-10-16:
[image]

yurem, 2020-10-16:
[image]

[image]

[image]

yurem, 2020-10-16:
@ NANYO-SHINONOME Thank you for your help. With latest Betaflight firmware Barometer work out of the box

motorello, 2024-01-08:
> @NANYO-SHINONOME Do you mean these 2 pins? [image]
> 
> Why blue wire on your board is needed?

@yurem Hi, sorry for reopening such an old thread... I'm trying to restore an old and burnt SPRacingF3 controller, I'm looking for the electrical schematic and I went through your partial screenshot of what seems to be what I'm looking for... Is there any chances to have the complete schematic?
Thanks in advance!
Francesco

NANYO-SHINONOME, 2024-01-08:
Hello, I'm sorry but I don't have the circuit diagram. I am sorry that I
can not be a help.