betaflight/betaflight #5346: Airmode does not work properly with throttle value below 1000

wuhao19920619, 2018-03-01:
The ESC protocol I use is DSHOT600 and my target is OMNIBUSF4SD. The betaflight version is 3.3-RC3.
My transmitter is X9D. 

Because I can't match the lowest throttle value to exact 1000, so I set it to 997. 

I notice that I can only arm my quad with throttle value below 1000, therefore this is the only setting I can apply.

But after I turned on airmode, with throttle value below 1000, it looks like only applied "half airmode", the position adjustment is stronger than which without airmode turned on, but the quad is not leveled. Only if I throttle up to 1000 or higher then the quad will suddenly leveled.

mikeller, 2018-03-01:
@wuhao19920619: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20min_command).

Also, you might want to check https://github.com/betaflight/betaflight/blob/master/docs/Controls.md#throttle-settings-and-their-interaction

closed 2018-03-01 by mikeller

wuhao19920619, 2018-03-01:
@mikeller Maybe you should read the last paragraph I think. The issue I think there might be is that airmode only "half applied" when throttle value below 1000. But I think logically it should works all the same no matter how much the throttle value is.

mikeller, 2018-03-01:
Yeah, I saw that. Not sure if that's actually a problem, or just a side effect of what seems to be a misconfiguration of your throttle levels, so I think adjusting them to recommended values and then checking again is a good first step.

wind0r, 2018-03-01:
@wuhao19920619  using a throttle value under 1000 isnt recomended. i even would stay it strongly advisted against doing so. the firmware is build around 1000-2000 range. 
In your case I simply would set it back to 1000 and ignore those 0.3% resolution

wuhao19920619, 2018-03-01:
@mikeller @wind0r Okay I guess betaflight is not be designed to deal with the throttle value under 1000. I will reconfigure it to match. Thank you

wuhao19920619, 2018-03-02:
@mikeller @wind0r I have just test setting min_check to 1010 and min_command and min_throttle keep default with 1000 and 1070 respectively. 

But I see that the airmode only applied after 1010(min_check). which is not "throttle 0". The throttle value between 1000 and 1010 is like "half airmode". I think maybe this can be considered as an issue？

mikeller, 2018-03-02:
@wuhao19920619: I think what you are seeing is PID control being disabled when below `min_throttle`. Try `pid_at_min_throttle = on`.

wuhao19920619, 2018-03-02:
@mikeller My pid_at_min_throttle is ON already. I think the pid is working all the time. But just the leveling adjustment strength is very weak when throttle value between min_command and min_check. I think it is not about min_throttle because my min_throttle is default 1070 and the airmode works good after 1010. 

Also, I think the min_throttle is not something about Dshot. It only works for PWM ESC protocol.

wind0r, 2018-03-02:
Your min_check is "throttle 0". Throttle will not be used there at that is not a airmode phenomen

wuhao19920619, 2018-03-02:
@wind0r So this should be a function required?
I personally think the airmode function allow PID fully control the quad in the air when throttle in the lowest position. But as your definition the "lowest position" here means the min_check. However during the fly when I low the throttle I can't distinguish whether it lower the min_check or not. I think in the most case the throttle will just be the lowest value, which should be min_command, at least between the min_check and min_command(I think this should be defined as "throttle 0"). Therefore I suggest the "throttle 0" for airmode should be defined as min_command.

AndersHoglund, 2018-03-02:
@wuhao19920619 you are mixing things up. `min_command` is a parameter on the ESC output end of the system, has nothing to do with the throttle on the RC-input end. The valid arameters to work with for RC throttle stick are `min_check` , `dshot_idle_value` and `airmode_start_throttle_percent`

wuhao19920619, 2018-03-02:
@AndersHoglund Sorry for the confusing description. The airmode I mean is that airmode in the mode tab which bind with an aux channel and can be turned on manually.

what I observe is: when I place my quad at an unlevel surface, then arm, there is almost no force to level it(this is okay), then I turn on airmode by an aux channel, the leveling force increased a bit, but still not strong enough to level it until I throttle up to min_check. 

Logically I think once  I turn on airmode with an aux channel it should level immediately, no matter the throttle is higher or lower than min_check. Because in the air when I low the throttle it is almost impossible to ask me to "low the throttle as low as possible but not lower then min_check"

AndersHoglund, 2018-03-02:
When powering on initially, airmode is not active until you throttle up above `airmode_start_throttle_percent`.

wuhao19920619, 2018-03-02:
@AndersHoglund yes, this is the normal process when you do nothing about airmode. But what if you manually turn on airmode or just turn the "airmode always on" on?

AndersHoglund, 2018-03-02:
Does not matter how you enable AM, permanent as a feature or on/off with a switch.
AM will not be activated until you throttle up above `airmode_start_throttle_percent` once, after that first throttle up AM will be active in the whole throttle range. It is a safety feature..... 
You can probably disable this by setting  `airmode_start_throttle_percent` to zero. But not charge me for your medical care bill. Settings safety features aside is NOT RECOMMENDED. Keep this enabled, keep seat belts on in your car, do not disable airbags etc etc.

wuhao19920619, 2018-03-02:
@AndersHoglund 
I think maybe you misunderstand the airmode switch.

ref:https://oscarliang.com/betaflight-airmode/
(Without Airmode, normally when you lower your throttle stick all the way down, the motors on a quadcopter should either stop or all spin at the same speed (depends on how you set it up).

But when Airmode is enabled, it keeps PID loops active when your throttle stick is at zero, which allows the pilots to steer and control the quad even in a free fall. That enables pilots to pull off some crazy aerobatics, and have better control over the entire throttle range.)

Therefore once I turn on airmode by switcher then the airmode should just be turned on permanently and the pid cycle works completely. It is nothing about throttle value once the mode be turned on manually.

wind0r, 2018-03-02:
@wuhao19920619 after arming you once need to have throttle above the threshold value to make airmode active. @AndersHoglund  is totaly correct there.

In your testing do you have the stick atleast once about the threshold value?

AndersHoglund, 2018-03-02:
@wuhao19920619 There are a lot of misunderstandings, incomplete information and des-information on the net. Including Oscars site....
Unfortunately, the BF docs and WikI do not describe this too well or not at all. So I can understand your confusion and frustration.

jcutcher01, 2018-07-15:
I am running 3.4.0 with airmode enabled via a switch. My min_check is 1010. I am noticing while airmode is enabled with my switch and I am just on ground at idle the motors sputter. When I disable airmode with my switch the motors are buttery smooth. This is telling me that the start percent parameter is not working when airmode is manually enabled with a switch. Am I wrong? My start percent is default at 32.