betaflight/betaflight #5348: BF 3.3.0 RC3 - Tramp VTX resets to Channel F1

realcris23, 2018-03-01:
Hi,

tested with Omnibus F4 and Matek F405 with BF 3.3.0 RC3.
Tramp VTX resets to Channel F1 if it is used as periphery.

It is possible to change the settings via LUA Script on the Taranis.
But it resets after the FC is restarted.

It is not possible to change the channel on the Tramp VTX itself.
In the Channel Tab (2x long blinking) it starts blinking continously if you want to change the channel.

The issues do not appear if the Telemetry wire (Tramp>FC) is disconnected.

The issues do not appear in BF3.2.5!

Kind regards,
Chris

(my first issue report - hope it's ok like that)

wind0r, 2018-03-01:
Did you update the lua scripts?
What does `get vtx` reports in cli?

realcris23, 2018-03-01:
Is it mandatory to update the lua scrips for BF3.3.?
Then thats the problem i guess - i did not update the lua scripts.

get vtx
osd_vtx_channel_pos = 377
Allowed range: 0 - 3071

vtx_band = 4
Allowed range: 1 - 5

vtx_channel = 1
Allowed range: 1 - 8

vtx_power = 1
Allowed range: 0 - 4

vtx_low_power_disarm = OFF
Allowed values: OFF, ON

vtx_freq = 5740
Allowed range: 0 - 5999

vtx_pit_mode_freq = 0
Allowed range: 0 - 5999

vtx_halfduplex = ON
Allowed values: OFF, ON

wind0r, 2018-03-02:
Yes is is mandatory to update the Lua scripts if you want to persist the channel changes or you need to change the channels via cli to make save them between reboots

realcris23, 2018-03-02:
...so this issue is closed

wind0r, 2018-03-02:
is it working? just click the close button then :)

realcris23, 2018-03-02:
yes all fine after updating opentx!
Thank you!

closed 2018-03-02 by realcris23