betaflight/betaflight #5351: New AKK vtx issue on BF 3.3 RC3 - Can't change to the power #4.

le3dmax, 2018-03-02:
Hey guys! 
Akk sent me some of their NEW vtx for testing that supposed to be working with BF 3.3,  
They seem to be working now, can change channels, power( number 1 to 3, 25mW/200mW/500mW), but I can't change to the highest power settings. (number 4, 800mW or 600mW, depends on the model).  Either on OSD or CLI, I'll test LUA tomorrow. 
I'm using the Bardwell F4 for this test.
Does anyone have the same issue?

set pwr_on_arm_grace = 5
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = OFF
set vtx_freq = 5740
set vtx_pit_mode_freq = 0
set vtx_halfduplex = ON

Thanks....

jflyper, 2018-03-02:
@le3dmax Can you get back to AKK and ask them which version (commit) they have tested the NEW vtx with? Once we have the information, we, with your help, can start bisecting which commit broke it.

closed 2018-03-02 by jflyper

le3dmax, 2018-03-02:
Great, I'll do that. Thanks.

codecae, 2018-03-02:
Please ask them to provide samples to our team.  We can solve issues better if we can actually physically interact with the hardware.

DieHertz, 2018-03-02:
I second that

AndersHoglund, 2018-03-02:
Even TBS has different power support on different products. Pro/Race/Nano.  But at least they differentiate by product name and clear specs. Is this "new" AKK VTX a race clone? (25/200), or a Nano clone (25/50). Or something in-between, 25/200/600 like Tramp? 
Any link to the "new" AKK?

codecae, 2018-03-02:
Better yet, have them develop their firmware to the specification and THEN deliver us samples.

The specification is how open to the public and they can utilize it in their design.

http://www.team-blacksheep.com/tbs_smartaudio_rev08.pdf

codecae, 2018-03-02:
@AndersHoglund We'll likely need to supply a generic list of "power levels" rather than mW ratings going forward to handle all of these different enumerations.  It's impossible to manage all of these one-off situations.

jflyper, 2018-03-02:
I propose going `MIN`, `LOW`, `HIGH`, `MAX`.

Manufacturers, including TBS, can put a table on a piece of paper they put in a package.

AndersHoglund, 2018-03-02:
Exactly. And we have an embryo of that generic list in the Spektrum VTX control code. Spektrum has a few generic power ranges, by index also. But the Spektrum VTX Control code needs to know all about all the other VTX vendor power indices to do a power index translations, both ways. Totally the wrong place, Spektrum code should not need to know anything about all the VTXs. VTX Common should handle all this in a generic way. As of now VTX Common is (partly) transparent, exposing VTX device details to the input end. It should perhaps generalize and isolate instead.
And this is not only about the Power tables. Band/Channel/Frequency/PitModePower/PitModeFreq/PowerOnOff/Polarization/MultiAntenna/ etc etc . I.e. some general model covering all VTX features.

le3dmax, 2018-03-02:
Yeah, they are sometimes vague about it but seems like they want to fix the issues. 
Here are the links to the VTXs they sent to me:
https://www.akktek.com/products/vtx/akk-fx3.html
https://www.akktek.com/products/vtx/akk-fx2.html
https://www.akktek.com/products/vtx/akk-x2p.html
https://www.akktek.com/akk-x2.html
Ok, I'll forward all this to them and I'll ask them to contact you about the samples. Thank you so much!

mikeller, 2018-03-03:
I second the removal of specific power output values - this potentially varies even within a given batch of the same VTX, if tolerances are high enough.

> I propose going MIN, LOW, HIGH, MAX.

What if a VTX has 5 levels?

Should we instead go for `1`, `2`, `3`,...

jflyper, 2018-03-03:
> What if a VTX has 5 levels?

We can add `FIRE` and `FRY`.

>_"Help!!! I have a problem with my vtx getting extremely hot. It's already smoking"_
_"What is your power level set to?"_
_"It's on FIRE."_
_"No wonder."_

le3dmax, 2018-03-04:
Hey. Installed the latest BF 3.3 and now everything works. Can change the power and channels now. OSD and Lua, all good. 
Thanks for your help guys. LD