betaflight/betaflight #5358: Bricked F4 Omnibus (green led of death)

flashez, 2018-03-02:
Hi,
I have a weird problem with two flight controllers (F4 Omnibus pro v2).
Both of them display a fixed green led and are not recognized by Windows (checked COM ports).
It was just before but when I unplugged the usb cable and plugged it back it displayed a fixed green led and the buzzer is yelling continuously.
I did check on google, it seems to be bricked but I don't understand why and how to unbrick them.

[image]
This one has already flown before.

[image]
This one is brand new from 1h ago but already bricked.

If you need additional informations just ask me.
Thanks for the help.

jflyper, 2018-03-03:
# Green LED is NOT a sign of death
# It is a main (MCU) 3V3 presence indicator.

jflyper, 2018-03-03:
That being said, I wants you to try re-flash the board by power cycling while the BOOT1 button is pressed. Have you done this before?

flashez, 2018-03-03:
@jflyper No could you explain please ?
Because nothing is happening when pressing BOOT button and plugging usb.

jflyper, 2018-03-03:
Read and watch this and do what they say.
https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

It's nice if you have a working FC so you know your PC/Mac is capable of detecting DFU mode MCU...

If you can't get them (the non-working FCs) into DFU mode, chances are that your MCUs are fried.

jflyper, 2018-03-03:
And since this is not a support forum, I'll be closing this. Please use other channels of support, e.g., RCG threads and BF FB.

closed 2018-03-03 by jflyper

jflyper, 2018-03-03:
That means those FCs are dud. Probably fried MCUs.

flashez, 2018-03-03:
@jflyper yes i think too, hard to believe it happened only with usb.
This model seems to be really bad.

jflyper, 2018-03-03:
Well, rumors are that these counterfeits are made with sub-standard components that are hand soldered. I would avoid them if I can.

etracer65, 2018-03-03:
From the pictures those certainly look hand soldered.

flashez, 2018-03-03:
Yes these are certainly really bad quality fc.
Gonna look for better ones

Koluy, 2021-05-15:
Omnibus F4 V3 Pro 
same problem fixed
Hold the Resset Button down dont release
dont select Dfu Mode and select install the right SD firmware in betaflight 
Regards