betaflight/betaflight #5361: 'Disarm regardless of throttle value...' option is no longer available in the Configuration tab for OMNIBUSF4 & OMNIBUSF4SD targets on 3.3

oleonav, 2018-03-03:
On OMNIBUSF4SD target the 'Disarm regardless of throttle value...' option is no longer available in the Configuration tab. Now disarming via switch directly kills the throttle value and drops the quad from the sky.

Tried setting this option to on via the cli. No success, option is not recognized anymore by the firmware

```Entering CLI Mode, type 'exit' to return, or 'help'

# get disarm_kill_switch
Invalid name

# set disarm_kill_switch = ON
Invalid name

#
```

- FC: Betaflight OMNIBUS F4 Pro (V2)
- BF Configurator:  10.2.0 Desktop on MacOS
- BF Firmware: 3.3

**Before on 3.2.5**
[image]

**On 3.3 firmware**
[image]

jflyper, 2018-03-03:
Duplicate #5297

closed 2018-03-03 by jflyper

floydhelios, 2018-03-13:
I installed 3.3 and crashed because disarm_kill_switch is gone.  I used this feature on all my quads and I'm used to it, so when I flipped the switch prior to coming in for a landing my machine dropped (luckily only broke a prop).  Please bring disarm_kill_switch back!  Leave it off by default and keep the option out of the GUI to prevent folks from accidentally setting it to 0 if the worry is that it's unsafe.  But leave it in the CLI so those of us that want it can use it.  I personally have never had a problem!