betaflight/betaflight #5371: Detect broken prop, Re-Activate Disabled Filters, Save on motor smoke.

JonnyPhenomenon, 2018-03-03:
in a previous feature request #3281 https://github.com/betaflight/betaflight/issues/3281
it was suggested to use ESC output to detect a damaged prop, However, couldn't this be done with pid errors, gyros, etc?   

More importantly, if any kind of "broken prop detection" could be used, it would be very useful if it could be implemented in a way that would Re-Enable any _manually disabled_ filters upon detection.

Disabling filters is all the rage, but they can make your motors run hot. this is exacerbated by unbalanced props.  Flying for more than a moment or two on a broken prop can destroy a motor if your filters are turned off, - its probably not GOOD to do even with all your filters on, but it could be whats needed to land your craft some where safely without frying a motor in the process. 

So if any kind of "broken prop detection" could be developed, it would be very useful if it could be implemented in a way that would Re-Enable any _manually disabled_ filters upon detection.

in a previous feature request #3281 https://github.com/betaflight/betaflight/issues/3281
it was suggested to use ESC output to detect a damaged prop, However, couldn't this be done with pid errors, gyros, etc?   

I had this idea when I broke a prop today on a branch (_while blackboxing!_), and landed about 3 seconds later. I ran over and picked up the drone and felt the motor. it was warm. the three others were cold.   Before I disabled the Gyro Notch Filter 1, I _think_ I could have flown all the way back to myself without that motor getting warm. 

In the interest of full disclosure, **_I_** cant actually tell from looking at the BlackBox logs that motor 3 was behaving any differently than the others, so maybe it isn't possible. But I dont have the first idea of what i am looking at either...

etracer65, 2018-03-04:
I marked this as a feature request.

I'm not really sure this is doable, but somebody might come up with a great idea.  The main problem I see is detecting the broken/damaged prop.  I'd presume we'd have to look for a net-increase in overall noise over time.  Continually monitoring the "noise floor" and looking for changes.  I presume it would be rather processor intensive but maybe it could be combined with the analysis logic in the dynamic notch filter.

Assuming detection is doable, then some algorithm would need to be in place to select the appropriate filtering.  Probably leave it up the the pilot by having low and high noise "filter profiles" or something like that.

DieHertz, 2018-03-04:
In fact the only thing that matters CPU-wise is the FFT computation, application of Dynamic notch is negligible. Have to collect some logs with `debug_mode = fft_freq` with a prop damaging event and observe the behavior

BrenLaw, 2018-03-04:
@DieHertz This is great news. I know it isn't necessarily a priority amongst all the other good work you devs have been doing, but having the Dynamic Notch be not only dynamic in the sense that it changes frequencies, but also dynamic in _when_ it's actually enabled would be a really great feature. You can fly without the added delay of the Dynamic Notch filtering when your setup is clean, but have the safety of the Dynamic Notch as soon as a prop gets damaged. 

There could be a user adjustable Threshold value which dictates the amplitude of motor noise required to make the Dynamic Notch enable. And also have a default (but still possibly user adjustable) cutoff value in Hz. which determines the lowest frequency motor noise that can still trigger the Dynamic Filter to turn on. This is important because typical in-flight movements (roll, pitch, flips, etc) are very high amplitude and would turn the Dynamic Filter on the entire flight.

I was thinking two CLI options:
- Dynamic Notch_threshold = ?? (don't know what scale would be used here) To set the noise amplitude required to enable Dynamic Notch.
- Dynamic Notch_cutoff = xxxHz. To set the frequency at which any gyro data lower than this point will not trigger the Dynamic Notch to turn on. My assumption is that by default this would likely be set somewhere near the soft Gyro Lowpass. Around 100hz.

My background is in audio engineering, so I've been following filter development in Betaflight very closely because it shares many similarities with EQing in audio. The core concept is essentially the same. So I've been trying to graft my knowledge into the software/firmware development realm, but I'm still by no means a coder. I don't have knowledge to try and create something like this myself. Just wishful, but hopefully well-grounded ideas.

JonnyPhenomenon, 2018-03-04:
@DieHertz - I break props constantly. I can get some logs... Although, my rig is a bit rickety, my pids arent all that good either, so you may not get very useful results... 

I'll enable debug_mode = fft_freq and go fly a few packs. its windy and cold out. perfect environment for prop breakage... :)

JonnyPhenomenon, 2018-03-04:
OK, I went out and broke a prop on my first flight. 18 seconds in. 
I have a couple rows of apple trees on my lawn and they have claimed more props than I can count.

I don't know if it will tell you much. once I hit the branch, I landed a second and a half later.  - not so much a crash, per se', but more of a "Hey, that was quick." followed by "Oh whoa, I have no throttle response wha-?" and then I was on the ground. 

the impact occurs at approximately 18.775 seconds into the log. 
I took a DVR of it too. 
you can download the pair from here: https://drive.google.com/open?id=1hb89MFSy31q6IjFvnNNyl_t6w5CTRSzd

JonnyPhenomenon, 2018-03-04:
if those are too short, I can try again... Also, I have other logs and videos where I DONT break a prop, if that might be useful..

DieHertz, 2018-03-05:
Just by the looks of it nothing unusual happens, the "noise floor" gets raised a bit, but there are much worse values during fast throttle maneuvres at 3-5s.
It looks as if this increase in the noise floor is indistinguishable from normal flying so far.
Need a longer log without immediate disarm after prop damage event.

flint723, 2018-03-07:
This log has a very bent prop in it but doesn't have the right debug.  Session 4 has the bent prop.
https://www.dropbox.com/s/eldr2wlum46x9pm/1407%20floss%20flight%2025.zip?dl=0