betaflight/betaflight #5377: BF 3.3.0 Matek F405 32khz sampling wont arm

NitroNAF, 2018-03-04:
For stable and RC3 my boards will not arm when the 'Use 32khz stampling' is enabled running a hexacopter motor mix.
The OSD shows a CALIB warning no matter the Gyro or loop speed set. e.g. 8khz.
When it is disabled and speeds are set to 8khz then it arms just fine. 
All overclock settings do not influence this bug at all.

Here is a dump of RC3.
# version
# Betaflight / MATEKF405 (MKF4) 3.3.0 Feb 25 2018 / 13:01:55 (b0ff928af) MSP API: 1.37

# name
name NAF_Katak_efaw

# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 A15
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
[... 525 lines cut ...]
set vtx_low_power_disarm = OFF
set vtx_freq = 5740
set vtx_pit_mode_freq = 0
set vtx_halfduplex = ON
set vcd_video_system = 1
set vcd_h_offset = 0
set vcd_v_offset = 0
set max7456_clock = DEFAULT
set displayport_msp_col_adjust = 0
set displayport_msp_row_adjust = 0
set displayport_max7456_col_adjust = 0
set displayport_max7456_row_adjust = 0
set displayport_max7456_inv = OFF
set displayport_max7456_blk = 0
set displayport_max7456_wht = 2
set esc_sensor_halfduplex = OFF
set led_inversion = 0
set dashboard_i2c_bus = 1
set dashboard_i2c_addr = 60
set camera_control_mode = HARDWARE_PWM
set camera_control_ref_voltage = 330
set camera_control_key_delay = 180
set camera_control_internal_resistance = 470
set pinio_config = 1,1,1,1
set pinio_box = 255,255,255,255

DieHertz, 2018-03-04:
Raise `moron_threshold`

NitroNAF, 2018-03-04:
Thank you. Setting it to 150 fixed it. Do you suggest a lower value?

DieHertz, 2018-03-06:
@NitroNAF May I ask you to set `moron_threshold` back to default and record a log with `debug_mode = gyro`?
I'd really appreciate it. I have some ideas how to get rid of this behaviour but need data.