betaflight/betaflight #5378: Betaflight 3.3.0 stable OSD element "Disarmed" bug

raimokr, 2018-03-04:
The OSD element "Disarmed" is covering other elements during it's not active state (armed state) so that these alements are fully or partially not visible. Please take a look at this DVR here (invisible textbox is covering corrhair and artificial horizon):
https://youtu.be/D0xgM8FBm8U

The same happens if I do the config via Betaflight Configurator GUI (in the menu as well as on the CLI via set var)

FC:
Holybro Kakute F4 AIO V1 (not the V2)

Firmware:
3.3.0 stable

etracer65, 2018-03-04:
@raimokr I don't have an OSD to test on at the moment but can you try the attached firmware to see if it fixes the problem?

[betaflight_3.3.1_KAKUTEF4.hex.zip](https://github.com/betaflight/betaflight/files/1778526/betaflight_3.3.1_KAKUTEF4.hex.zip)

raimokr, 2018-03-05:
@etracer65 I flashed this hex file but still the same issue...
# version
# Betaflight / KAKUTEF4 (KTV1) 3.3.1 Mar  4 2018 / 08:41:17 (6b5ddab) MSP API: 1.37

closed 2018-03-12 by mikeller