betaflight/betaflight #5379: BF3.3 - SmartAudio on TBS Unify HV Race does not save VTX parameters anymore...

RipperDrone, 2018-03-04:
Did a new build with smartaudio connected to TX3. Seems I cannot save the parameters, however editing / changing them is working ok.

When trying to save, Taranis tells me 'Saving...', then 'Retrying...', then cancels w/o parameters saved.

Is there any particular settings I need to do in CLI to get it working? Or is it a new bug? :-/

RipperDrone, 2018-03-04:
seems related to https://github.com/betaflight/betaflight-tx-lua-scripts/issues/104 since I have R-XSR lates FPort beta firmware installed, and maybe its about timing glitches in FPort transmission...

Gozzz, 2018-03-04:
On my BFF3 target running 3.3 stable, R-XSR (180228 FCC beta f.port firmware), Unify Pro Race HV, Taranis (otx 2.2 w/lua 1.0.1) I get similar behavior but the settings are applied. When saving changes I get "saving".. "saving".. "retrying", then the page reloads with the updated settings.

I also get the same behavior on a test setup with same hardware/firmware parameters, only difference being it has a Unify Pro Nano hooked up to it.

RipperDrone, 2018-03-04:
I am getting the same sequence as you described, and the Taranis screen shows the altered settings. However, if I reboot the FC and Taranis after that, settings are lost. So it seems they are just mirrored to the Taranis screen, but saving is not successful.

RipperDrone, 2018-03-04:
HAve done some more tests: It seems that both filters page and VTX page cannot be saved. PID page CAN be saved, though. I'm puzzled...

Gozzz, 2018-03-04:
Mine both save after reboot. Only difference I can think of is I'm not using the inverter hack on the RX, I'm sending the f.port right from the connector. 

What FC are you using?

RipperDrone, 2018-03-04:
FC = DYS F4 Pro V2. How are you running FPort without hardware mod of RX? You have an inverted signal TX pin on your FC then? Or are u running it over SoftSerial?

ijustwannafly, 2018-03-04:
I went back to the old lua script that only had pid, rates and vtx pages because the "latest" scripts with more menus is a pita with otx 2.2.1.

I had the retrying no save issue on the latest scripts with otx 2.2.1.  The old script gives me no issue.

RipperDrone, 2018-03-04:
which one is the 'old' script as per your definition?

ijustwannafly, 2018-03-04:
I will have to grab it when I get home.

Gozzz, 2018-03-04:
@RipperDrone I didn't think the inverter hack was needed when using f.port. From reading here at least- https://oscarliang.com/setup-frsky-fport/

I'm running it this way on two targets (BFF3 & CLR4). 
Using (UART 1) on the BFF3 with the following diff- 
serial 0 64 115200 57600 0 115200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set tlm_halfduplex = OFF

And on the CLR4 it is going straight to the Telem pad (UART 3) with the following diff-
serial 2 64 115200 57600 0 115200
set serialrx_provider = FPORT

Can you post a diff?

The old script I would imagine is v1.0 on the releases page- https://github.com/betaflight/betaflight-tx-lua-scripts/releases

RipperDrone, 2018-03-04:
@Gozzz the script does NOT work on a multitude of targets w/o hardware-hacked RX since it needs an inverted TX singnal handling which most of the time is not provided by the FC (it only has the RX inverted mostly).

RipperDrone, 2018-03-04:
Excerpt from diff:

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

set serialrx_provider = FPORT
set serialrx_halfduplex = ON

Anything else you are looking at? (I can send more detail if needed :-))

mboarman, 2018-03-04:
For what it's worth, I'm having the same issues without FPort. I'm using BetaFlight 3.3 on a BetaFlight F3 Flight controller and XSR (invert hacked) receiver. When attempting to switch the VTX power it gives me a Retrying message and than drops back to the original menu. However, upon a reboot of the Flight Controller, it shows the VTX power changes were never saved. I also tried this with an SPRacing F3 and a X4R-SB receiver with the same results. I flashed back to BetaFlight 3.1.7 and everything started working again?

Gozzz, 2018-03-04:
@RipperDrone So you are using uart 1 for f.port, do you have the f.port wire hooked up to the sbus pad on your board? Also are you getting *full* telemetry back to your radio (specifically looking for an active vbat reading?).

RipperDrone, 2018-03-04:
@Gozzz yes, I am using TX1 pad of UART1 for FPort. It's NOT hooked up to the SBUS pad (which would be RX1), but to TX1, explicitly. And yes, I am getting FULL telemetry back to my Taranis X9D+, namely VFAS (=VBat), A4 (=cell voltage), SPd, Hdg, ... so definitely telemetry is working. IT's just about the .lua script. Tried both the 1.0.1 lua and the 1.0.2 (Kalman edition) versions. Running BF3.3, not tried older versions yet like @mboarman.

RipperDrone, 2018-03-04:
no worries, it seems @mboarman and I have reproduced the same issue on different FCs, with and w/o FPort. So it seems related to BF3.3 and the .lua script...

ijustwannafly, 2018-03-04:
@RipperDrone 

@Gozzz posted the correct link to the lua I am using.

Forgot to mention, I had the issue on bf 3.2.5 non fport rx.

I started getting the issues with otx2.2.1 and the latest lua scripts.

etracer65, 2018-03-04:
Somebody mentioned on the Slack channel that they had some kind of multi-protocol module installed in their module bay and that was interfering with the Lua script.  Long-shot, but a FYI.

mboarman, 2018-03-05:
@etracer65 thanks for the suggestion, but I don't have any module nor did I flash the multi option in my otx 2.2.1 build.

RipperDrone, 2018-03-05:
No module here as well...

Bahecuk, 2018-03-05:
Having same issue with BF 3.3 (release) on airbot F4, my R-xsr is running through f.port connected to uninverted pad on RX and normal (not sbus) uart TX pin on FC. My Tx is q x7 with latest oTX and LUA script. Telemetry is bugged both with R9M inserted in radio (just sitting there) and without a module.
Yesterday this exact quad and tx combo was working fine with XSR connected exactly the same way (f.port, same uninverted hack, same uart tx pad), but I swapped RX to R-xsr and it started going into "retrying" state after saving lua script page.

Saxin, 2018-03-06:
I have same behaviour on rxsr, connected to OBF3, doesnt happen on X4R - this looks like rxsr specific, Im not running Fport FW

mboarman, 2018-03-06:
@FPVbug can you confirm that it actually did save the change after unplugging the battery and restarting the script? I thought mine was changing also, but it appears after the retry it just defaulted to the change on screen but didn't save it to the VTX. Just curious.

mboarman, 2018-03-07:
@Saxin Strange because on my X4R-SB I have the same problems when trying it with a SP3Racing board (no inverter hack needed)

FPVbug, 2018-03-07:
@mboarman

kalosjann, 2018-03-11:
I'm having the same issue mentioned above. My scenario:
FC: Joshua Bardwell F4
Fw: Betaflight 3.3.0
VTX: TBS Unify pro hv
RX: XSR (using s.port)
Radio: Taranis X9D+ w/ OTX 2.2.1 and Lua script (1.0.1) (No heli, luac, eu)

All the pages are working fine, as well as the saving.
When it comes to the VTX page i can visualise and change parameters but, when i save i get only a bunch of '---'. The parameters are saved but, when I reboot the drone (un-plug/plug the battery) the setting are the old one. 

Not tried on the Betaflight 3.2, but seems that one of my friend is not having the same issue.

Gozzz, 2018-03-11:
I've noticed the band, channel and power are now listed when you do a diff in cli. Is it perhaps not updating this info to the most recent change when the FC is power cycled?

ethomas997, 2018-03-11:
As of Betaflight version 3.3.0, CLI settings to modify the VTX configuration are supported and are applied at startup, so you'll definitely want to take them into consideration when testing.  I just posted a doc page about them to the wiki:  https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings

closed 2018-05-23 by stale[bot]