betaflight/betaflight #5380: Spektrum Satellite always in bind mode when powerd by lipo

Gutzebutze, 2018-03-04:
I've a Super_S F4 2S FC and a Super_S BS06D 4in1 Blheli_S 6A ESC. I'm connecting a LemonSatRx to it and i am binding via the CLI part of the Configuration GUI.

After activating Serial RX mode for UART1 in the Ports screen and then selecting "Serial-based receiver/SPEKTRUM2048"  in the Configuration screen, i use TAER1234 in the Reciever screen. I saved the done changes step by step and rebooted the FC if neccesary. I then run this in the CLI to initiate bind mode:

set spektrum_sat_bind=7
set spektrum_sat_bind_autoreset=ON
save

I then disconnect the usb-cable, and reconnect it to restart my FC.

This works fine, and the red light starts to flash.
I then bind my radio (DSMX/22ms), and the light goes solid red. I can then go into the Receiver screen and my sticks make the correct items move.

If I then restart my FC vie USB everthing works like it should, the RX stays bound. 

However, it goes direct into bind mode again when powered by lipo only. If i unplug the lipo and plug in the usb-cable the rx recognizes the tx again, everything works. 

If i plug in usb, turn on the tx (light stays solid), then plug in lipo and unplug the lipo ->quad can be armed and fly`s. 

If i try to bind the rx to the tx after plugging in the lipo the bind failes.

I started running BF 3.1.7 (Stable) and I've just compiled 3.2.5 from Git  and tried that too, but neither release the FC from bind mode (whitch is`nt working) when lipo is used to power up the FC.

githubDLG, 2018-03-06:
hi, guys   , how about try this:
1.after bind complete, type    **set spektrum_sat_bind=0**  in to the CLI and then save , restart the FC
this is a way to disable bind feature of the FC , maybe help, just try it man~

Gutzebutze, 2018-03-06:
Thank you for this info, but i tried it this way allready. Makes no difference. 
Thx

jflyper, 2018-03-07:
@Gutzebutze There are cases that spektrum receivers autonomously going into bind mode without proper bind pulses generated by flight controller.

What does `RX_BIND` line of CLI command `resource list` say?

Can you issue `resource RX_BIND none` (then `save`) and see if the issue persists?

Gutzebutze, 2018-03-07:
This issue can be closed. There was no issue at all, it was just me. The Rx was to close to the ESC. After giving it a "shild" of 0,2 mm carbonplate it worked out.

I am sorry for bothering you guy. And a BIG thank you!

closed 2018-03-07 by Gutzebutze