betaflight/betaflight #5383: Matek f405 aio ctr ibus not working

eta232, 2018-03-04:
I wired my a8s rx to rx2 pin and set up in uart2 rx. In configurator i set serial rx and ibus but no movements oft Sticks shown in channel map. This happend after flashing Bf Form 3.2 to 3.3 with chip erase. 
If i solder ppm cable to rx2 pin it works. Tried also different rx pads and different UART ports but nothing works with the ibus wire. Can share diff info tomorrow, but maybe some suggestions until now?

etracer65, 2018-03-05:
Check the `serialrx_inverted` CLI parameter.  You may need to switch its value.  From the release notes:

> The parameter sbus_inversion has been changed into serialrx_inverted, and the way it is applied has been changed as follows: It now applies to all RX protocols, not just SBus, and instead of switching the UART to normal when off and inverted when on, 'off' now means that the port is set to whatever the default is for the selected protocol (i.e. inverted for SBus, not inverted for SUMD), and 'on' means that the port is inverted from default (i.e. not inverted for SBus, inverted for SUMD)

McGiverGim, 2018-03-05:
The CTR has two pads for the same UART, the SBus (inverted) and the RX2 (uninverted) in theory. So in the RX2 it must work. If it does not work for any of the RX pads, then:
- Share the `diff` to see if something is wrong
- I suppose the A8S is bind correctly to the TX (you said the PPM worked).

ttjs82, 2018-03-05:
@eta232  I've the older Matek F405 AIO board wired with Flysky X6B receiver. Like @McGiverGim said there are two pads for the UART2. If you have it connected to the UART2 Rx pad it should work. Other thing to note is that if you have the positive power wire from the receiver going to any other pin than the 4V5 pad you'll have to connect the lipo to get the receiver powered. The 4V5 pad is the only one that gives power also with just usb connection.

I didn't have to change any CLI settings, but just in case you should check that the `serialrx_inverted = OFF` and `serialrx_halfduplex = OFF`. Quick way to see the current values of these settings is writing `get serial` in the CLI.

eta232, 2018-03-05:
Hi @all, thanks for your support. I had used flysky x6b before with same issue, i thought he's gone defect also changed to a8s. I've soldered power to 4v5 and signal to rx2. I will try later your suggestions and post results including diff and dumb results.

eta232, 2018-03-05:
Hi, here my diff output:
'''diff
# version
# Betaflight / MATEKF405 (MKF4) 3.2.2 Nov 11 2017 / 00:35:15 (77782e602) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = IBUS

# profile
profile 0

# rateprofile
rateprofile 0

eta232, 2018-03-05:
and here my dump file
[BTFL_cli_20180305_165404.txt](https://github.com/betaflight/betaflight/files/1781611/BTFL_cli_20180305_165404.txt)

eta232, 2018-03-05:
sry updated to bf 3.3 again, so here my dump file again:
[BTFL_cli_20180305_165404.txt](https://github.com/betaflight/betaflight/files/1781633/BTFL_cli_20180305_165404.txt)

and also diff again:
'''diff
# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.3.0 Mar  2 2018 / 03:52:04 (177472b4f) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature
feature CHANNEL_FORWARDING

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set serialrx_provider = IBUS

# profile
profile 0

# rateprofile
rateprofile 0

#

McGiverGim, 2018-03-05:
Maybe is a silly question, but, have you checked the serial in the ports tab? I don't see nothing about it in the diff

eta232, 2018-03-05:
[image]

eta232, 2018-03-05:
that was the issue!! now it works. Just find out why it doesnt arm, but i think i will find it.
Many thanks so far!!!

ttjs82, 2018-03-05:
That's nice to hear, well done. In the Betaflight 3.3.0 you can't arm the quad while you have usb connected and the Betaflight configurator open. It should arm without problems after you just set up the switch arming correctly and test the arming without props and using the lipo. Additionally, you can do "soft" testing of the arming on the modes tab after you have configured the aux switches the way you like. The arm mode line will "light up" when you switch the arming on but the props won't spin just as a safety thing.

etracer65, 2018-03-05:
You can arm in the configurator if you first go to the motors tab and change the safety switch.  NO PROPS!!

eta232, 2018-03-05:
Many thanks just wondering but now everything works fine!!!!!!

closed 2018-03-07 by etracer65

thowitz, 2019-09-21:
Hi, I had the same problem and I know the receiver is definitely in ibus mode, I'm using a foxeer f405 aio flight controller and a flysky x6b receiver with a flysky fs-i6s transmitter which is also in ibus mode, the receiver is connected to UART 2 on the FC and I've enabled it in betaflight.
Many thanks
Tim

etracer65, 2019-09-21:
@thowitz You're commenting on an issue that's been closed for 1.5 years. Issues are not the proper place to obtain configuration or hardware support.

Try:
Facebook group: https://www.facebook.com/groups/betaflightgroup/
Slack group: https://slack.betaflight.com
RC Groups: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread