betaflight/betaflight #5386: Magnetometer IST8310 not found

Ralfde, 2018-03-05:
Omnibus F7V2 Betaflight 3.3
IST8310 I2C Addresss like inav 0x0C
The magnetometer is not found

Ralfde, 2018-03-05:
Yes, please.

McGiverGim, 2018-03-05:
I've generated a file to test: https://drive.google.com/drive/folders/1lPa_x3hJF4CTfbA3KxEot4wPNM5ljfcI?usp=sharing

I've not the compass so I don't know if it will work.

Ralfde, 2018-03-06:
The magnetometer is still not found. Could you maybe check with the inav 1.9.0 RC2 code. This one does detect the IST8310 correctly.

McGiverGim, 2018-03-06:
Yes, I have took the code from iNav, but is not exactly the same, so I have make some changed. I will recheck the code.

Have you assigned the `resource` pins correctly if needed for your board?

Ralfde, 2018-03-06:
Isn't the I2C used automatically like at inav, when a magnetometer is choosen?

McGiverGim, 2018-03-06:
Some FC need to define which pins are the I2C, specially when shared with a UART.
The resource command will give you a list of what is configured.

McGiverGim, 2018-03-06:
Take a look:
https://github.com/betaflight/betaflight/blob/0430fcb67dfd87958357e50c834eb6957337b6ba/src/main/target/OMNIBUSF7/target.h#L127-L130

You must do something like this:
```
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
```
if you are using the UART3 pins for I2C.

Ralfde, 2018-03-06:
# resource
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource LED 1 E00
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource SPI_MOSI 4 E06
resource ESCSERIAL 1 A02
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource BARO_CS 1 A01
resource SDCARD_CS 1 E04
resource SDCARD_DETECT 1 E03

Do I need to set it to I2C 1, 2 or 3?

McGiverGim, 2018-03-06:
Depends where the compass is connected, if it is connected to the UART3, the I2C 2:
```
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
```

Ralfde, 2018-03-06:
I set it to I2C 2 now and the magnetometer is found. Great. Thanks a lot.
Could you please also make firmwares for the IST8310 also for the Omnibus F4 and Omnibus F4SD?

McGiverGim, 2018-03-06:
Yes, be sure that it works, check that there's not any I2C errors in the bottom bar. If you confirm that all is right, I will do the firmwares.

McGiverGim, 2018-03-06:
Generated firmwares, based on the latest master, from:
- OMNIBUSF7V2
- OMNIBUSF4SD
- OMNIBUSF4
in the same link than before.

McGiverGim, 2018-03-06:
@Ralfde I've modified the firmware again, with some changes asked by the developers to my code (https://github.com/betaflight/betaflight/pull/5397). If you downloaded them before, could you download and test some of them again to see if it works?

Thanks!

Ralfde, 2018-03-06:
I2C error 3, direct after starting. But then not increasing with Omnibus F7V2.
With Omnibus F4SD V3 no errors at all.

McGiverGim, 2018-03-06:
In Omnibus F7 V2, this is with the latest firmware or the first? With the first, did you see some I2C error?

McGiverGim, 2018-03-06:
And take a look to the sensors tab, you can see if the MAG is reading values or not when moving it.

Ralfde, 2018-03-06:
Was the new file with the 3 errors. Values are changing.
[image]

McGiverGim, 2018-03-06:
Did you have the old file? If you have it, please test it too see if the problem is with the latest changes or not. If you don't have it, I can generate it again.

Thanks!

Ralfde, 2018-03-06:
Yes, I have tested now also the old file. Exactly the same behaviour.

McGiverGim, 2018-03-06:
Ok, I will ask the other developers. 

If I'm not wrong:
- F7: shows some errors, but works
- F4: no errors, and works

I'm right?

Ralfde, 2018-03-06:
Yes. And the F7 does show the errors directly after starting it. The number does not increase. I concern more about the strange X-values at the moment.

McGiverGim, 2018-03-06:
Why you think they are strange?

Ralfde, 2018-03-06:
The Y and Z graph look normal and the values are <1.
The X Values are looking sometimes normal ,but most time 0 or >16, so the line is at 0 or on the upper limit of the graph, which I think is not right.

McGiverGim, 2018-03-06:
Did you execute the calibration?

Ralfde, 2018-03-06:
Yes

Ralfde, 2018-03-07:
Is there a debug mode for the magnetometer, so I could make some logfiles for you?

Ralfde, 2018-03-07:
[image]

McGiverGim, 2018-03-07:
The values seem between 0 and 1, I don't see a > 16. Am I missing something?

McGiverGim, 2018-03-07:
In theory the calibration must solve this. It calculates the maximum and minimum value for each axis, to know where the limits are. Maybe the "tethering loggin" tab (in english) can give some data. And a `get mag` in cli can show us the calibration values detected.

Ralfde, 2018-03-07:
# get mag
align_mag = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

mag_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

mag_i2c_device = 2
Allowed range: 0 - 4

mag_i2c_address = 0
Allowed range: 0 - 119

mag_spi_device = 0
Allowed range: 0 - 4

mag_hardware = AUTO
Allowed values: AUTO, NONE, HMC5883, AK8975, AK8963, QMC5883, IST8310

mag_declination = 2300
Allowed range: -18000 - 18000

mag_calibration = -1,31,-40
Array length: 3

Ralfde, 2018-03-07:
[image]

Ralfde, 2018-03-07:
Here you see an X value > 26

Ralfde, 2018-03-07:
How to upload the log file?
[image]

Ralfde, 2018-03-07:
[BTFL_log_20180307_104225.zip](https://github.com/betaflight/betaflight/files/1788598/BTFL_log_20180307_104225.zip)

McGiverGim, 2018-03-07:
The X calibration value is too low, only -1, so it seems something is wrong with the calibration. Usually the three values must have more or less the same magnitude.
Have you tried to calibrate it again? The value is the same? For both, F4 and F7?

Ralfde, 2018-03-07:
after calibrating again: mag_calibration = -2,33,-30

Ralfde, 2018-03-07:
F4V3 board Omnibus F4SD firmware:
mag_calibration = -1,33,-31

McGiverGim, 2018-03-07:
Strange, the calibration is more or less the same for all of them.  I will check your data and the code again.

McGiverGim, 2018-03-07:
Seeing your data, the calibration for X must be near -30 (if the data has been collected moving the compass in all the directions 360º).

I don't know why it is not calibrating right, I will recheck the code, but if I'm not wrong, you can test with:
`set mag_calibration = -30,33,-40`

This is only a patch to make usable the compass, but I need to be sure where the problem is and correct it.

McGiverGim, 2018-03-07:
I've rechecked the code and all is right. All the axis (X, Y, Z) use the same code for the calibration, and the values readed by the MAG (between 0 and 30 in your data, for example) can't end with a value of -1 or -2 in the calibration.
Do you have other model of MAG to test with this firmware to see if it produces the same result? I'm starting to thinking that the problem is not with the IST8310 MAG code, maybe in other general code affecting other models too. And the one I have is dead and I'm waiting for a replacement.

Ralfde, 2018-03-07:
I did set mag_calibration = -30,33,-40

But still the same behaviour for the X-axle.
[image]

Ralfde, 2018-03-07:
I will try now with an HMC5983. Need a while for wire it.

Ralfde, 2018-03-07:
The HMC5983 show correct values, but strang calibration values: 
mag_calibration = 10,-74,6
second try: mag_calibration = 1,-95,2
[image]

McGiverGim, 2018-03-07:
Thanks for your tests. At this moment I have no idea what can be wrong. I think I need some other developer with access to the hardware. I will ask.

Ralfde, 2018-03-08:
I checked the values with the Omnibus SD F4V3 board with inav and the calibration values X=-28 and the values looks fine on this one with inav 1.9.0

McGiverGim, 2018-03-08:
Thanks for your test. Seems clear that something is wrong with the calibration in Betaflight with my code. I will try to take a look to it, but I don't know if will get to something.

Ralfde, 2018-03-13:
I try to organize samples for you. Will take a while, because I don't have enough right now. But you may tell me where to send it, when I get them.

McGiverGim, 2018-03-13:
You can contact me in the official slack chat of Betaflight.

Ralfde, 2018-03-13:
Sorry, I don't know how to use this slack chat. 
@martinbudden : Could you please give my mail address to McGiverGim?

McGiverGim, 2018-03-14:
@Ralfde My mag QMC5883 has arrived, and it presents some strange values too. In the sensors tab, the X is more or less correct, but the Y and Z are zero. Maybe the origin of the problem is at the same place. I will try to solve this problem first to see if it solves yours too.

Ralfde, 2018-03-14:
Ok. I will wait for the result.

McGiverGim, 2018-03-14:
@Ralfde it seems that finally there is a problem in the code of Betaflight itself, not in my code for the IST8310 driver. If you take a look to your HMC5883 capture, the values are only positives, but the MAG must show values between -1 and +1.
I've pushed a PR that tries to solve it: https://github.com/betaflight/betaflight/pull/5451 but at least for my QMC5883 this not seems the final solution.
I've built new firmwares for you, if you want to test it, but as I said, I don't know if this will be the final solution.

McGiverGim, 2018-03-14:
@Ralfde it seems that the problem is with the configurator, that does not know how to scale the values, it uses 1090 as maximum value for all models of MAG and this is not true. So my latest firmwares are correct and must show, at least, positive and negative values in the configurator. Can you test it?

Ralfde, 2018-03-14:
Sure. Please send me the links. Do I need to download also a changed configurator?

McGiverGim, 2018-03-14:
The link is the same as the latest time: https://drive.google.com/drive/folders/1lPa_x3hJF4CTfbA3KxEot4wPNM5ljfcI?usp=sharing
No, there is no new version. By the moment the configurator will have the bug.

Ralfde, 2018-03-14:
The files are from 05. and 06.03.2018. Are these the correct files?
And how can I check the correct values? 
Does the configurator show values between +-1, then?
Or maybe with a blackbox logfile?

McGiverGim, 2018-03-14:
I see the modified today. Verify it. The configurator maybe will show values bigger than 1, but must show from -X to +X for each axis. In the earlier version it showed only positive numbers.

McGiverGim, 2018-03-14:
I'm wrong, give me 1 hour and I will verify the files, please.

McGiverGim, 2018-03-14:
I've verified them. The date of creation is when you say, but they have been updated today. So they are correct.

Ralfde, 2018-03-14:
Yes, it does show now positive and negativ values. Just too big values for X:
[image]

[66 more replies]

closed 2018-05-07 by Ralfde