betaflight/betaflight #5387: "set report_cell_voltage on" OMNIBUS target not working

Nico9n, 2018-03-05:
On OMNIBUS target "set report_cell_voltage on" ends in "Invalid name". Tested on Betaflight 3.2.x and 3.3

etracer65, 2018-03-05:
After looking at the code it appears that `report_cell_voltage` is part of IBUS telemetry which is only included for F4 and F7 processors.

closed 2018-03-07 by etracer65

ctzsnooze, 2018-03-11:
@etracer65 this should be re-opened, it's a bug.  I've used report_cell_voltage on my Motolab F3 boards for years.  Now in 3.4 it has stopped working on F3 boards but not F4 boards.  That's wrong, it's a feature of the radio, not the board, it's built in to FrSky telemetry regardless of board.

etracer65, 2018-03-11:
@ctzsnooze Settings don't work that way.  They are not contextually available based on other settings.  You can set lots of options for things you may not be using.  For example, you may not be using led strip (`FEATURE LEDSTRIP`) but you can still configure the LEDs in the CLI.

Contextual parameters is the realm of the GUI and there it is appropriate to "hide" settings that are not applicable based on other settings.  Again see the led strip example.

If `report_cell_voltage` was exposed in the Configurator GUI then yes it would be appropriate to hide it if iBus wasn't selected.

ctzsnooze, 2018-03-11:
OK wondered about how the CLI was managed: defines can change what is shown there, but user settings can't.  

It makes sense that generally it was removed from the CLI to force FrSky pilots to use A4 so that FAS would be total voltage always.  This will annoy some people.