betaflight/betaflight #5388: BF 3.3 does not arm

Ralfde, 2018-03-05:
Omnibus F4V5
SUMD on UART1
does not arm disarm flags 17,18

BF 3.3

did work with 3.2.5

etracer65, 2018-03-05:
Arming disabled flags can be read about here: https://github.com/betaflight/betaflight/wiki/Arming-Sequence-&-Safety

17 = MSP (connected to configurator)
18 = Arm switch in unsafe position.

Please read the release notes:  https://github.com/betaflight/betaflight/releases/latest

> Arming is now disabled when the flight controller is connected to the Betaflight configurator. This was added in order to keep users from accidentally arming their craft when testing their RX / mode switches with a battery connected. In order to enable arming for bench testing, go to the 'Motors' tab and enable the 'Motor test mode' switch (REMOVE ALL PROPELLERS FIRST);

Ralfde, 2018-03-06:
Thanks a lot. Seem to work right.

closed 2018-03-06 by Ralfde