betaflight/betaflight #5390: BF3.3 - SP F3 - FPort

DerElch87, 2018-03-05:
Hi there,
I got a strange issue with trying FPort.
BF is not recognizing the RX when attached to UART 3.
On top of that the Controller hangs in boot loop when powered on only via USB, so no LiPo or external power. Only when I power the FC before with an external supply it is possible to connect via USB.
`serial 2 64 115200 57600 0 115200
set serialrx_provider = FPORT
set serialrx_halfduplex = OFF
set serialrx_inverted = ON`

The same on UART 2 is working without problems. RX gets recognized and controller can be powered without external power.
`serial 1 64 115200 57600 0 115200
set serialrx_provider = FPORT
set serialrx_halfduplex = OFF
set serialrx_inverted = ON`

Strange is that before (BF 3.1) it could be powered via USB on UART 3 without problems.

I'm kind of lost here.
Not a problem for me as I can also use UART 2, but probably there is a bug somewhere.
Only wanted to mention.

BTW:
Thanks to all developers for the great work. I really highly appreciate it.

Controller: SP F3 Racing
Betaflight 3.3
RX:  XSR with FPort Firmware

ttjs82, 2018-03-05:
According to the manual http://seriouslypro.com/files/SPRacingF3-Manual-latest.pdf UART3 (page 4) is only outputting 3.3 volts and the XSR need 4 volts. You might wan't to check and verify this with a multimeter.

ttjs82, 2018-03-05:
@DerElch87 my friend said that he has the SP Racing flight controller working with XSR reveiver on the UART 3 with F.Port protocol, he has it soldered on the UART 3 solder pads. The smartport pin from the receiver needs to be connected to the TX pad on the UART. He told that at least on his board the usb alone won't power the receiver and that the flight controller needs to be powered through 5V BEC to forward power to the receiver also.

DerElch87, 2018-03-06:
I think I explained it too difficult.
Let's split it into two different problems.

Problem 1
The FC is not booting when the XSR is attached to UART3 and ONLY the FC gets powered via USB.
Only setting I changed is FPort.
BEFORE (BF 3.1) there was never a problem like that - same hardware.

Problem 2
When I first boot the FC and RX via external power (5V) and I connect in Betaflight there is no signal recognized by the FC.
When I only change the UART from 3 to 2 everything works. No other settings changed.
Even booting works like before.

For me that's not a problem as I found a solution for myself, only wanted to mention it.

Should I close the issue?

ttjs82, 2018-03-06:
Nice that you broke down the issue more, it helps to take a catch of this better. This is surely an interesting issue. I suppose that you were using SBUS when using BF 3.1 as it didn't have the F.Port support yet? To use SBUS connection you would had to use different pins on the receiver and the flight controller compared to what you would need to use the F.Port protocol properly. So are you saying that you didn't even change the wiring of the flight controller and receiver (other than moving them between UART2 & UART3)? In all the cases where I've been now using the FPort protocol I've had to use positive, negative and smartport pads on the receiver with the smartport connected to some UART TX pad on the flight controller side.

It's possible that the radio controls work properly with F.Port even without changing the wiring from the way they were with SBUS. To make sure that the Fport connection itself is working correctly you should check that you are seeing all the telemetry data on your radio controller if it just has the telemetry features. I'm not sure if the booting and rx recognition issues you are seeing could be caused by potentially having problems with the wiring. But I'd like to at least rule out any issues with the wiring. Picture of the connection between the flight controller and the receiver would help a lot.

severteka, 2018-03-11:
My SP F3 with BF3.3 work great with XSR (F.Port firmware) connected to TX3 (UART3). But,  unfortunately, on BF3.3 current sensor work not properly. With BF 3.2.2 was all OK.