betaflight/betaflight #5392: Feature: Allow rearm when throttle not low (accidentally disarmed in flight)

jj-uk, 2018-03-05:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

If i accidentally disarm mid-flight, i have to lower the throttle then rearm. It would be useful and faster if BF would allow me to rearm instantly, ignoring the throttle level, if the rearm is done within, say 2-3 seconds of the disarm.

etracer65, 2018-03-07:
From a safety perspective I really don't think this is a good idea.  Arming with non-zero throttle is only going to end poorly.  The point is that you shouldn't be disarming accidentally while in the air anyway and you need to fix that issue.  Maybe change which switch you use for arming so you don't accidentally hit it.  

Introducing a serious safety issue for everybody is not the best way to solve your problem.

jj-uk, 2018-03-08:
Arming with non-zero throttle will end badly, yes. But my suggestion was to allow that condition ONLY if the craft was armed less than a few seconds ago.

etracer65, 2018-03-08:
That doesn't really address the safety concerns.  What if you had just landed, disarmed and then accidentally bumped your arming switch?  Not good either way because the motors will start spinning, but at high throttle it would be really bad.

You're real request to allow non-zero throttle re-arming _**only while still in flight**_.  The problem is detecting an "in flight" condition is very difficult and not reliable enough for a safety issue such as this.

I still stand by my suggestion that you need to address the reason you're accidentally disarming in flight and/or train yourself to lower the throttle before attempting to re-arm.

I've re-armed in flight many times after intentional disarming on the onset of what I thought was going to be a crash (like a gate clip at speed).  The natural response in the crash is to lower the throttle and disarm.  Then upon realizing that I can possibly save it I simply need to re-arm because the throttle is already low.

AndersHoglund, 2018-03-08:
This is a dangerous proposal. Even with timing constraints etc etc. You must consider Murphy:

> Anything that can go wrong will go wrong.

Even O'Tool might be applicable.

tavdog, 2018-03-09:
This can be addressed in your radio. Simply disallow your disarm switch if your throttle is not low.

AndersHoglund, 2018-03-09:
@tavdog You are inviting Murphy again. Do not do this.