betaflight/betaflight #5395: failed to open serial port message. I changed some setting and now I cannot get betaflight to connect. Is there a way to "go back" to what I had on there? I was trying to see if another port would allow me to connect. Newbie on betaflight. It's a Hak rc micro f4 fc

Supersix4, 2018-03-06:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

closed 2018-03-07 by jflyper

Supersix4, 2018-03-07:
It's a betaflight issue.  Why did you close my problem?? You are providing zero assistance  I need help with betaflight programming.

fujin, 2018-03-07:
@Supersix4 The giant section of the new issue template states the following:

> If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

I suggest you DFU flash your FC (jump the BOOT0 pins, or press BOOT button), and re-flash firmware with full chip erase. This will wipe any misconfiguration.

Alternatively, please use RCGroups or Slack for support. GitHub is for firmware issues.

jflyper, 2018-03-07:
This is not a place to assist users having configuration problems. Please read the pre-formatted text (which you didn't even care to read, follow and remove).

Supersix4, 2018-03-07:
I did the dfu. It's fixed. Now to muddle thru all the settings. Thx!

szafranski, 2019-05-25:
Try upgrading firmware with "Full chip erase" and "Manual baud rate" set to 115200 - even if it's your default. Cheers