betaflight/betaflight #5398: Runcam Split record starts let the ESC´s reset

Gregmaan, 2018-03-06:
Hi,
when I power up my quad with 3.3 and above and I want to start recording of the Split the ESC´s reboot.
This happens only the first time I wan´t to start recording via RC. ESC´s have firmware 32.31.
Second time I command recording all is fine.
With 3.2x I never had this issue and going back solved it.
```
# Betaflight / MATEKF405 (MKF4) 3.3.0 Mar  2 2018 / 03:52:04 (177472b4f) MSP API: 1.37

# name
name QAV-R

# resources
resource MOTOR 6 NONE
resource SERIAL_TX 11 A08

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE
feature ESC_SENSOR
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 2 16384 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 6,12::C:2
led 1 6,11::C:2
led 2 6,10::CO:2
led 3 7,10::CO:2
led 4 7,11::C:2
led 5 7,12::C:2
led 6 8,12::C:2
led 7 8,11::C:2
led 8 8,10::CO:2

# color

# mode_color

# aux
aux 0 0 2 1300 2100 0
aux 1 2 0 1300 1700 0
aux 2 13 1 1650 2100 0
aux 3 26 2 1725 2100 0
aux 4 30 0 1750 2100 0
aux 5 32 4 900 1300 0
aux 6 33 4 1700 2100 0
aux 7 34 5 1700 2100 0
aux 8 35 1 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 s 1000

# master
set gyro_lowpass_hz = 0
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set gyro_stage2_lowpass_hz = 85
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set fpv_mix_degrees = 20
set max_aux_channels = 6
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set bat_capacity = 1200
set vbat_scale = 109
set ibata_scale = 183
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 5
set pid_process_denom = 1
set frsky_vfas_precision = 1
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_vbat_pos = 2519
set osd_rssi_pos = 2074
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2551
set osd_flymode_pos = 2497
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 2250
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2060
set osd_mah_drawn_pos = 2540
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2476
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2529
set osd_esc_rpm_pos = 2081
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set debug_mode = NOTCH
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732
set vcd_video_system = 1

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass = 95
set dterm_notch_hz = 245
set dterm_notch_cutoff = 150
set vbat_pid_gain = ON
set dterm_setpoint_weight = 77
set p_pitch = 48
set i_pitch = 52
set d_pitch = 23
set p_roll = 37
set d_roll = 19
set p_yaw = 60

# rateprofile
rateprofile 0

set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 105
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
```
I think it´s a DShot related issue and I don´t think it´s critical.
Cheers
Gregor

jflyper, 2018-03-07:
@Gregmaan 
Is this the correct description of your issue?

1. With 3.3,
2. when trying to start recording of RC Split,
3. ESCs reboot (with 3 + 2 tones? ), (unknown RC Split status)
4. but trying to start the recording for the second time
5. ESCs don't reboot and RC Split starts recording.

RoadyFPV, 2018-03-20:
Confirmed this issue at brainfpv radix with runcam split 2

mu2004, 2018-03-26:
I can also confirm having this exact same behaviour on BF 3.3.0.

ghostface, 2018-04-08:
I got the esc reseting but no other reaction on using the buttons.
Upgrading the firmware with the latest from the website solved it for me.

phillipsj70, 2018-04-08:
I experiencing similar issue. Also radix with runcam split. This reset happens while flying causing drone to drop and video to go black.

Gregmaan, 2018-04-09:
> I got the esc reseting but no other reaction on using the buttons. Upgrading the firmware with the latest from the website solved it for me. <

I cannot confirm that updating anything solves the issue

- Betaflight 3.3.1

- BLHeli32_32.4

- Latest runcam split firmware
Still the same issue.
I would recommend to start recording first, before arming, let the esc´s reset, and after that for me all works normal.

ghostface, 2018-04-09:
I maybe wasn't very precise with my feedback - ESCs still reset but the switches now work.
Also for anyone checking their split mini firmware version, the old and new version has the same version number - so when in doubt just download it from the website and flash it via the sdcard.

David-VG, 2018-04-13:
Same problem here. Just a video:

https://youtu.be/F012PgKZIHg

bullitproof, 2018-04-17:
Same here

- Betaflight 3.3.1

- BLHeli 32.4

- RunCam Split 2 with newest firmware

SvenHenning, 2018-04-20:
I can confirm this bug, I have the same configuration as bullitproof.

therealjonnyd, 2018-04-20:
Exact same problem, OmnibusF4SD target and Split Mini. Same behavior on LUXF4OSD.

RobbieHood, 2018-04-22:
Confirmed!

JonnyPhenomenon, 2018-04-24:
I noticed the same issue with mine.  
Runcam split V2, latest firmware
Betaflight F4, with 3.3.1.  

I have the split configured to start recording automatically on boot.  but some times I want to stop recording while I get set up to fly, so if I flip the recording switch and the ESC's reboot, but the switch doesn't actually stop the recording until I flip it a second time.

I have yet to try this while armed.  I had already been using a logical switch on my TX to disable my camera control switches while armed so I couldn't hit one in flight and lose video.   
I could only imagine what would happen if someone decided to start recording while in flight...

wesk-ms, 2018-04-25:
I've got a variant of the issue.  I'm using the new Runcam camera control board connected to a Foxeer Monster V2.

Repro:
Center throttle
Yaw right.
ESCs reset (2 beeps)
Unable to get camera settings menu to display

Matek F722
Betaflight 3.4.0 build #787

realpastaman, 2018-05-03:
Same issue as Gregmaan BF 3.3.1 JB AIO f4 Testing on the bench it doesn't look like it could cause a esc reboot in flight.

ghostface, 2018-05-04:
I have to check again but I am pretty sure while pid tuning I noticed that if you go into the OSD and hit save+reboot. And then use the runcam split record toggle switch the esc reboot doesn't happen and it works straight away.

Maybe someone else can confirm?

realpastaman, 2018-05-05:
RoadyFPV what is the firmware version ? 

On 4th May runcam said - 

RunCam  Support (Customer Services)   May  4, 16:43 CST  Thanks  for your feedback. We have noticed about it and will try our best to  find a cure for it as early as possible.   Regards,Stella  T.RunCam Customer Service | RunCam  Support (Customer Services)   May  4, 16:43 CST  Thanks  for your feedback. We have noticed about it and will try our best to  find a cure for it as early as possible.   Regards,Stella  T.RunCam Customer Service
-- | --
RunCam  Support (Customer Services)   May  4, 16:43 CST  Thanks  for your feedback. We have noticed about it and will try our best to  find a cure for it as early as possible.   Regards,Stella  T.RunCam Customer Service

RoadyFPV, 2018-05-05:
FW is 2.1.0

David-VG, 2018-05-05:
Nice know that!

Can we get this fw?

RoadyFPV, 2018-05-05:
[RCSplit.zip](https://github.com/betaflight/betaflight/files/1977078/RCSplit.zip)

Here is the file. i dont know is a final version. take care

David-VG, 2018-05-05:
Thanks, will try as soon as i get a new pigtail (hehehehe).

realpastaman, 2018-05-05:
Thanks RoadyFPV. Anyone else going to try this out :)

realpastaman, 2018-05-07:
From Runcam support
RunCam  Support (Customer Services)   May  7, 18:33 CST  Thanks  for getting back. We will release version 2.1.0 soon. You may keep eye  on.     Regards,Stella

Gregmaan, 2018-05-12:
Did someone test the 2.1 FW?

RoadyFPV, 2018-05-22:
My TX is back from Service. And i test again but whats that? the esc reset again by switching WIFI toggle and again by start record? 
Can someone confirm that 2.1 again restart esc?

David-VG, 2018-05-29:
> My TX is back from Service. And i test again but whats that? the esc reset again by switching WIFI toggle and again by start record?
>Can someone confirm that 2.1 again restart esc?

I still get resets with 2.1 :(

ghostface, 2018-05-30:
Hmm I checked the commits between 3.2 and 3.3 and the only directly runcam related changes I found were this: https://github.com/betaflight/betaflight/commit/2d84608daa9cbdf7f37e0716374bd89573bd58f3#diff-e293e0889da47c994ca5e3c692180810 I don't really see how this would break anything tho. I might try to remove it and test if it solves the issue.

It's possible that some other change is responsible for the behaviour tho, but checking the obvious commits first might be worth a try.

alancastell, 2018-05-31:
Same issue on 2 different setups.  One a Maetek F405CTR on 3.3 and a Holybro Kakute F4 with 3.3.  Both setups running BL32 ESC's and Split 2 cameras.  Didn't happen on my prior versions of BF.

romecode, 2018-06-01:
Can confirm issue on BF 3.3.2
Omnibus F4 Pro Corner Wraith 35A ESC 32.4
Latest FW from RunCam (according to updater)

If i go into the OSD menu for RunCam, and exit out, the buttons seem to work properly afterwards.

ghostface, 2018-06-04:
Tried removing https://github.com/betaflight/betaflight/commit/2d84608daa9cbdf7f37e0716374bd89573bd58f3#diff-e293e0889da47c994ca5e3c692180810 but it made no difference.

ghostface, 2018-06-04:
Honestly I didn't check the code in depth and I am not very familiar with the betaflight code base. 
It was just worth a quick try as I was comparing the changes between 3.2 and 3.3 and it seemed simple enough to just give it a shot, and since I did I wanted to report my findings :)

On another note I can add that the esc rebooting also happens when using soft serial, second try works fine just as reported by the others.

Gregmaan, 2018-06-05:
Maybe the issue is related to the DMA burst feature?
I am only a "user" but I think there was a change from 3.2.x to 3.3.x

azolyoung, 2018-06-05:
@mikeller 
Hi, I'm sorry for the late reply, I've got your message. I also found the issue was caused by the non-asynchronous  communication, and I'm re-writing the rcdevice protocol implementation to async style, sorry again, there are several things that need to be dealt with these days,  I need do it one by one, and now, this issues is my next job:)

mikeller, 2018-06-05:
All good, @azolyoung, good do know that you are looking into it!

delike, 2018-06-15:
Any news on this? :)

ghostface, 2018-06-15:
It's no longer happening for me on betaflight 3.4 rc1 using soft serial.

delike, 2018-06-17:
today i have tried Betaflight 3.4.0 too, but now now the recording wont start anyway..
Configuration is the same as on 3.3 ( i have compared it by screenshots). but nothing is happen when i push the switch. :(

Gregmaan, 2018-06-17:
On 3.4 RC1 I can´t control the split anymore. No more esc reboots, but also no function.

delike, 2018-06-17:
i'am also on Matek405 Platform..

ghostface, 2018-06-17:
This is odd - at first it was working fine on 3.4 on my split mini via soft serial.
Now it's not working until I do a reboot via osd (save exit + reboot)

alancastell, 2018-06-18:
Have a few other flight controllers  and don't want to update till I know this is fixed.

JManDoo, 2018-06-22:
I have this issue on the OMNIBUSF4SD Target as well on 3.4.0-RC2

azolyoung, 2018-06-23:
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4SD.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF405.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_CLRACINGF4.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_BETAFLIGHTF4.hex

Hi, I've submit a PR(#6196) to BF to solve this problem, I was test the PR in my quad that also has this problem, and it work fine. There are some firmwares that solve this problem. If you have time, please help test it. Thank you!

Gregmaan, 2018-06-24:
@azolyoung I would like to test the MatekF405 today but the links are not useable for me. Can you please update your links
Thanks!

JManDoo, 2018-06-24:
I get access denied too

Jason

delike, 2018-06-24:
Right-Click on the Link and click "Save Link to..." ;)

Gregmaan, 2018-06-24:
I think you we should not flash a 1kb file to our boards ;-)

azolyoung, 2018-06-25:
OH, yes, I forgot to make it public read:(, the links should work now!

azolyoung, 2018-06-25:
here is matekf722 link: https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF722.hex

mikeller, 2018-06-25:
The 3.4.0-RC3 release contains these changes as well...

Gregmaan, 2018-06-25:
I tested the #6204 build and all seems fine. Many Thanks!
There seems to be other benefits. My LED strip works much better too ;-)

azolyoung, 2018-06-25:
@Gregmaan Great! Thanks for your help!

mikeller, 2018-06-25:
@Gregmaan: Can you please close this issue then?

ghostface, 2018-06-25:
I still get the same behavior using bf 3.4 rc3 on split mini with softserial on ppm/led pad using a zeus f4 (omnibusf4 target). Cam Button won't work until I soft-reboot via the OSD.

alancastell, 2018-06-25:
I am still having the same issue on a Maetek F405.  Just updated, and checked that switches and uart was correct.

mikeller, 2018-06-25:
@alancastell: There's another fix coming in #6213.

alancastell, 2018-06-25:
Thanks, tested the reboot in OSD and then it started working.  Appreciate this @mikeller

azolyoung, 2018-06-26:
Hi @ghostface, @alancastell  I'm sorry, the previous version still has bug about the initializing, thanks for your help), and the following are the new firmwares that build from #6213, it fixed the problem that need reboot FC
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4SD_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_OMNIBUSF4_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF722_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_MATEKF405_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_CLRACINGF4_PR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_3.4.0_BETAFLIGHTF4_PR6213.hex

alancastell, 2018-06-26:
Great. Got one for the Brain Fpv Radix?  Thanks so much guys.

azolyoung, 2018-06-26:
Hi @alancastell , Sure), here are the firmwares based on brainfpv rc3, and merge the changes of #6213:
https://s3.amazonaws.com/rctmpfiles/betaflight_v3.4.0-RC3-BRAIN-1_BRAINRE1_BFPR6213.hex
https://s3.amazonaws.com/rctmpfiles/betaflight_v3.4.0-RC3-BRAIN-1_RADIX_BFPR6213.hex

[15 more replies]

closed 2018-07-08 by Gregmaan