betaflight/betaflight #5404: Feature Request: Add blackbox loop recording

wuhao19920619, 2018-03-07:
I use a 512MB sd card so it is very often that the space is full. the last log file will just stop where there is no space. I think loop recording can make things easier. It's like when the free space lower than 50mb, then delete the earliest files until the free space larger than 100mb.

wuhao19920619, 2018-03-07:
especially I set the log rate is 8hz. single file could be larger than 100mb.

etracer65, 2018-03-07:
Or maybe just use a larger SD card - they're cheap.  For it to only be 512MB it must be ancient and likely speed limited as well.

wuhao19920619, 2018-03-07:
@etracer65 Yes I know. The thing is you have to clean logs regularly, otherwise blackbox will stop working. I usually keep loging all the time so if someday accident happen I can find the result. Just wondering if this can be like runcam recording which support loop recording.

Ozzy33, 2018-03-09:
There has been many loop recording requests, and they say it can not be done.

wuhao19920619, 2018-03-13:
@Ozzy33 I tried to search "loop" and "blackbox" but I don't find any issue about loop recording. And why it can not be done? The mechanism seems pretty clear and not that difficult.

Ozzy33, 2018-03-13:
Sorry, I am not a programmer, just quite a few of us wanted this feature and it was said it can't be done.

mikeller, 2018-03-21:
@wuhao19920619, @Ozzy33: There have been a number of requests to add loop recording for onboard flash (e.g. #3343). The reason that this is not possible is that erasing flash takes a lot longer than writing flash, so that loop writing would drastically reduce the maximum possible logging speed.

As for logging to SD card, the general recommendation is to use a large enough card, since they are dirt cheap these days.

wuhao19920619, 2018-03-29:
@mikeller Yes I understand. I mean this not have to be "overwrite" or something simultaneously, For example if the free space is lower then 20%, It can just automatically delete the old logs when no arming until the free space >=then 20%. this 20% can be set to 20%/50% or even larger, just depends on how often you plug in your quad without flying or the size of your card. Once the quad arm, The earsing just stop(no matter if one file be deleted completely or just in the halfway) to make sure the recording can work fine. Once disarm, it then start for where it stops.

mikeller, 2018-03-30:
@wuhao19920619: Have you looked into setting up BLACKBOX_ERASE on a switch on your TX? This will allow you to erase the onboard flash blackbox in the field.