betaflight/betaflight #5412: Feature Request: Add date and time information to blackbox log.

wuhao19920619, 2018-03-07:
It seems no date and time information for blackbox log to indicate when does it be created or modified. Then it seems very hard to find a log several hours or days ago. The only information you can use is the number in log name but it is kind of useless in this case.

DieHertz, 2018-03-07:
When realtime clock value is sent over telemetry from your transmitter - that's possible, but in general there's no way to keep track of time on an FC

wuhao19920619, 2018-03-07:
OK I see... For now I don't have any better idea about this...

zuim, 2018-03-22:
Additionally the gps time could be used, if a gps module is connected.

zuim, 2018-03-23:
#5520 adds time to logs, when there is a gps connection

JonnyPhenomenon, 2018-03-27:
I just found this thread because I had a feature request idea and wanted to see if anyone else had thought of it.
So my idea was actually for "goggle DVR" recordings, so you would have some idea of when the flight was recorded, but this could work for the blackbox logs too. 
Basically, if we could somehow get the date and time from the taranis (or whatever) on power up, and display it on the OSD, for a few seconds on bootup, or maybe a second or two after arm, 

Unlike the flight controller, and every set of goggles _I've_ ever seen, the Taranis _DOES_ keep track of time, and has a battery inside for that purpose. it would be great to have a "time and date update" take place each time the radio pairs up with the receiver, and somehow upload that information to the flight controller so it could show it on the OSD.  

So how easy(hard) would this be to do? I mean, could it be dome with some clever LUA scripting and just a little bit of code in betaflight? or would it need some serious collaboration with the OPEN-TX folks too?

binlan, 2018-06-28:
or is it possible to hook up a rtc to the flight controller? something like [that](https://www.amazon.de/DaoRier-Pr%C3%A4zision-Echtzeituhr-Arduino-Raspberry-Schwarz-x-1/dp/B06XYMCG7P/ref=pd_sim_107_2?_encoding=UTF8&pd_rd_i=B06XYMCG7P&pd_rd_r=16fe5cae-7ae8-11e8-b634-dbf30619051a&pd_rd_w=lOR2G&pd_rd_wg=c7sbJ&pf_rd_i=desktop-dp-sims&pf_rd_m=A3JWKAKR8XB7XF&pf_rd_p=5296994838746851949&pf_rd_r=4D38X1BVQYXBY1XWSDWP&pf_rd_s=desktop-dp-sims&pf_rd_t=40701&psc=1&refRID=4D38X1BVQYXBY1XWSDWP).

mikeller, 2018-06-29:
This is already done, if the time is set by any of the supported means (GPS / from TX via MSP):

https://github.com/betaflight/betaflight/blob/e260c37be3b45c26659281d4f4d8489a80dfc08a/src/main/blackbox/blackbox.c#L1201

@wuhao19920619: You can probably close this.

mikeller, 2018-07-07:
@wuhao19920619: Can you please close, this has been implemented.

closed 2018-07-08 by mikeller