betaflight/betaflight #5421: The idea of the dispatching of processing resources

flash-v, 2018-03-08:
There is one idea, maybe stupid. Now the flight controller processes the data from the gyroscope and counts how much throttle  it should give to each motor. This loads the CPU heavily. And this is the main problem of BF for applying new effective filters for gyro at 32/32 mode.
Is it possible for the ECS to calculate the necessary throttle based on the transmitted data from the gyro? That is, we get rid of the multishot and dshot protocols, and transfer the gyroscope data to the ECS, and they calculate the required level of throttle. Thus, part of the processing power is transferred from the main processor to the controller processors.

jflyper, 2018-03-08:
You mean offloading filtering and PID computation to processors on ESCs?

If so, what makes you think those processors can handle filtering + PID + motor control without being loaded heavily?

DieHertz, 2018-03-08:
Main load is gyro filtering and feeding ESCs with already commuted data anyway, off-loading PID to ESCs won't save a significant amount of CPU time, and ESCs are struggling already anyway.

flash-v, 2018-03-08:
"You mean offloading filtering and PID computation to processors on ESCs?" 
I think so, that's what I mean. Now there are boards with two processors F3 and F4 for calculation determination. Why can not it be allocated to an ESC with 32 bit processors?
"Main load is gyro filtering and feeding ESCs with already commuted data anyway, off-loading PID to ESCs won't save a significant amount of CPU time, and ESCs are struggling already anyway."
 I mean, on the main processor, leave the gyroscope calculations and filtering, and transfer the PID and throttle to the ESCs.

flash-v, 2018-03-08:
guys, you're just a genius, if you realize that, then it will be a revolution. :)

closed 2018-03-09 by flash-v