betaflight/betaflight #5426: SPRacing F3 Acro/Delux 10DOF: VCC voltage is very low when powered with USB

myuunosuke, 2018-03-09:
I'm trying to setting up a small drone with barometer.
I've got a SPRacing F3 10DOF with barometer and Motor driver for brushed motors.
When I was trying to configure receiver, I noticed that the receiver is not powering up when powered from USB...
I disconnected everything and measured voltage level on VCC of the UART2, UART3, Motor out, and IO Port..., they are all around 1.6V.
When I connect regulated (step up from 4.7V)  battery power to the output port VCC(VCC of M6) and GND(GND of M5), the VCC and GND of the UARTs get 5V and the receiver powers on..., However the Betaflight software lose communication from the FC.

My setup
- SPRacing F3 Delux 10DOF
https://www.banggood.com/Upgrade-NAZE32-F3-Flight-Controller-Acro-6-DOF-Deluxe-10-DOF-for-Multirotor-Racing-p-1010232.html?rmmds=myorder&cur_warehouse=CN

- Motor driver for Brushed motors
https://www.banggood.com/Micro-Brushless-Motor-Driver-for-Naze32-Spracing-F3-Flight-Controller-p-1079832.html?rmmds=myorder&cur_warehouse=CN

- Flysky Receiver
https://www.banggood.com/818CH-Mini-Receiver-With-PPM-iBus-SBUS-Output-for-Flysky-i6-i6x-AFHDS-2A-Transmitter-p-1183313.html?rmmds=myorder&cur_warehouse=CN

Is this normal thing with SPRacing F3 Acro/Delux?

If this is normal thing... how do you configure it to have receiver working while connected with USB???

[image]

[image]

BrenLaw, 2018-03-09:
Many numerous flight controllers do not power up peripherals when plugged in via USB. This is not a special case with just your flight controller. It's identical behavior that's shared with many other different flight controllers of all brands, shapes, and sizes.

However, it is abnormal that having a battery/power source connected makes you unable to connect to configurator. I don't quite know what to make of that.

etracer65, 2018-03-09:
First, I'm going to close this issue because it is not related to the Betaflight firmware.  The best place to seek help on this would be the rcgroups thread.  https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

But yes, this is normal.  Most flight controllers do not provide power to the motor outputs while connected via USB.  Otherwise the ESC's would also power up when connected via USB which wouldn't be a good thing.  Some will provide 5V power to other pads like specifically a dedicated RX connection but not all.  My experience with the SPRacingF3 is that it doesn't.  Even less information about this specific one as you apparently have a knock-off flight controller and not a real SPRacingF3.

closed 2018-03-09 by etracer65

myuunosuke, 2018-03-09:
Thank you very much for the information from both of you.
It turned out that I was connecting my receiver module to UART1 which is also used for USB and that was distracting the USB communication with my FC.
Thank you,