betaflight/betaflight #5428: Betaflight appears to crash(software) in flight.

tavdog, 2018-03-09:
1. I had two apparent failsafe crashes today which turned out not to be failsafe but an apparent crash of betaflight itself, or at least the blackbox logging part of it.  I dropped out of the sky twice, no phase 1 (shouldn't gone into horizon mode), no phase 2, should attempted to land itself.  The blackbox log for this time in the flight simple cuts off.  

2. No known way to reproduce at the moment.
3. Matek F405 with FCHUB-VTX with R-XSR.
4. BF 3.3.0
5. ```# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.3.0 Mar  2 2018 / 03:52:04 (177472b4f) MSP API: 1.37

# name
name QAVR

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15
# feature
feature LED_STRIP
feature ANTI_GRAVITY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# map
map TAER1234

# serial
serial 4 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 3,0::CI:10
led 1 4,0::C:3
led 2 5,0::CB:12
led 3 6,0::CB:12
led 4 7,0::C:3
led 5 8,0::CI:10

# color

# mode_color

# aux
aux 0 0 0 1825 2100 0
aux 1 2 1 1875 2100 0
aux 2 13 3 1325 1675 0
aux 3 26 0 900 2100 0
aux 4 28 1 900 1600 0
aux 5 35 3 1875 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1900
rxfail 10 s 1000

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set min_check = 1010
set rssi_channel = 11
set airmode_start_throttle_percent = 35
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 50
set failsafe_off_delay = 40
set failsafe_throttle = 1100
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1300
set beeper_dshot_beacon_tone = 1
set small_angle = 180
set gyro_cal_on_first_arm = ON
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_warnings = 63
set osd_cap_alarm = 1300
set osd_tim1 = 2561
set osd_vbat_pos = 1
set osd_rssi_pos = 2437
set osd_tim_1_pos = 2054
set osd_tim_2_pos = 33
set osd_flymode_pos = 405
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2453
set osd_mah_drawn_pos = 2440
set osd_craft_name_pos = 2062
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 192
set osd_pid_pitch_pos = 160
set osd_pid_yaw_pos = 128
set osd_debug_pos = 1
set osd_power_pos = 320
set osd_pidrate_profile_pos = 334
set osd_warnings_pos = 362
set osd_avg_cell_voltage_pos = 2446
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2219
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 3
set vtx_freq = 5695
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 5000
set p_roll = 50

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 31
set yaw_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 80
set pitch_srate = 80```
[image]

[LOG00008.BFL.zip](https://github.com/betaflight/betaflight/files/1795885/LOG00008.BFL.zip)

etracer65, 2018-03-09:
The blackbox log simply shows a disarm - you can even see it in the picture you posted.  The log doesn't cut off.  It ends normally due to the disarming.  If the flight controller really had crashed there wouldn't be a disarm event logged and the graphs would show a break in the data (small empty square markers).  Everything looks perfectly normal to me.

Possibly you accidentally disarmed?  Of maybe you have a failing switch on your radio that's not making good contact.  All it would take would be a momentary blip from the disarmed position.

tavdog, 2018-03-09:
Interesting.  Now that you say this I have see rapid disarm/rearm events in the log before with no ill effects.  I guess this could be a receiver issue then ? Is there a way to force betaflight to not disarm if the throttle is high ?
[image]

Mechotronic, 2018-03-09:
@tavdog how tight is your mode setting for arm? You might have the edges too tight in the settings and maybe you can lower the bottom slider more?

SRWieZ, 2018-03-09:
Rooster - Betaflight F4 - Betaflight ESC 32bits - RXSR - TBS Unify Pro - Hypetrain - BF version 3.2.4

I have a similar problem on one of my quad. I got twitches wen i was flying so I synced the DVR with blackbox to see what's causing the problem. That was one week ago, unresolved yet.

You can see that my 3 pos switch (disarmed/angle/air) keep switching fast between angle and air mode.
Sorry I have only take theses pictures to send to a a friend. But believe me, it twitches so fast and sometimes it keep doing it for 1 second.

[image]

Here you can see it on the bench where I wasn't doing anything with my radio.

[image]

It's a new build, others drones doesn't do that so I assuming that there is some bad batch of RXSR. It's append randomly. I think is related to RSSI, it doesn't seem to happen in Good RF Situation.

I should mention that I had some trouble binding and flashing this receiver (I have about 3 quad with RXSR with no problem). 
This one is flashed with [171103](https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/R-XSR/FW-RXSR-171103.zip). 
I just saw there is a new version [180228](https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/R-XSR/FW-RXSR-180228.zip) of [RXSR firmware ](https://www.frsky-rc.com/r-xsr/). 
I will try tomorrow to see if it fixes the problem.

Hope this will help you :)

etracer65, 2018-03-10:
I’m going to close this issue because it seems to be either a hardware issue or some problem with the RX firmware and not a  Betaflight issue.

closed 2018-03-10 by etracer65