betaflight/betaflight #5435: Resource Remapping Not Appearing to Save

lukedempsey, 2018-03-10:
I'm trying to remap my MOTOR 4 (B09) to my TX3 pin (B10) on a BFF3

I save and reboot but it doesn't update the resource list.

Diff:
```
#Betaflight / BETAFLIGHTF3 (BFF3) 3.3.0 Mar  2 2018 / 03:38:14 (177472b4f) MSP API: 1.37
resource MOTOR 4 B10
resource SERIAL_TX 3 NONE

set acc_hardware = NONE
set motor_pwm_protocol = DSHOT600
profile 0

rateprofile 0
```
Sorry if this is a PEBCAK, but I thought I would post it here because if it is a misunderstanding, the [wiki I referred to](https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping) could be updated.

jflyper, 2018-03-10:
@lukedempsey 
You wanted to wire motor 4 to TX3 pad?

closed 2018-03-10 by jflyper

lukedempsey, 2018-03-10:
Thanks for your help.

If any one is here because they ran into the same problem, I found the available remapping ports under the [BFF3 Board Wiki Page](https://github.com/betaflight/betaflight/wiki/Board---BETAFLIGHTF3#remapping-for-motors)