betaflight/betaflight #5436: Could not load background script 'js/eventPage.js'.

mrakyhub, 2018-03-10:
chrome send this error, when i want to add extension in linux(mint):
Failed to load extension from: /my/directories/betaflight-configurator-10.2.0
Could not load background script 'js/eventPage.js'.

{
    "manifest_version": 2,
    "minimum_chrome_version": "38",
    "version": "10.2.0",
    "author": "Betaflight Squad",
    "name": "Betaflight - Configurator",
    "short_name": "Betaflight",
    "description": "Crossplatform configuration tool for Betaflight flight control system",

    "offline_enabled": true,

    "default_locale": "en",

    "app": {
        "background": {
            "scripts": ["js/eventPage.js"],
            "persistent": false
        }
    },
    "permissions": [
        "https://maps.googleapis.com/*",    
        "https://*.github.com/",
        "https://*.githubusercontent.com/",
        "http://*.baseflight.net/",
        "https://*.amazonaws.com/",
        "serial",
        "usb",
        "storage",
        "fileSystem",
        "fileSystem.write",
        "fileSystem.retainEntries",
        "notifications",
        "alwaysOnTopWindows",
        {"usbDevices": [
            {"vendorId": 1155, "productId": 57105}
        ]},
        "webview"
    ],
    "sockets": {
        "tcp": {
            "connect": "*:*"
        }
    },
    "webview": {
        "partitions": [
            {
                "name": "map",
                "accessible_resources" : ["tabs/map.html", "js/tabs/map.js", "/images/icons/cf_icon_position.png"]
            }
        ]
    },

    "icons": {
        "128": "images/bf_icon_128.png"
    }
}

etracer65, 2018-03-10:
This is a configurator question.  Please post issues/questions about the configurator here:

https://github.com/betaflight/betaflight-configurator/issues

closed 2018-03-10 by etracer65

mrakyhub, 2018-03-11:
> Komu: betaflight/betaflight <betaflight@noreply.github.com>
> eventPage.js'. (#5436)
> "

thanks!
---------- Původní e-mail ----------

https://github.com/betaflight/betaflight-configurator/issues
(https://github.com/betaflight/betaflight-configurator/issues)

(https://github.com/betaflight/betaflight/issues/5436#issuecomment-372073939)

"