betaflight/betaflight #5439: FPVM Betaflight F7 Maxing out CPU Load

SupaCoop, 2018-03-11:
Hi, I recently attempted to make the switch to butterflight and am having issues running 32khz sampling. I'm running the betaflight f7 on 3.4.2 with the icm20602 gyro enabled and it reads 100% cpu load on 16khz/16khz without the dynamic filter. I've seen guys with f4's running 32khz/16khz with the dynamic filter enabled and they were running just fine. I tried to contact fpv model and they literally told me to wait for the next firmware version to come out. An f7 should have plenty of processor cycles to spare even with 32khz/32khz and the dynamic filter, right? Any ideas?

fujin, 2018-03-11:
Butterflight is not supported.

This is a duplicate of #3191.

FPVMODEL's recommendation to wait for the next firmware version is accurate at this time, F7 optimization fixes are in-progress / planned for 3.4 and looking pretty good so far.

closed 2018-03-11 by fujin

DieHertz, 2018-03-11:
I'm working on the F7 improvements, wait for it.

mmakam2, 2018-05-24:
I also have the same issue on mine..

 Too bad distributors don't mention this when they sell them...

DieHertz, 2018-05-24:
F7 works quite good in 3.4 development builds.
It's your responsibility as a user to conduct the preliminary research, F7 performance issues weren't a secret

mmakam2, 2018-05-24:
Mine only works on 16khz/16khz on the 3.4 build, 32khz maxes the CPU still.

Teeheeze, 2018-05-24:
I just flashed my MatekF722 with the latest build and it has fixed most of the issues I had with the F7. The only thing I've noticed is that when using 32khz/32khz, the CPU reads around 50% at first for about a minute, then after accessing the configuration tab, it randomly jumps to 100% again. This only happens when using 32khz/32khz. When I change it to 32khz/16khz, the CPU reads around 26-28% and stays around there permanently. Just thought I'd point out what's happening with me... Thank you so much to all the developers working on this issue, especially DieHertz!

mmakam2, 2018-05-25:
I can confirm the same behavior on the Betaflight F7 board as Teeheeze.

jflyper, 2018-05-26:
Those of you having issues F7 CPU maxing out, please attach your CLI `diff` output please.

mmakam2, 2018-05-26:
this is the config that maxes it out:

$M>lj$M>ÍtARM;FAILSAFE;ANTI GRAVITY;BEEPER;OSD DISABLE SW;TELEMETRY;BLACKBOX;FPV ANGLE MIX;BLACKBOX ERASE (>30s);CAMERA CONTROL 1;CAMERA CONTROL 2;CAMERA CONTROL 3;FLIP OVER AFTER CRASH;PREARM;VTX PIT MODE;PARALYZE;É$M>e$ @dE$M>$ @dµ
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / FPVM_BETAFLIGHTF7 (FBF7) 3.4.0 May 25 2018 / 11:09:21 (2d94ccd8a) MSP API: 1.39

# name
name CHAMELEONTI5

# resources
resource MOTOR 2 E11
resource MOTOR 3 B01
resource MOTOR 4 E09

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0
aux 1 13 0 1375 1600 0
aux 2 26 1 1300 1700 0
aux 3 31 1 1825 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 7
set serialrx_provider = CRSF
set input_filtering_mode = ON
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 32000
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_rssi_pos = 2209
set osd_tim_2_pos = 2433
set osd_vtx_channel_pos = 2337
set osd_crosshairs_pos = 2253
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2241
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5725

# profile
profile 0

set vbat_pid_gain = ON
set p_pitch = 59
set i_pitch = 36
set p_roll = 41
set i_roll = 26

# rateprofile
rateprofile 0

set thr_expo = 5
set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 105
set tpa_rate = 24
set tpa_breakpoint = 1425

Strangedave, 2018-06-13:
I am also experiencing the same issue. There also seems to be a problem with the black box logs and the osd also. The black box logs are showing motor noise up in the 1300 KHZ range. I was not able to move the crosshair in the OSD. I am including the most recent black box log, and Diff. I did try stock settings, the motors sounded like they were sputtering, so I went back to my old PID's. I need to work on filters, but the logs are confusing me. By the way logs are very slow to down load, 3.3.3 was also slow, but I left this quad plugged in over night to get 16MB out of it.

Setup is 
Xhover Win 5
Betaflight F7 FC
32Khz Gyro 16Khz Pid
DAL Engine 4-in-1 esc
T-Motor F40 Pro II
Tramp HV
RSXR tx
Tattu 1550 MAH 75C 
HQ Prop 5X4.8X3 V1S

[Diff 3-4-0 v1.txt](https://github.com/betaflight/betaflight/files/2099071/Diff.3-4-0.v1.txt)

[BTFL_BLACKBOX_LOG_WIN_5_20180612_212806.zip](https://github.com/betaflight/betaflight/files/2099072/BTFL_BLACKBOX_LOG_WIN_5_20180612_212806.zip)

mmakam2, 2018-06-13:
I can help with atleast one issue. The OSD position value changed int he CLI, version to version, so put it back to the build's default value instead of restoring from cli backup. You can adjust the position with osd crosshair set value.

Strangedave, 2018-06-13:
Thanks I'll give that a try... Ran a few packs this afternoon with out the crosshairs, and I'm considering removing them.

Hideloop, 2018-06-15:
Sorry to ask. Issue is closed but persists in Rc1. Is there a build fixing it for Fpvm betaflight f7?

DieHertz, 2018-06-15:
The original issue was about 100% CPU load at 16/16.
Does 16/16 caused 100% CPU load for you?

Hideloop, 2018-06-15:
I'm sorry. Missed that. 
32/16 is now working fine at 22 percent cpu. I was in the thought that 32/32 is possible in bf 3.4 with f7. So if I understand you right this is not the case and possible or is this worth of a new issue?

DieHertz, 2018-06-15:
Not sure it'll be possible in foreseeable future due to amount of new features added into PID loop :)
I'll surely look into optimizing further for BF 3.5/4.0, maybe some change of architecture would be required, maybe something is missed still.

For now 32/32 on F722 targets is prevented by GPS rescue, we're working on extracting out so users who don't use it aren't affected.

Hideloop, 2018-06-15:
Thank you very much. If I can support with testing or anything let me know. And thank you for your great work.

Strangedave, 2018-06-15:
Thanks for the info, I am sure I posted this in the wrong place. 32/16 is noticeably better than 8/8, so I am super happy today and look forward to future  versions.. if I can be of assistance please let me know.