betaflight/betaflight #5440: Reduce FFT sampling rate for 8k and above gyro loop frequency

ctzsnooze, 2018-03-12:
rav-rav who did the FFT coding for the dynamic filter has said that it was optimised for a 4k gyro loop frequency.  See:  https://github.com/betaflight/betaflight/pull/5401#issuecomment-371797617

Furthermore, he says that running it more frequently is not beneficial, performance-wise.

The dynamic filter becomes a significant CPU load when we run a gyro loop frequency higher than 4k.  We should get significantly less CPU load if we don't run the FFT calculations faster above 4k gyro loop.

So I am suggesting that the code should cut the frequency of FFT update according to gyro loop frequency to keep the absolute number of FFT calculations constant at what they would have been if the gyro loop was running at 4k.  Should make a big difference at 32k.

ledvinap, 2018-03-12:
@ctzsnooze : Aliating should be taken into account when reducing sample rate. Lowpass filter is probably necessary ...

DieHertz, 2018-03-12:
What if we take samples after high-cutoff targeted lowpass? That way we'll also get rid of some 32KHz-mode noise.

ctzsnooze, 2018-03-12:
rav-rav said it was designed for 4k use and - I'm assuming - the MPU6000 with its 250Hz hardware filter.  

There is so much noise in 32k mode or 8k experimental I would have thought it best to do it after at least one software PT1 filter.  I anticipate that these gyro modes won't fly without one low and one high PT1 at least.  

Hence it would seem reasonable to sample after the higher of the two lowpass gyro filters when in 32k mode and in 8k experimental mode.  In all other modes sample the incoming gyro.

Main problem is the user could set either gyro LPF to off, and either to any value.  

Perhaps in 32k mode it would be best to just do it after whatever gyro filtering is enabled.  The peak will still be there.

DieHertz, 2018-03-12:
If user disables to higher lowpass which is necessary for 32K - it will not fly, just like disabling all filters is likely to burn your motors today.
I don't think it's that big of an issue, the main issue is we don't want it to be "dynamic" in terms of where it can be applied, that may affect CPU performance, it should have a fixed place in the filter chain to minimize branching.

ctzsnooze, 2018-03-12:
OK I agree with that.  Perhaps when the dual stage gyro code arrives (soon, I hope) we can put it after the first, and provide advice that the first filter should in 32k modes be something like 250Hz.

I've now flown these codes a lot and even with the hardware 250Hz filter enabled, I get fantastic noise reduction results with 150/300 and very good results with 125/250 on the gyro filters.  Both fly at least as well or better than a single 100Hz PT1 I think.

rav-rav, 2018-03-12:
There is no aliasing, as all samples are used.
The current implementation was targeted for 4kHz sampling. It collects 4 samples and calculates the average of those. Then it calculates the FFT and new filter settings in 4 steps. So you get your signal downsampled to 1kHz but get no aliasing and less noise compared to skipping samples.

> If multiple samples are taken of the same quantity with uncorrelated noise added to each sample, then averaging N samples reduces the noise power by a factor of 1/N.[4] If, for example, we oversample by a factor of 4, the signal-to-noise ratio in terms of power improves by factor of 4 which corresponds to a factor of 2 improvement in terms of voltage.
[Oversampling Wikipedia](https://en.wikipedia.org/wiki/Oversampling)

If you run the FFT on for example 2kHz data, the only difference will be additional info on frequencies above 500Hz, but no additional info on anything up to 500Hz. But it will cost twice as much computations.

rav-rav, 2018-03-12:
There's no need to change anything except the hard coded 4 into the appropriate sampling rate.

rav-rav, 2018-03-12:
Ok I should have checked the code first.
There is no hard coded 4. It's already scaled correctly.
fftSamplingScale = samplingFrequency / FFT_SAMPLING_RATE;

Nothing to be done here actually.

ctzsnooze, 2018-03-12:
So if the gyro loop is at 32k, the FFT code will only collect every 8th sample to get averaged, smooth four, then do the FFT?  And we don't need to do anything, the code already has it right?

ledvinap, 2018-03-12:
@rav-rav : Averaging works as low-pass filter and it is probably enough for intended use.

Sampling is still subject to Nyquist criterion. Gyro noise in not white noise, motor and commutation harmonics are quite strong in 500Hz + band, these would be aliased to 0..500Hz.

rav-rav, 2018-03-12:
@ctzsnooze in 32k 32 Samples will be averaged, as the target is to have 1kHz as input for FFT.
@ledvinap I agree with everything. The power above 500Hz however is already quite low even before averaging and should not really interfere with the analysis.

Does anyone have some logs with 32kHz data to take a look?

apoc, 2018-03-12:
Hi, I just glanced to the source code so I hope I am not miss-interpret the thing but this is what I have found:

Every gyroUpdateSensor is called gyroDataAnalyse, non filtered value is taken there and added to the accumulator for each axis. After each fftSamplingScale(samplingFrequency / FFT_SAMPLING_RATE, for 32k gyro loop it is 32k / 1k so ~ 32 values) the accumulated samples are averaged sample = fftAcc[axis] / fftSamplingScale, this sample is filtered out by biquadFilterApply(&fftGyroFilter[axis], sample) in BPF mode, so the data comming to FFT are actually filtered and calculated center frequency is also filtered by LPF centerFreq = biquadFilterApply(&fftFreqFilter[axis], centerFreq).

About performance impact the main caveat is that gyroDataAnalyseUpdate(notchFilterDyn); is called with every gyro loop "tick" what is not neccesary because the fresh data are not sampled yet, for update you need 4 steps aka 4x call the gyroDataAnalyseUpdate, so I think the only thing needs to be done is:

`
// fftIdx is 0 means we have full window to analyze, also fftAccCount == 0 in this case, we need 4 steps to calculate FFT and update filters
if (fftIdx == 0 && fftAccCount < 4) {
    // calculate FFT and update filters
    gyroDataAnalyseUpdate(notchFilterDyn);
}
`

The proposed condition should work for sampling rates >= 4kHz

apoc, 2018-03-12:
@rav-rav I have some 32KHz log files, take a look at this one for example
[LOG00008.zip](https://github.com/betaflight/betaflight/files/1803624/LOG00008.zip)

apoc, 2018-03-12:
Or the gyroDataAnalyse can be limited in gyro loop

[aaa](https://github.com/apoc/betaflight/commit/6331548a6c568c5d6ccc3dfffd7fa696d1bd318d)

rav-rav, 2018-03-13:
The approach using fftAccCount < 12 is better. If you limit the rate in which gyroDataAnalyse is called, only 1/8 of the data will be used for FFT and this will make aliasing a lot worse.
One could also change the update steps for 32kHz. Some are grouped right now in order to have 4 steps which take approx the same time. This could be changed to 8 smaller steps for 32kHz rate but would need to stay the same for smaller rates.

apoc, 2018-03-13:
I also have idea to put gyroDataAnalyseUpdate into the main gyro loop, thus sampling and actual calculation of filters can run on independent "ticks".

I also little bit "cleaned up" the gyroDataAnalyse code, removed some variables: [code](https://github.com/apoc/betaflight/commits/limit_fft_sampling_rate)

DieHertz, 2018-03-13:
I can work on regrouping as I got used to measuring performance lately.

apoc, 2018-03-14:
Ok, I guess I have final version. I experimented also with FFT sampling frequency change just to see the possible performance benefit, but it is almost same as my current implementation. The CPU load dropped from 45% to about 39% (shown in configurator).

[branch](https://github.com/apoc/betaflight/commits/dyn_filters_improvements)

There is also BB log taken after changes
[LOG00007.zip](https://github.com/betaflight/betaflight/files/1811447/LOG00007.zip)

closed 2018-05-05 by ctzsnooze