betaflight/betaflight #5444: D setpoint weight

Yamaford, 2018-03-12:
Would it be possible to add another slider to setpoint?
I would like to see error generated from RC input seperated from error generated from propwash or other externals. It seems to me we could set the slider for RC lower softening up D term from stick inputs and stiffen up if the error is generated from the gyro.

etracer65, 2018-03-12:
I'm confused by what you're asking.  The setpoint has nothing to do with error from measurement (the gyros).  It only responds to error from rcCommand.  The actual changes in the sticks are what drive the D-term boost scaled by the setpoint weight.

The only place we can see the effects of propwash is in the gyros and the normal PID response is responding to it.  Don't see any way we can distinguish propwash measurement error from any other.

Yamaford, 2018-03-12:
If the rc command isn't moving much and the gyro is, then the error is from
gyro. If the gyro gets a sudden error jolt I can correct for this by
raising the settings weight as it is now completely removing propwash but
then my stick input is more harsh than I like. I can lower my setpoint
weight and get the feel I want in the sticks but propwash handling isn't
perfect.
What do you mean weight doesn't have anything to do with measurement? Afaik
it is the deciding factor between D term responding to error or
measurement.

etracer65, 2018-03-12:
D-term always responds to measurement error.  Setpoint weight is like a "boost" given to D-term based on stick velocity.  When the quad is commanded to roll the gyro measurement lags because of inertia and motor spin-up delays.  So the stick movement can be used as a clue that we need to accelerate the reaction (push harder) initially to overcome the delay.  So it's a hybrid between the old measurement vs. error D-term calculation.  Measurement all the time but an extra push from changing stick error.  Measurement is more accurate and is how you want your flight controller to behave, but it lacks that connected feeling on initial stick input (because of the delays mentioned).  Using pure error-based D-term calculation gives the stick connection, but does a worse job of actually flying (no connection to reality).  So the combination gives you the best of both.