betaflight/betaflight #5452: switch off osd and stats

Gaz-B, 2018-03-14:
Could we have the osd off switch also switch off the stats screen. landing to change camera setting would be soooo much easier without a completely full screen of text.

etracer65, 2018-03-14:
Seems reasonable.  I'll look into it.

wind0r, 2018-03-14:
i think this allready got added with 3.3.
if the user disables all stats no stat screen will be shown.

https://github.com/betaflight/betaflight/blob/master/src/main/io/osd.c#L1246

etracer65, 2018-03-14:
Yes I thought that at first as well but then re-read and he's not asking to always disable stats.  Only when the OSD DISABLE SW mode is on.  You land, disarm and want to adjust your camera but the stats are on the screen in the way (at least until they timeout).

Gaz-B, 2018-03-14:
That's it and time out is a very long wait.

Gaz-B, 2018-03-14:
Thanks didn't know that. I was quite surprised when the switch didn't switch the stats off that's the only time i normally want the screen clear.

m013, 2018-03-16:
what if  people have   cross-hairs or other stats (battery, throttle value ...) placed the way it overlaps camera control osd?

etracer65, 2018-03-16:
Then setup the OSD DISABLE SW on the modes tab.

etracer65, 2018-03-18:
Created https://github.com/betaflight/betaflight/pull/5463

m013, 2018-03-18:
> Then setup the OSD DISABLE SW on the modes 

i did. but
the idea was  not to use another function for a SW. some people fly with 8ch mode, so the number of free switches is limited.

etracer65, 2018-03-18:
@m013 While it would be nice to disable the OSD when entering camera control, it's really not possible to make this work well.  The problem is that the commands to enter the camera OSD are "blind" and are simply emulating the camera dongle.  There's no feedback from the camera to the FC that it's actually in the camera OSD.  Even worse, there's no way to know that the user exited the camera OSD and therefore the FC OSD should re-enable.  So the only viable option is to let the user disable the FC OSD on demand when appropriate.  Even running with only 4 aux channels you have options.  Use mixes on your radio to emulate more channels, etc.

m013, 2018-03-18:
seems legit. thanks

PR #5463 (2018-03-20): Make OSD stats page follow the state of the OSD disable switch mode
Fixes #5452

Previously the OSD stats would always display even if the OSD disable switch mode was active.

Additionally, since the OSD stats page must now be dynamically refreshed rather than only rendered once at disarm, "live" stats like RTC clock and the timer "ON TIME" continue to update and display current information.  All other stats related to the previous flight remain static as of the disarming.
closed 2018-03-20 by mikeller