betaflight/betaflight #5453: Low pass filter gerated through assembly code

quadmx, 2018-03-14:
I am not familiar with really large projects such as Betaflight but I was good at writing code at assembly. Is there any benefit of having the code for the low pass filter used at the gyro to be in assembly? I guess it has less delay, but that depends on when and how it is called. If it is beneficial I can write some code on it.

DieHertz, 2018-03-14:
Nope, it's not, don't bother :)

quadmx, 2018-03-14:
O well, I tried. Hey I'm from the youtube channel in spanish Quadmx. It's similar to Joshua's but in spanish and Im not that large, but if you guys need help translating stuff, testing things or doing how to guides I will be glad to help out. You guys do a lot for the community.
Regards

DieHertz, 2018-03-14:
Good job, I think there's a translation of BF Configurator to Spanish, but reviewing it is always welcome 👍

mikeller, 2018-03-21:
@quadmx: If you would like to help us with translations, join the Betaflight slack (registration at http://betaflight.ch/), and make yourself heard in the #team_translation channel. We can help you get set up from there on.

quadmx, 2018-03-22:
I have tried to get all day but it just tells me that the site is taking too long to respond. Is there another way of getting in? Is this the same? https://betaflightgroup.slack.com/ 
Regards

mikeller, 2018-03-22:
What's your email address, I'll send you an invite.

mikeller, 2018-03-23:
Done.