betaflight/betaflight #5454: Matek F405 CTR BF 3.3 Wrong Compass Direction. Unable to Align to North

jalao, 2018-03-14:
I just installed a new Matek F405 CTR. Connected Neo 8M for both GPS and Compass direction. I calibrated the Mag, Accelerometer as well. Used my mobile phone compass to identify North,  I went to the configuration tab and tried aligning my quad with North. It failed. I  tried all the options from CW 0 - 270 and CW Flip 0 - 270 and was never able to get BF align North. I tried resetting the Z-axis offset from the home menu and still unable to align north. My barometer is covered and the drone flies perfectly, but the compass arrow points in the wrong direction. 
I don't want to switch to iNAV since I am a BF fan. Please help me. 
I am literally unable to go long range unless I get this compass issue resolved. 

Thanks

McGiverGim, 2018-03-14:
Have you tested if out of the configurator works? There is an issue, but only affects msp and the configurator. In the OSD for example it must work. https://github.com/betaflight/betaflight/pull/5451

hundemann, 2018-03-15:
I understood McGivenGim you are referring to a known issue with the MAG?! 
Maybe independant, but:
My CTR will not connect into the configurator and gets somehow stuck, so the configurator is not usable under 3.3.0.
Blue LED is just constantly on and non-blinking. I faced this just on the 3.3.0 versions, 
3.2.5 works without any issue. 
Could this issue result in such behaviour? I might be the only one, as it seems no oher CTR user has this problem

McGiverGim, 2018-03-15:
No, It does not seem a related problem. I can't help with this. Have you selected the erase data when flashing?

hundemann, 2018-03-15:
Yes, I tried many things, but as I'm somehow alone with this issue on this FC, I assume it might rather be related to some connected HW I use.
I just found a suggestion in another issue [#5303](https://github.com/betaflight/betaflight/issues/5303), I need to check this, as I'm using also an older 2Gb SD card for Blackbox logging

jalao, 2018-03-15:
Hi. Yes. In configurator, I am unable to align it to north. I tried changing all the Yaw offsets and the closet I got was 45 degree north.

jalao, 2018-03-15:
I don’t have this issue. I am able to install 3.3 on my CTR without any issue and it flies perfectly. I do remember though that I was getting some strange beeping noise when installing the firmware. So I basically had to power cycle, replace the battery and everything was fine.

jalao, 2018-03-15:
In addition, a colleague of mine had the same issue, installed iNav and had no problems.

jalao, 2018-03-17:
Can I please get an update on this request. I know you all are very busy but it will be great to know if iNAV will be a better option to run GPS. Thx.

McGiverGim, 2018-03-17:
If you need navigation, yes, iNav is better option. If you only need show position or compass in osd then you can use betaflight. As I said, this problem is a bug in the communication with the configurator, but the firmware is working without problem.
You can use a nightly build too if you prefer because the fix had been merged.

jalao, 2018-03-17:
Hi Miguel,

Thank you for your response. How can I get access to the Nightly build where the code has been merged. I would rather use betaflight than INav. INav has issues with crossfire at the moment. 

Thank
Jacob.

McGiverGim, 2018-03-17:
You can grab them from here https://ci.betaflight.tech/job/Betaflight/

flyandride, 2018-03-20:
Jalao, I’ve noticed that if I point the quad north, take off and fly north for about 50 feet, the home arrow appears pointing the correct direction.  For the rest of the flight the arrow will turn and always point home.    If I take off and fly S, E or W, the arrow will point in the wrong direction, keeping it’s wrong orientation for the rest of the flight.  GPS is a VK18U7 with no compass.

dbutz, 2018-03-24:
Jalao, 
Could you tell me how to get the mag to get recognized? I have matek f405-ctr and a bn880 wired to i2c slc and sda pads on the bottom of the board but I can't get the mag sensor to light up in bf. Do I need enable mag or remap resources to the i2c pins?

Thanks for any help you can provide,
David

jalao, 2018-03-24:
What version to Betaflight are you running.

flyandride, 2018-03-24:
I'm running BF 3.3 on a Matek F722.  I'm not worried about getting a compass on a BF quad as BF does not have a working RTH function.  The distance, speed and arrow home are good enough to find your way back.

jalao, 2018-03-24:
David,

I enabled Mag, and barometer. I made sure sda connects to sda and slc connects to slc. I also enabled the gps section and selected Ublox and set it to auto.  
Let me know if that works.

jalao, 2018-03-24:
Mike,

I cant get my quad to point north in BF. If I can get the arrow to work that will be great. What did you select as your Mag alignment?

jalao, 2018-03-24:
Miguel,

Thank you. I tried the code and it actually works. I got my quad to point north after selecting Mag alignment c270. My quad sees about 16 satellites but does unable to get GPS fix. Also with the drone resting on the floor, the GPS number changes rapidly with huge variations. It changed rapidly from 83.3456554 to 79.45445. Is there a fix for this?

Thx. 
Jacob.

jalao, 2018-03-26:
Hi Miguel,
THanks for your help, but can you help please? GPS Fix does not work at all, but the Compass Arrow now faces north...IF you can compile the cold with the current stable code, everything will be perfect. I am to get GPS fix on the current stable code but the compass arrow does not work, while in the new code you gave me, compass arrow works but I am unable to get GPS fix with almost 16 Satellites. Thanks.

McGiverGim, 2018-03-26:
@jalao I will build one for you based in the 3.3 release (give me some time, I'm out of my computer) but the GPS code has not changed in some time at all. I'm not too sure this will solve your problem.

And as I said many times, you can use the stable 3.3 release, you will see bad compass direction in the configurator but it will show the correct direction when flying.

dbutz, 2018-03-26:
@jalao Thanks! I flashed betaflight 3.3 and the mag sensor lit up right away. I've calibrated and put in my declination for this area and did a test hover. Now I just need to fly for real and see what happens. 

Does return to home arrow and distance only get calculated once you fly some distance away from yourself first? In my front yard it was just a dash (-) and I was thinking maybe I need to travel a hundred feet or so from where I plugged in before it can tell me that info. Otherwise the compass seems to work and GPS has coordinates.

McGiverGim, 2018-03-26:
@dbutz The dash appears when you have no satellites lock:

https://github.com/betaflight/betaflight/blob/83e614c1392d653bc9702ac87d94d258cf7012fd/src/main/io/osd.c#L447-L450

How many satellites show the OSD?

McGiverGim, 2018-03-26:
@jalao here you have, version 3.3 with the fix for the MAG in the configurator, if I've do it right:

https://drive.google.com/drive/folders/1zF2erYC2n1fRSAi3COGa00q_AW-9Eomw?usp=sharing

dbutz, 2018-03-26:
@McGiverGim I had at the time 9+ satellites and GPS coordinates.

McGiverGim, 2018-03-26:
The home arrow must show the dash (-) when no gps fix, and a symbol similar a triangle/up arrow (^) when the distance to the HOME position is less than 1 meter. 

If you see the - with 9+ satellites, something is wrong with the fix or with the Betaflight code.

dbutz, 2018-03-26:
@McGiverGim Hmm good point. I was walking around with it in my front yard to test the sensors and gps lock. I'll fly it at lunch today and get a better idea. Thanks for the tips!

jalao, 2018-03-26:
Miguel,

You are the best. Thx. I will try it and revert back. 

Jacob

tonisa, 2018-05-03:
I have too Matek F405 CTR and Ublox M8N clone from Banggood. For me mag alignment = CW 270 flip did the trick. Compass direction is correct also in Configurator. Betaflight 3.3.1

jalao, 2018-06-04:
All. When will you help us fix this issue in BetaflIght?

closed 2018-07-11 by stale[bot]