betaflight/betaflight #5455: Roll of death after flips in 3D mode 3.2.5

sygel, 2018-03-14:
I have a Graupner Alpha300Q and it flies great with the Graupner FC
The same quad with F4 Kakute AIO V2 Betaflight has roll of death problems after hard flips in 3D mode.

I added two blackbox logs (3 flights) with different settings.
I tried to lower the P (58 -> 40) and also a higher min throttle 4.5, 8 and 14%
I tried with the original Graupner oneshot esc and with new dshot600 esc, always the same problem.

[BTFL_BLACKBOX_LOG_ALPHA300Q_20180314_172444.bbl.zip](https://github.com/betaflight/betaflight/files/1813200/BTFL_BLACKBOX_LOG_ALPHA300Q_20180314_172444.bbl.zip)
[BTFL_BLACKBOX_LOG_ALPHA300Q_20180314_123714.bbl.zip](https://github.com/betaflight/betaflight/files/1813202/BTFL_BLACKBOX_LOG_ALPHA300Q_20180314_123714.bbl.zip)

etracer65, 2018-03-14:
Please include a `diff` as indicated in the issue creation instructions.

sygel, 2018-03-15:
Do you mean a diff between graupner gr18 fc and kakute betaflight fc? I don’t think that is possible. The grauper gr18 fc does not use betaflight.  I only wanted to demonstrate that it is not a hardware problem with the frame, motors, esc or lipo.

mikeller, 2018-03-15:
@sygel: From the instructions when opening an issue:

> 5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and

sygel, 2018-03-15:
Ok here is the diff:
```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / KAKUTEF4V2 (KTV2) 3.2.5 Feb 11 2018 / 00:21:39 (6e69ff00c) MSP API: 1.36

# name
name ALPHA300Q

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -OSD
feature 3D
feature AIRMODE
feature ANTI_GRAVITY

# beeper

# map
map AET21R34

# serial

# led

# color

# mode_color

# aux
aux 0 0 1 900 1300
aux 1 1 2 1700 2100
aux 2 2 2 1325 1725

# adjrange

# rxrange

# vtx

# rxfail

# master
set baro_hardware = NONE
set max_check = 1950
set blackbox_device = SPIFLASH
set max_throttle = 1800
set dshot_idle_value = 1500
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set 3d_deadband_low = 1450
set 3d_deadband_high = 1550
set 3d_neutral = 1500
set 3d_deadband_throttle = 35
set deadband = 10
set yaw_deadband = 20
set pid_process_denom = 1

# profile
profile 0

set p_pitch = 40
set i_pitch = 40
set d_pitch = 30

# rateprofile
rateprofile 0

set rc_rate = 80

# 
```

etracer65, 2018-03-15:
I don't see anything wrong with your config but the blackbox logs show gyro overflow detection kicking in and in at least one case I can see evidence of an actual gyro overflow/inversion instance.  The ICM series gyros (used on the Kakute) have a hardware bug that can cause them to return inverted gyro data under high spin rates or sometime triggered by noise.  The gyro overflow detection is designed to prevent a flyaway situation due to the falsely inverted gyro data.

Your rates don't look high enough to be triggering the overflow through command input so it's likely noise and/or vibration affecting the gyro chip.  The events also only appear to be happening when you're at very high throttle so that also points towards the vibration or electrical noise scenarios.

Some things to try:

1. Upgrade to Betaflight 3.3.  It has some improvements to gyro overflow detection logic.
2. Capture a more detailed blackbox log of the event to better see the gyro output.  Try `set debug_mode = GYRO_RAW` (don't forget to `save`) in the CLI and increase your logging rate to at least 2KHz in the blackbox tab.
3. Electrical noise has been known to effect the more sensitive ICM-series gyros.  Possibly add a capacitor to your power to filter out noise from the motors/esc's.
4. Check for damaged props or motor bearings.

sygel, 2018-03-15:
Thank you for looking into this.
I upgraded to 3.3 and made a blackbox log on the bench (motors spinning at max RPM).  Can you see in this log if I have excessive vibration? I have no idea what is considered ok.
If it looks good to you I will do another test flight.

Thank you.

[BTFL_BLACKBOX_LOG_ALPHA300Q_20180315_213527.BBL.zip](https://github.com/betaflight/betaflight/files/1817013/BTFL_BLACKBOX_LOG_ALPHA300Q_20180315_213527.BBL.zip)

```

# diff

# version
# Betaflight / KAKUTEF4V2 (KTV2) 3.3.0 Mar  2 2018 / 03:48:43 (177472b4f) MSP API: 1.37

# name
name ALPHA300Q

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -OSD
feature 3D
feature AIRMODE
feature ANTI_GRAVITY

# beeper

# map
map AET21R34

# serial

# led

# color

# mode_color

# aux
aux 0 0 1 900 1300 0
aux 1 1 2 1700 2100 0
aux 2 2 2 1325 1725 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set airmode_start_throttle_percent = 35
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set max_throttle = 1800
set dshot_idle_value = 1000
set motor_pwm_protocol = DSHOT600
set 3d_deadband_low = 1450
set 3d_deadband_high = 1550
set 3d_neutral = 1500
set 3d_deadband_throttle = 35
set small_angle = 10
set deadband = 10
set yaw_deadband = 20
set pid_process_denom = 1
set debug_mode = GYRO_RAW

# profile
profile 0

set p_pitch = 40
set i_pitch = 40
set d_pitch = 30

# rateprofile
rateprofile 0

set roll_rc_rate = 80
set pitch_rc_rate = 80

```

etracer65, 2018-03-15:
It's difficult to tell much from the log when spinning on the bench without props.  You're probably going to have to capture an in-flight log to see if you can replicate the problem.

I did notice that your accelerometer z-axis was showing over 2G when you did some full throttle runs so it indicates there is a moderate amount of vibration there, but I'm not sure if it's enough to be a problem just by looking at the log.

sygel, 2018-03-16:
Ok here is an in-flight log.  First some half flips at about 180deg/s which work fine.  Then some full throttle punch outs which also were fine.  Then a half flip and another from upside down to normal at  300deg/s where it all goes wrong between 41.9 and 42sec.  When the motors are not spinning all that fast motor 1 starts to go to full throttle without roll or yaw input.

diff is the same as in my previous post.

[BTFL_BLACKBOX_LOG_ALPHA300Q_20180316_085015.BBL.zip](https://github.com/betaflight/betaflight/files/1818454/BTFL_BLACKBOX_LOG_ALPHA300Q_20180316_085015.BBL.zip)

etracer65, 2018-03-16:
Well the good news is that the raw gyro data looks okay and doesn't seem overly noisy.  So after going back and looking at all three flight logs I think there are a couple of different things going on.

**BTFL_BLACKBOX_LOG_ALPHA300Q_20180314_123714.bbl**
This log is the only one that appears to have an obvious and true overflow/inversion from the gyro chip.  At 53.270 seconds the pitch gyro goes from +230dps to -2000dps in about 20ms.  This is not physically possible and is exactly what an ICM gyro overflow inversion looks like.  The gyro overflow logic properly detected this event and zero's the PID's to start recovery at 53.282 seconds.  The quad would have recovered but you disarmed before that could happen.  The only unknown is why the ICM gyro overflowed as you weren't even close to exceeding the rate.  Unfortunately we don't have raw gyro data for this log so it's hard to say whether there was excessive noise affecting the gyro.  There's always the possibility that this particular board has a faulty or overly sensitive gyro - no way to tell.  But Betaflight did perform correctly to stop the overflow and possible flyaway event.

**BTFL_BLACKBOX_LOG_ALPHA300Q_20180314_172444.bbl**
In this log it appears you have a de-sync or some other loss of thrust from motor 4 starting at about 44.170 seconds.  This appears to occur when the motors transition from reversed (inverted) direction to normal.  The lack of thrust from motor 4 causes the quad to go into a roll and pitch tumble that eventually exceeds the gyro overflow detection threshold at 44.795 seconds.  Again the gyro overflow detection properly kicks in to prevent a flyaway event.  Assuming motor 4 recovered thrust control would have returned but again you disarmed before that occurred.

**BTFL_BLACKBOX_LOG_ALPHA300Q_20180316_085015.BBL**
This event appears almost identical to the previous - another de-sync or lost of thrust starting at 41.890.  As before it's right after the motors are changing direction however in this case it's motor 1 that loses thrust.  This again causes the roll/pitch tumble however motor 1 recovers at 42.460 seconds and the quad starts to regain control.  The spin rates never reached the gyro overflow threshold so no overflow protection was triggered or needed.  Again you disarmed before the quad could completely recover.

So multiple de-syncs affecting different motors usually points towards ESC problems.  3D mode which requires stopping/reversing/restarting the motors puts the most stress on ESC's and will expose any flaws they may have.  Perhaps the ESC's you're using are not up to the task.  The uncontrolled flips in the last two logs were undoubtedly caused by loss of thrust to one motor - most likely a de-sync but could also possibly be a loose prop that was slipping.

Whether or not your gyro is faulty in regards to the unexpected overflow/inversion is unknown.  Even if it was, in the first example Betaflight would have recovered the spin and regained control had you not disarmed (but can't really blame you).  No way to really test whether the gyro is faulty and in general the raw data looked okay in the later log.

So I think the first task would be to address and resolve the de-sync issues.  After that you may not have any more problems.

sygel, 2018-03-18:
Thank you for the detailed analysis.
I don't agree however with the assumption that you make that the craft would have recovered.  I was doing these flips about 1 meter above the ground (I don't fly FPV). When I see a problem with the craft I always immediately hit disarm to minimise damage.  But at 1 meter I'm sure the quad was already hitting the grass before I could disarm.
Loose prop is also impossible with the graupner props and motors.

I my opinion there are 2 possible causes, since the ESCs/motors fly just fine with the graupner gr18  FC (I don't think it uses betaflight, but I don't really know what is inside that graupner FC)

1) faulty Kakute  FC
2) problem with betaflight software or settings in 3D mode with less than "optimal" hardware.

I did track down some vibration coming from motors 1 and 4 by checking g forces on the bench while spooling up the motors individually.   I will fix this first and try again.  I will also try some lower startup power( current setting 0.5) in BLheli and maybe different timing (currently high).

Which ESCs would you recommend for 3D mode?

fftunes, 2018-03-18:
Betaflight 3D mode still has zero protection against too hard motor direction transitions afaik. So imho it's pretty safe to blame code for those desyncs, especially since you say your flight style works fine with the graupner fc.

I'm sorry this post doesn't help at all.

etracer65, 2018-03-18:
It's not the flight controller firmware's job to manage the motors - that's the job of the ESC's.  The only thing the flight controller firmware does is command a throttle level for the motor.  Everything else is up to the ESC.  It's possible that Betaflight is exposing weaknesses in the ESC firmware/hardware by not being "gentle" on them but that's only a guess.  Do other flight controllers attempt to work around these problems? - not sure.

etracer65, 2018-03-18:
@sygel Nowhere in your posts was there an indication that you were only doing LOS and flying at low level.  So yes the events would have recovered assuming you were flying with more altitude margin.  This additional information might also help explain the gyro overflow on the first log in that it might have been caused by the impact with the ground which can induce extremely high gyro rates triggering the overflow.

Also, I can't see where you've ever stated which ESC's you were using.  What firmware and version?  I see from your diff that you still appear to be using the default ONESHOT125 protocol.  While not a problem exactly, it does limit your PID loop to 2KHz.  However using better protocols will depend on your ESC and what it supports.

It's hard to "recommend" an ESC as there are so many good ones.  However 3D is still kind of niche and I'm not sure if there are any that are more optimized to support it.  Not a 3D flyer myself so can't offer any suggestions here.

sygel, 2018-03-18:
Ok, here some more info while I'm waiting for the new bearings.

Motors:
https://www.graupner.de/ULTRA-2809-2300KV-brushless-Motor-CCW/S7052/

ESC that works with the Graupner FC in 3D mode with zero issues doing very fast flips and tic-tocs, rainbows,...
https://www.graupner.de/Regler-3D-Copter-Brushless-ULTRA-CONTROL-20A-3D-ohne-BEC/S3057/
This ESC only supports Oneshot, firmware 14.2 (installed by graupner), also tried 14.3 and 14.8

The same ESC with katute and betaflight -> roll of death after mildly aggressive flips

Then I tried to solve this issue and bought these ESCs:
https://www.banggood.com/4X-DYS-XSD-20A-3-4S-ESC-BLHeli_S-Supports-Dshot600-Dshot300-For-High-KV-Motors-p-1187166.html?cur_warehouse=CN
I do use DShot600, as indicated in the diff file.
BLHeli Firmware 16.7

Do these settings do anything if I use DSHOT?
set 3d_deadband_low = 1450
set 3d_deadband_high = 1550
set 3d_neutral = 1500

only this is relevant right?
set dshot_idle_value = 1000

fftunes, 2018-03-18:
@etracer65, imho it IS the firmware's job to command the escs with some... let's call it "common sense".

You wouldn't throw in reverse gear while driving your car at 80mph on the highway.

sygel, 2018-03-18:
If betaflight or blheli could add some optional parameter, I would be willing to test that.
set 0 power between transitions for x milliseconds.  
I’m also going to test if I can see if the graupner fc does this with a slow motion camera...

sygel, 2018-03-19:
If I look at the logs I see that the transition (pos-neg or neg-pos) always takes 10ms.  If I fly inverted and flip back to normal the power to the motors goes first to max negative!  
Is this a fluke in the blackbox logger or do these values actually get send to the ESC?

so flipping inv -> normal
motors around 200 -> 999 max full negative ?
around 1000 full stop
1000->1200 normal spool up

This would explain why all my problems occur from inverted to normal which is the worst possible scenario.

[image]

etracer65, 2018-03-19:
This is just an artifact of the blackbox logging.  It doesn't properly represent the inverted portion of the motor outputs.  Not sure if it's a problem in the viewer or the data written to the log at this point.

sygel, 2018-03-19:
Ok but what values are sent to the ESCs during those 10ms?

etracer65, 2018-03-19:
The code is complex, but here's what I think happens.  As the throttle stick enters the 3d deadband, the throttle will go to the minimum defined by the digital idle in the direction the stick is coming from.  So inverted to entering deadband will spin at reversed -4.5% for default digital idle settings.  However this is only the minimum as the PID controller is still active and stabilizing the quad so the individual motors may be commanded to higher levels.  As the stick exits the deadband on the other side the motors are reversed and subject to the digital idle as the minimum and will spin a +4.5%.  Higher throttle levels above that scale appropriately.  This may not be absolutely correct as the code is quite complex, but I think it gets pretty close.  If you want to look at the code yourself it's in:

https://github.com/betaflight/betaflight/blob/master/src/main/flight/mixer.c

There is no "10ms" period.  You're just seeing the sample rate from your receiver which is about 9ms and the steps in throttle level to reverse the motors.  So there's going to appear to be a ~9ms gap where the throttle reverses but it's just the next reading from the RX that happened to be past the threshold where the previous was below.

There doesn't appear to be any big jump in throttle as the motors reverse - assuming the pilot transitions the throttle through the deadband smoothly.  To get the visualization correct in the blackbox viewer you need to mentally flip the lower half of the motor outputs.  Below is a photoshopped view of what one of your transitions should look like in the blackbox viewer.  Not exactly correct but you get the idea.

If you haven't yet, I would strongly suggest you post any questions about appropriate settings for 3D and/or issues about how to remedy de-syncs in the BLHeli_S rcgroups thread.

https://www.rcgroups.com/forums/showthread.php?2640796-BLHeli_S-Smooth-as-Silk

[image]

fftunes, 2018-03-22:
> However this is only the minimum as the PID controller is still active and stabilizing the quad so the individual motors may be commanded to higher levels. 

This is exactly the problem of betaflight 3D mode. PID commanding motors to higher levels right before transition prevents simultaneous reversing and in worst case leads to desyncs.

sygel, 2018-03-22:
Well I read the code and it does do a little bit to help the motors:

`612            // keep ITerm zero for 250ms after motor reversal`
`613            pidResetITerm();`

But I think the logging needs to be fixed first to see what really goes on around the reversal.
Because it is not just a visual problem in blackbox, if you look at the values it shows max negative just before the transition.  This could be a logging problem like etracer65 says, but if we cannot trust the logs it is hard to know what is going on.  I'll file an issue in the logging section.

fftunes, 2018-03-22:
> Well I read the code and it does do a little bit to help the motors:

Yes but that's after reversing, not before.

fftunes, 2018-03-23:
But yeah, i suggested changes and improvements often enough that usually go ignored since the main devs show zero interest in 3D.

etracer65, 2018-03-23:
@fftunes Rather than criticizing why don't you contribute then?  We're always happy to have people contribute code particularly if it's an area in which they have experience.  And yes, we could use more developers who share an interest in 3D.  None of the current devs have a problem with 3D, but if it's not something we have experience using then it seems unfair to expect a high level of development in this area.  Everybody here is a volunteer.

sygel, 2018-03-23:
If I cannot solve this by changing the startup power or timing in BLheli, I will change the code a little bit.
First thing I will try is 0 power for x milliseconds between transitions.  I think I can add this to the code fairly easy.  If you have other suggestions or advice why this would be a bad idea please let me know.

etracer65, 2018-03-23:
Unfortunately I don't think the zero power at transition will be a good idea as it will cause instability during the reversal (worse than already occurs when stopping/reversing motors).  That's going to be a generalized problem if you do anything affecting the motor power programmatically. So it might improve your de-sync issue but in general lead to worse control and bigger bobbles on transition.

It still goes back to it being the ESC's responsibility to manage the motors.  Any workarounds in the flight controller to "be nicer" during the transition would be a bodge to make up for something the ESC is not doing a good job with.

Have you tried playing with the startup power in BLHeli_S?  I'm not an expert in this area but that seems to be related.  Also there are the motor timing and demag comp that can be tuned but the BLHeli_S thread is the best place for advice.

sygel, 2018-03-23:
> Have you tried playing with the startup power in BLHeli_S

I will try that first, but unfortunately I have to wait a couple of days for new motors, the last roll of death bent the motorshaft...

fftunes, 2018-03-23:
@etracer65, if i can't code things myself (i wouldn't even know where to look for 3D motor things), what else can i do than suggest solutions and improvements based on logic, so to say?

Yet usually the arguments to not change anything in code comes from people that don't fly 3D themselves. Don't take that personal, i respect the work you do for betaflight, your anti taz feature turned very nice.

Anyway back on topic, i see 2 options, either:
- ramp down PID to zero from throttle center towards the further end of 3d deadband, or
- override motor levels to stay at idle for a short amount of time (preferably configurable, 10-25ms will be enough) before reversing.

The latter has already been done by someone but only for 3D-on-a-switch mode, i might try to find it later when i'm at my computer, it's a bit tedious from phone.

fftunes, 2018-03-23:
@etracer65, regarding the instability during reversal - it does not matter. In the very moment when the motors stop and reverse, the quad is dead weight anyway. The pilot knows that and accounts for it. 

I'm convinced the main reason for any wobbles after reversing is because of asynchronous reversing of individual motors.

etracer65, 2018-03-23:
@fftunes If you're not a developer there's nothing wrong with making suggestions for changes.  Just because they don't always get implemented it's not because anybody thinks the ideas are bad or you're being ignored.  Instead it comes down to having available developers that are interested in or have experience with 3D flight and that's an area where the project is lacking.  Developers choose what they work on - there's no assignment of work.  So I personally wouldn't choose to work on 3D flight-related code because:

1. I don't have any 3D capable equipment to test with
2. I'm personally not interested in 3D flight at all

But that's just me and other developers might want to jump on any suggestions you come up with for enhancements.

If you have specific enhancement requests I would suggest creating a separate issue(s) that detail them so they can be tracked more easily rather than buried in this support thread.

sygel, 2018-03-23:
> ramp down PID to zero from throttle center towards the further end of 3d deadband

I don't like this, I need full PID if I'm at throttle center in a stall turn.  Also the FC cannot predict which side of the hysteresis I will exit.

> override motor levels to stay at idle for a short amount of time (preferably configurable, 10-25ms will be enough) before reversing.

Ok, this would add some lag and is an interesting variation to my suggestion of zero power for a couple of ms between reversals.  Why do you think idle is better than zero power?

mikeller, 2018-03-24:
I think the 'zero power' / actively brake the motors suggestion has got some merit here. Ideally, this should take into account how fast the motors are still spinning, to minimise the delay time. Has anybody ever suggested this for addition to the ESC firmware?

KenImhof, 2018-03-26:
I tried for several months before finally being successful with getting 3D reversing working with blheli.  There is a bug in blheli that was eventually addressed in blheli_32 version V32.2.0.1+.   3D reversing is now working great with the latest version of blheli_32 and KISS 24A race edition.  With blheli_32 the esc ramp-up power must be set  to 150%.  Older versions of blheli_32,  blheli_s, and blheli would crash my quad (death roll) each time I attempted inverted flight.

sygel, 2018-03-26:
@KenImhof Can you share a link to the bug.  How was it solved, can it be ported to blheli and blheli_s?  Blheli_32 is closed source right?

@mikeller Part of the discussion is, where should this be fixed,  ESC or FC?  You could argue that it is  the sole responsibility of the ESC.  If a patch can be written for every ESC, without negatively impacting the rest of the flight characteristics (when I'm not reversing, which is more than 99% of the flight time) then fine by me.  But my old graupner ESCs running BLheli 14.2 can do agressive flips, tic-tocs all day long with the original graupner FC.  Both FC and ESC have information the other doesn't have.

Anyway if shutting down the motors for x milliseconds helps it can easily be added to betaflight and later to any BLHeli version.  I'll do some testing the next weeks and share my results.

mikeller, 2018-03-26:
@sygel: My point wasn't about _not_ adding it to Betaflight. It was about the potential to add a better version of it to the ESC, since the ESC has got information about the actual motor speed.

KenImhof, 2018-03-26:
Sygel, I tried to address this problem with blheli, blheli_s, and blheli_32. After about 100 crashes, I was ready to give-up on 3D operation.  I then tried KISS 24A ESCs and it worked perfect with every protocol and MPU I tested.  I eventually contacted the blheli developer and send him my quad to witness the problem and work on a solution.   He then provided an updated blheli_32 release V32.2.0.1, which completely solved the problem with a 150% ramp-up power configuration.  It also does not work well with the auto-timing option.  I do not know if he ported the fix into the other blheli versions. You may want to try and contact him (sskaug) directly on the blheli_32 forum.

sygel, 2018-03-26:
@KenImhof Thank you for this input, I will contact him.  I'm at about 10 crashes now...

I managed to try the zero power patch by changing the mixer.c file.  Compiling and installing went great, I really like this docker setup, great stuff.  The result however was not good on the bench.  Rapid throttle changes stall the motors for about 1 second.

So I tried a slight variation, for 100ms I'm applying the minimum idle throttle after the reversal (so not before).  This seems to work on the bench.

I'll see if I can lower the 100ms and maybe try a combination of the zero power and idle after reversal.

Teracis, 2018-03-26:
Hi Team,

I've just read through this entire list and you lot are clearly working on something that is important to a lot of us 3D flyers.
Are @sygel @fftunes  and @KenImhof in the betaflight slack? Maybe we could get some 3D development discussion going on over there?
Otherwise we've been doing a whole lot of chat on exactly this in ZoeFPV's discord. I've also tried to look through some of the code on this and just keep coming back confused.
The other approach we were thinking about (for FC) was similar to JoeLucid's torque boost feature, where the motors are driven harder through the reverse to get them not to stall.
It's great that @KenImhof has been working with sskaug on the ESC development, as the ESC has direct control of the motor it makes the most sense to make the improvements there, but it's also hardest to make them there.
So far with BL Heli_S I've just tried to tune it as best I can for reversing, which usually ends up in getting the timing as close as possibly and then ramping up startup power, enabling "brake on stop" is also a requirement, even though the documentation says it does nothing, we've shown clear improvement to motor direction changes by enabling it.
I'm no dev either but happy to try and help, maybe we can all put our heads together and work with @mikeller and @etracer65 to get some 3D development happening?

mikeller, 2018-03-26:
@Teracis: Maybe try to get @rotcehdnih on board as well? He's an accomplished 3D flyer and a developer, so his skills would be very helpful here.

Teracis, 2018-03-26:
@mikeller I've spoken to Hector before but didn't realise he's a developer! I'll definitely try to get him on board with the whole "lets get reversing sorted" thing.
I'm expecting the slack will be the best place to have the conversation, should we keep a separate channel so as not to fragment the discussion around the main dev discussion or just see how we go?

mikeller, 2018-03-26:
@Teracis: Yes, we can create a #3d channel.

Teracis, 2018-03-26:
3d-development channel created in he beta flight slack. It would be great to see all of you there!

SparkyJosh, 2018-03-27:
@sygel @mikeller 
some punishing LOS 3d flying from this weekend. and a few lazy fpv 3d with  logs

https://drive.google.com/drive/folders/1v2Kw9a80cV1LxDl-8BNtyfip8zRcjt3-?usp=sharing

stock BF 3.3 and stock esc settings except for brake on stop = 100%

fftunes, 2018-03-27:
@sygel 
> So I tried a slight variation, for 100ms I'm applying the minimum idle throttle after the reversal (so not before). This seems to work on the bench.
Try it before reversal. Simultaneous reversing works fine if all motors are actually at idle speed. Even my old 2204 motors can do this with 6" props. Also i bet you can make it way shorter - 50ms is enough to let a quad completely freak out, so to slow down motors from "a bit above idle" to idle shouldn't take more than 10-20ms max.

And regarding zero power through transition, glad you found it doesn't work well. It could even create more problems since not all escs do actively brake on zero power, like the old kiss 30a for example.

sygel, 2018-03-27:
This is how the patch looks in blackbox.  In this case I added 50ms idle after each reversal. 

[image]

@fftunes I will try it before reversal as well, thanks for all the info.

I added min_idle_time_3d so I can configure this at the field without compiling. I guess I will start with 20ms... 

How does betaflight slack work, do you need to be invited or can you just join?

Teracis, 2018-03-27:
I think you can just join, betaflightgroup.slack.com should be the main and then just join the 3d-development channel. It's not active yet but we'll need to get a list of stuff to work on soon and then get cracking!

fftunes, 2018-03-27:
> But as you can see in the logs, before the reversal it is already about 10ms at idle.

But this is on the bench, right? Individual motors might look much different if PID interfer in flight.

Teracis, 2018-03-27:
If you're using BL Heli_S maybe you can follow these steps first (on standard firmware) to try and get your reversing tuned as well as possible first (unless it's already as good as you can get it):
BL Heli_S Tuning

For ALL tests, safety glasses on (props will be on for step 4 onward), quad held in hand or locked in vice and pointed away from everything, Airmode DISABLED, monitor motor temperature. Take notes for each test to compare.

1. Get all motors spinning in correct directions and 3D mode enabled on FC with default BL Heli_S settings other than direction.
2. Set idle values to not stall in either direction when touched lightly, this is the same process for 2D setups but needs to be done in BOTH directions for 3D.
3. Enable brake on stop.
4. Fit props and safety equipment. Use props intended for flight, or heaviest props intended for flight.
5. Disable Airmode or any similar function.
6. Change motor timing and test which has the smoothest direction changing. This can be subtle and it's worth testing all options (Low, Med/Low, Med, Med/High, High) confirm the final selection by going back and forwards between the other options and the selected best.
7. Change startup power to determine the smoothest direction changing. This can also be subtle, however initial testing has indicated that higher startup power has been the smoothest and quickest to recover from a desync.
8. Check the idle values again, lift these if direction changing is still problematic.
9. Tune PID values (if not already complete) and test hover to confirm no motor heating issues.
10. Test fly and confirm motor direction changes are satisfactory, if not, start at step 4 again, potentially with lighter props.

If a single motor is lagging severely, consider swapping motors around to see if the problem follows the motor (motor issue) or stays on that arm (esc issue) or even swapping the motor bells around or the entire motor for a spare.

That's pasted from somewhere else when we were trying to develop a system, I went from not being able to use 1g+ props for 3D to using them successfully on the micro rig with no changes to FC settings.

sygel, 2018-03-27:
> But this is on the bench, right? Individual motors might look much different if PID interfer in flight.

This is a screenshot of an in-flight reversal, flip inv -> normal,  (without the patch), you can see that the PID is most active after the reversal, just before it does not do a lot.

[image]

Teracis, 2018-03-27:
That would be expected though wouldn't it? Given that the quad is going through a period of essentially uncontrolled movement because the motors are changing direction and therefore the PID controller is building up huge amounts of error as it's got no control over the orientation of the quad during that transition through the motors being stopped.

sygel, 2018-03-27:
Yes, and that is why I'm blocking the PIDs for x ms to let the motors spool up after the reversal.

fftunes, 2018-03-27:
Ok, but in my experience, what throws the quad off the most is: motors not reversing at the same time. Think of the effect of a single motor lagging behind in reversal (because it was commanded higher than the others right before transition).

Teracis, 2018-03-27:
I'm inclined to agree with both of you here, it's very likely worth driving all motors to idle in the current direction, then reversing them all at once with the same command (possibly far higher than idle for a predefined time) whilst also resetting the PID error at the end of the reversal time period.

I say driving them at higher than the other direction's idle speed for reversing as @joelucid has had great success with his "torque boost" feature, which just allows the escs (and therefore motors) to be commanded with a far higher value for a brief moment when changing throttle position. This results in a similar boost one could expect from a closed loop esc control.

This might be all well and good for static reversing, but we'll have to dial this in for complex manoeuvres as well. At least we're all thinking along the same lines!

KenImhof, 2018-03-27:
What works for me is to keep the motor throttle very high.  With Dshot, I am run 20%.  With Multishot, I am running 1350, 1500, and 1650.  A good test is to check the motor reversal on the bench without props and if you can see the motors stop (I look for the prop screw holes) between reversals, it will not work.

sygel, 2018-03-27:
@KenImhof  Wow, 20% is higher than I ever tested (15% was max I tested).  Does this mean that you have problems reversing when you run lower?

I changed the code a bit, I now have the DSHOT idle at  12% and reversal idle at Y%.  So for X ms after the reversal I give it a constant Y%.

If I read all your comments I'm thinking Y = 20% and X = 20ms (originally I thought Y = 6%)

Problem with all this is that I cannot reproduce any of this on the bench.  I can't see any stops (even without the patch).  I think I have to make some setup so I can test with props.  I know some people hold it in their hand, but not me...

KenImhof, 2018-03-27:
I like to run a high idle to prevent bobbles in fast transition moves such as tic-tocs.  For me 20% is still a little low so I will probably patch the CLI so I can try a value of 25%.  I have been flying with BLHELI_32 for around a year without a single desync and strongly recommend upgrading your ESCs along with installing a LOW ESR capacitor on each ESC, which should completely address your motor reversal issues.

sygel, 2018-03-31:
Did some more flying/testing today.  It still has the occasional desync even with max startup power.
I also tried an even bigger window 60ms-  60ms so 120ms in total.  It still flies really good even with such a big window.

Next step for me will be new ESCs, today a saw someone fly 3D with zero issues with the
BLHeli_32 Lumenier 35A.  He had no capacitors and still no issues.

Even if these ESCs do not have the desync issue I will still try this patch to see if it improves handling after a reversal.  Like @fftunes pointed out, it might help starting the 4 motors completely in sync.

I also tried a mid air motor restart feature after desync.  It kind of worked but best is of course to have no desyncs in the first place.

If I detect a desync I apply 3% idle for 120ms and then continue...

[image]

KenImhof, 2018-04-01:
I have not had a desync since the ESC firmware was fixed.  I have been flying for over a year with no issues.  Sample gentle flight from earlier today:
BetaFlight:https://youtu.be/NXWMI2Fgo_U

sygel, 2018-04-01:
@KenImhof great, it is not that I dont’t believe you.  I will also upgrade to blheli_32.

But fact remains that my old graupner Esc with blheli 14.3 flies without desync with graupner fc and always has desyncs with betaflight.

I really should do 2 extra tests with my patch before I switch to Blheli_32, but I am getting tired of crashing...
Try capacitors on my dshot blheli 16.7 escs
Try my graupner Esc

[16 more replies]

closed 2018-06-01 by stale[bot]