betaflight/betaflight #5457: Benewake Lidar on Omnibus F7V2

Ralfde, 2018-03-15:
Can I connect just the Lidar TX to UART7 RX?
I just adjusted Peripherie to UART7 to Benewake lidar.
I use the TF mini.
Or do I need to use the UART1 with RX and TX?

Anything else I need to do?
Where can I see, if it works correctly?

joeostrander, 2018-03-15:
If I'm not mistaken, it's still a compile-only option.

in the target.h:
#define USE_RANGEFINDER
#define USE_RANGEFINDER_TF
#define USE_ALT_HOLD

in CLI:
feature RANGEFINDER
set rangefinder_hardware = TFMINI
serial 1 32768 115200 57600 0 115200
aux 3 22 1 1300 2100 0

of course you'll have to change the serial and aux lines to match your config

After this, you should see data when you're on the sensors tab.

Ralfde, 2018-03-16:
what is the aux for? Could you please compile me a version like this for OmnibusF7V2?

closed 2018-05-05 by Ralfde