betaflight/betaflight #5460: Blackbox sampling rate does not take pid_process_denom into account

flint723, 2018-03-15:
In 3.3.0 and 3.4 for sure, not sure about older versions.  Configurator 10.2.0 (but should not matter).

Sample rate (1 in 4 for 8k and 2khz logging) stays the same, when lowering the pid loop frequency without lowing gyro frequency.

8k gyro, 8k pid, 2khz logging: works correctly, 0-1000hz visible in spectrum analyzer
8k gyro, 4k pid, 2khz logging: file is half the size and 0-500hz visible in spectrum analyzer (1khz log)
8k gyro, 2k pid, 2khz logging: file is 1/4 the size and 0-250hz is visible in spectrum analyzer (500hz log)

Here's a log of 8k8k, 8k4k, and 8k2k with 2khz logging set.
[Blackbox log 8k8k, 8k4k, 8k2k comparison.zip](https://github.com/betaflight/betaflight/files/1816480/Blackbox.log.8k8k.8k4k.8k2k.comparison.zip)

mikeller, 2018-03-21:
@flint723: I think this a configurator problem, and not a firmware problem actually. Since 3.2, the blackbox logging rate is identical to the PID loop rate, and `blackbox_p_ratio` just controls the ratio of full data frames that is written, see https://github.com/betaflight/betaflight-configurator/issues/661#issuecomment-329701843 This change is not yet reflected in the configurator. Can you please open an issue for this in https://github.com/betaflight/betaflight-configurator/issues ?

flint723, 2018-03-21:
Done, Thanks.

mikeller, 2018-03-21:
Thanks. Can you close this one then please?

closed 2018-03-21 by flint723