betaflight/betaflight #5461: Omnibus F4 Pro corner IRC Tramp not working

TwoToneEddy, 2018-03-15:
Hi,
After following the guide here:

https://github.com/betaflight/betaflight/wiki/Board---OMNIBUS-F4-Pro-Corners

And following the configuration example 2 (IRC Tramp on pin M5, Softserial 2) i don't seem to be able to get the VTX control working. SBUS (uart 3) and Smart Port (soft serial 1) work fine but using either the lua script or the OSD i can't change or read the VTX configuration. Strangely though I can put it into pit mode (i'm using a holybro atlatl vtx), when I say i can, the led flashes which indicates pit mode.

I wouldn't normally post something about an issue like this but i've got the same VTX on 2 other quads with an XSRF4PO and i've never had an issue. Do I need to do anything to allow the soft serial to use bidirectional communication?

This is betaflight 3.3.0 on the omnibusf4sd target, this is the first time i've used 3.3.0 and haven't tried reverting to 3.2.x, perhaps I will try that tomorrow.
Thanks

m013, 2018-03-16:
i have the same issue with hakrc 20x20   f4 FC (omnibusf4 target). only pitmode works via softserial.

TwoToneEddy, 2018-03-16:
Yeh, that's what I see. Moved it to a real UART in the end but it would be good to understand what's going on.

m013, 2018-03-18:
i cant afford that. the pinout is wrong, so i stuck with a softserial

jflyper, 2018-03-18:
- Tramp is known to be very picky on where it is connected and how it is controlled (hardware UART or software serial). Even different UARTs on the same FC exhibit different results (see #3692).
- The problem was never investigated in detail. I suspect no one with sufficient tools and knowledge never had the issue.
- For hardware UARTs, people pragmatically found adding weak pull-up to 3V3 would help.
- For software serial, the same pull-up may work, but haven't had any reports of success.

m013, 2018-03-19:
>That being said, I would like to ask those of you having the symptom of "Only pit mode works via softserial", if the exact same setup works flawlessly in any of the previous releases (or nightlies).

i only tried it with some nightly build of betaflight 3.3 , butterflight 3.4.2 and betaflight 3.2.4 -  nothing changed, only pit mode worked.

the vtx can be controlled correctly via 'dedicated smartaudio port' (uart6) on dys f4 pro v2  . 
haven't tried to do softserial on that fc for tramp though

Saxin, 2018-03-20:
@jflyper  - same on my side Omnibus Pro Corner, Tramp HV wired to TX1 (soft serial), doesn't work.
hope you recall I've contacted you regarding this on Slack, support thread.

jflyper, 2018-03-20:
Did you try the weak pull-up workaround?

Saxin, 2018-03-20:
no .
pull to 5v ? what resistor values to check ?

jflyper, 2018-03-20:
R10K to 3V3.

Saxin, 2018-03-20:
ok will try that and update, thanks.

Saxin, 2018-03-23:
@jflyper Remapped Tramp Tel. to led - works perfectly, (BF 3.2.5).
Couldn't reach FC bottom to be able to test weak pull up as suggested - sorry about that.

jflyper, 2018-04-12:
Those of you having trouble with Tramp on hardware UART1_TX of majority of F4 boards (those with the TX1 mapped onto PA9)  may find a working solution now; try latest nightlies.

closed 2018-05-20 by stale[bot]